springfall2008/batpred #4: solcast entity nomenclature

dbrb2, 2023-05-28:
This is a fantastic tool. 
One thing I noticed, was that though I had the solcast integration installed, the entities created didn't have "solcast" in the entity names, so did not match the default config file. Easy to fix - I simply removed the bolt sections below.

I mention it only because the comments in that file suggest they won't need changing, so if it is a change in nomenclature by the solcast integration it might be good to know. Cheers either way!!
  
# Set these to match solcast sensor names (unlikely to need changing)
  pv_forecast_today: sensor.**solcast**_forecast_today
  pv_forecast_tomorrow: sensor.**solcast**_forecast_tomorrow
  pv_forecast_d3: sensor.**solcast**_forecast_d3
  pv_forecast_d4: sensor.**solcast**_forecast_d4

garreth64, 2023-05-29:
I notice the same and have changed the names in the yaml and all good now. Presume this is a change to the Solcast integration.

springfall2008, 2023-05-29:
Are you using this plugin? https://github.com/oziee/ha-solcast-solar

dbrb2, 2023-05-29:
I am, yes

mpartington, 2023-05-30:
How odd, I'm using the solecast integration and am not seeing this issue. I did install it 12 months ago though, so maybe on 'original' entity names if they have changed

tim-mawson, 2023-05-30:
The Solcast integration added the "solcast_" prefix back in June of last year, perhaps you're using an older version?

dbrb2, 2023-05-30:
I'm running 3.0.47, which seems to be the latest - though I did install it last year, so maybe when it was updated it maintained the older names...

tim-mawson, 2023-05-30:
You might be right there, perhaps that's it

springfall2008, 2023-05-31:
I'm going to close this now, I did put a comment in the Readme saying it maybe different for some people