springfall2008/batpred #5: Scheduled discharge?

dbrb2, 2023-05-29:
I noticed that scheduled charge was working fine - however scheduled discharge was being ignored by the system. 

On investigating the givTCP entities, enable charge schedule was "true"
Enable discharge was "false" - and if set to "true" manually, would quickly revert to false 

Could that be why there is a problem? 

I see in config we have: 
```
scheduled_charge_enable: 
   - switch.givtcp_{geserial}_enable_charge_schedule_2
```

But there is no corresponding entry for scheduled discharge enable? 

(note, the _2 suffix was added due to an issue with givTCP recently where all MQTT entities were renamed)

springfall2008, 2023-05-29:
This is handled by inverter mode, which is set to Force Export during the actual discharge slot and then back to Eco afterwards. 

Did you see if the discharge actually went ahead?

 inverter_mode: 
   - select.givtcp_{geserial}_mode

dbrb2, 2023-05-29:
Interesting! I can see from the HA history that I seemed to go into timed discharge mode, though when I checked, MQTT, the actual value of that export was very low. 

The window chosen from the logs was 1630 to 1830 

However, looking at the plot below, showing the export over time, something doesn't seem right - there was a spike, but I think that may have been me manually setting the discharge via the web portal. 

[image]

I'll check it again during the next scheduled force discharge, this time without touching the system, and grab the logs / see what happens

dbrb2, 2023-05-29:
I don't know if this is relevant to the above, but initiallty I was getting errors in the logs: 

`AttributeError: 'PredBat' object has no attribute 'adjust_force_discharge'

The problem seems to be on line 2091 - self.adjust_force_discharge should, I think, be inverter.adjust_force_discharge

Making that change stopped the errors, though there were still plenty of warnings of: 

`2023-05-29 16:46:59.164624 WARNING pred_bat: pred_bat: Entity predbat.car_soc_best not found in namespace `

But perhaps when I made that change I broke something else!

springfall2008, 2023-05-29:
Your fix seemed good and I pushed it, although I don't know if it resolves the force discharge issue
The other warning isn't important it's just when a new sensor is created.
If you can grab a log again as you suggest it could help.

springfall2008, 2023-05-29:
One thing to note is I've found register writes unreliable in my system, so sometimes it doesn't work until a second attempt (5 minutes later).
I've also added a new REST mode which bypasses MQTT and maybe more reliable but I'm still testing it. If you update you could try it out but maybe change one thing at once.

dbrb2, 2023-05-29:
Cheers - I'll update the ticket tomorrow. Thanks for all the work you've put in to this!

dbrb2, 2023-05-30:
Interesting - it looks like the discharge started, then stopped 5 minutes later: 

[image]

[image]

Unfortunately the appdaemon logs only seem to go back a few minutes - I'll see what I can do about that! 

But the logs from just now seem to show the discharge is being requested, just not actioned: 

2023-05-30 17:12:13.168313 INFO pred_bat: Discharge windows now [True, True] [{'start': 1020, 'end': 1080, 'average': 34.91}, {'start': 2400, 'end': 2490, 'average': 34.91}]
2023-05-30 17:12:13.285686 INFO pred_bat: Charge window will be: 2023-05-31 02:00:00+01:00 - 2023-05-31 05:00:00+01:00
2023-05-30 17:12:13.293095 INFO pred_bat: Configuring charge window now (now 05-30 17:12:00 target set_window_minutes 30 charge start time 05-31 02:00:00
2023-05-30 17:12:13.302839 INFO pred_bat: Inverter 0 charge window is 02:00:00 - 05:00:00, being changed to 02:00:00 - 05:00:00
2023-05-30 17:12:13.309803 INFO pred_bat: Next discharge window will be: 2023-05-30 17:00:00+01:00 - 2023-05-30 18:00:00+01:00
2023-05-30 17:12:13.319311 INFO pred_bat: Inverter 0 Adjust force discharge to Timed Export times 17:00:00 - 18:00:00, current mode Timed Export times 17:00:00 - 18:00:00
2023-05-30 17:12:13.325226 INFO pred_bat: Not setting charging SOC as we are not within the window (now 05-30 17:12:00 target set_soc_minutes 30 charge start time 05-31 02:00:00
2023-05-30 17:12:13.331082 INFO pred_bat: Completed run status Discharging

springfall2008, 2023-05-30:
Can you please check your HA log of the Inverter mode?
[image]

springfall2008, 2023-05-30:
Yes, Force export should be somehow set. Not sure why it didn't do that. I have a feeling it's kind of buggy in GivTCP. Do you have debug logs for GivTCP - I keep mine on Info mode?

dbrb2, 2023-05-30:
Oooo....One more thing I noticed: 

`Target_SOC=100`

Does that mean it is trying to export, but also trying to keep the battery at 100% ? I can see that might not work....

dbrb2, 2023-05-30:
`2023-05-30 15:32:14,260 - write - [INFO] - Setting Discharge Slot 1 to: 16:00 - 17:00
2023-05-30 15:32:21,840 - write - [INFO] - Setting Discharge Slot 1 to: 16:00 - 17:00
2023-05-30 16:02:15,216 - write - [INFO] - Setting Battery Mode to: Timed Export
2023-05-30 16:27:14,271 - write - [INFO] - Setting Discharge Slot 1 to: 16:00 - 17:30
2023-05-30 16:27:24,172 - write - [INFO] - Setting Discharge Slot 1 to: 16:00 - 17:30
2023-05-30 16:27:25,147 - write - [INFO] - Setting Discharge Slot 1 to: 16:00 - 17:30
2023-05-30 16:32:14,203 - write - [INFO] - Setting Discharge Slot 1 to: 16:30 - 17:30
2023-05-30 16:32:20,425 - write - [INFO] - Setting Discharge Slot 1 to: 16:30 - 17:30
2023-05-30 17:02:17,616 - write - [INFO] - Setting Discharge Slot 1 to: 17:00 - 17:30
2023-05-30 17:02:19,021 - write - [INFO] - Setting Discharge Slot 1 to: 17:00 - 17:30
2023-05-30 17:02:23,974 - write - [INFO] - Setting Discharge Slot 1 to: 17:00 - 17:30
2023-05-30 17:02:29,909 - write - [INFO] - Setting Discharge Slot 1 to: 17:00 - 17:30
2023-05-30 17:02:34,067 - write - [INFO] - Setting Discharge Slot 1 to: 17:00 - 18:00
2023-05-30 17:02:41,446 - write - [INFO] - Setting Discharge Slot 1 to: 17:00 - 18:00
2023-05-30 17:32:16,694 - write - [INFO] - Setting Discharge Slot 1 to: 17:30 - 18:00
2023-05-30 17:32:18,890 - write - [INFO] - Setting Discharge Slot 1 to: 17:30 - 18:00
2023-05-30 17:32:19,214 - write - [ERROR] - Setting Discharge Slot 1 failed: (<class 'AssertionError'>, AssertionError('Unexpected response from remote end: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response'), <traceback object at 0x7fb28b5780>)
2023-05-30 17:32:28,797 - write - [INFO] - Setting Discharge Slot 1 to: 17:30 - 18:00
2023-05-30 17:32:29,750 - write - [INFO] - Setting Discharge Slot 1 to: 17:30 - 18:00
2023-05-30 17:32:33,949 - write - [INFO] - Setting Discharge Slot 1 to: 17:30 - 18:30
2023-05-30 17:32:40,104 - write - [INFO] - Setting Discharge Slot 1 to: 17:30 - 18:30
2023-05-30 18:02:13,823 - write - [INFO] - Setting Discharge Slot 1 to: 18:00 - 18:30
2023-05-30 18:02:18,691 - write - [INFO] - Setting Discharge Slot 1 to: 18:00 - 18:30
2023-05-30 18:02:25,504 - write - [INFO] - Setting Discharge Slot 1 to: 18:00 - 18:30
2023-05-30 18:02:28,801 - write - [INFO] - Setting Discharge Slot 1 to: 18:00 - 19:00
2023-05-30 18:02:36,838 - write - [INFO] - Setting Discharge Slot 1 to: 18:00 - 19:00
`

From MQTT explorer, I think the set has worked, as the timed slots seem to be 1800-1900 now, but not the actual export 
I assume the reason it has gone 1800-1900 is because the code has realised the desired effect wasn't acheived during the 1600-1700 slot

dbrb2, 2023-05-30:
Ah - I've been doing some digging, trying setting registers by hand: 

To get the system to discharge between 1920 and 1926 local time (BST) I did the following: 

Left the target_SOC unchanged at 100 
Left the "force discharge" unchanged at "normal" 
Set the "discharge start time slot 1" to 1820 
Set the "discharge end time slot 1" to 1826

This seemed to work as expected 

Setting the slot values to 1920 and 1926 had no effect 

Perhaps that is the problem? The discharge slots are set in UTC, but the tool is setting them in BST - so the slot doesn't activate?

dbrb2, 2023-05-30:
Ah - I may have it - and it may have been my fault: 

The inverter time zone was set to GMT+1 - which I chose because currently the local time is GMT+1
It turns out that was a mistake - I should have chosen the GMT timezone and let GE work out the BST offset

That meant my inverter time was double BST - at 1900 GMT inverter time was 1900+1+BST Offset = 2100 

Corrected - meaning hopefully tomorrow will work!

dbrb2, 2023-05-30:
Looks like that has done the trick: 
Set a discharge window of 3 minutes, using local time, now that my inverter agrees with me about what time it actually is - and behaviour was exactly as expected! 

[image]

[image]

springfall2008, 2023-05-30:
That's good.
I pushed an update for timed export that might help make it more robust, but it probably wont impact you.

It maybe worth trying REST mode at some point also, I find my GivTCP MQTT crashes quite often.

dbrb2, 2023-05-31:
Interesting - that seemed to work today, with a but: 

It seems that I hit some kind of battery level bug, so the reported SoC dropped from 51% to 1% 
However the timed export is still set 

Perhaps when the battery reports this low, a timed export shouldn't continue? 

[image]

springfall2008, 2023-05-31:
Have you install the battery update from GE yet that fixes the SOC bug? I suspect it relates to that rather than anything else.

it’s a good point however, to cancel the discharge if the soC is too low, I think it will happen within the next 5 minute slot?

dbrb2, 2023-05-31:
Cheers
I had the battery f/w - as of just now I also have the inverter f/w, so we
will see

Interestingly, your tool initially moved the export slot to between 1800
and 1930 (from 1600-1730) - presumably because less than expected was being
exported

It was though eventually set back to Eco mode before 1930 by the tool, at
1846 - which perhaps was down to the low SoC?

springfall2008, 2023-05-31:
Yes the low soC would cause it to abort exporting

I guess you should contact GE about the glitch?

dbrb2, 2023-05-31:
Yep - I posted on the GE forum, and they pushed me a new f/w version, so we
will see
It is interesting it didn't stop exporting at the next 5 minute window, but
did eventually

Cheers in any event,

Ben

springfall2008, 2023-06-02:
Closing this ticket for now, please open a new one if you have any new issues

closed 2023-06-02 by springfall2008