springfall2008/batpred #7: set_window_notify sends notifications to all phones [feature request]

mervin39, 2023-05-31:
When set_window_notify (also set_discharge_notify, set_soc_notify, set_reserve_notify) are set to True, Home Assistant will send a notification to any phones with the companion app installed. This is fine and useful for me, but my SO doesn't need to know the ins and outs of battery charging. It would be great to have section in apps.yaml to configure which devices receive the notification, instead of all-or-nothing

treforsouthwell-sifive, 2023-05-31:
I'm sure I can do that, I'll put it on the list

springfall2008, 2023-05-31:
I'm sure I can do that, I'll put it on the list

springfall2008, 2023-05-31:
Should be ready for testing now

mervin39, 2023-06-01:
Looks great - testing now. Do you know of an easy way to trigger a notification?

springfall2008, 2023-06-01:
I normally change my set_window_minutes to 1000 and then toggle on or off scheduled charging (so the tool has to changed it back)

mervin39, 2023-06-01:
Yep that did it. Notifications were sent to only the devices in the list.
Thank you!

closed 2023-06-01 by springfall2008