springfall2008/batpred #9: Car Charging doesn't use sensor

paulgee31, 2023-06-02:
Looking at the prebat.py 4.0 source, I can't see any refence to the responses listed in apps.yaml? Am I missing some aspect to how the car charging prediction now works?

Setting the car_charging_planned to a sensor results in:
INFO pred_bat: WARN: Return bad int value EV Disconnected from car_charging_planned
2023-06-02 12:14:21.136105 INFO pred_bat: Previous message repeated 848 times

apps.yaml
car_charging_planned_response:
    - 'yes'
    - 'on'
    - 'true'
    - 'connected'
    - 'ev connected'
    - 'charging'
    - 'paused'
    - 'waiting for car demand'
    - 'scheduled'
    - 'enabled'
    - 'latched'
    - 'locked'
    - 'plugged in'

predbat.py:
   planned = self.get_arg('car_charging_planned', False)
                if isinstance(planned, str):
                    if planned.lower() in self.get_arg('car_charging_planned_response', ['yes', 'on', 'enable', 'true']):
                        planned = True
                    else:
                        planned = False

springfall2008, 2023-06-02:
Good spot, there was a bug in the code after some re-factoring that caused this warning and hence it not working, I've submitted a fix for this now and it should work - please let me know.

The code you cut/pasted is actually correct, it's just it was trying to turn the text into a int() by mistake.

paulgee31, 2023-06-02:
It seems happy now with the fix. I’ll put the car it on charge in a bit.

springfall2008, 2023-06-03:
Resolving, feel free to re-open if you get an issue

closed 2023-06-03 by springfall2008