springfall2008/batpred #10: KeyError: 'geserial'`

dbrb2, 2023-06-06:
It looks like the most recent release as of June 6th may have a small problem? 

`2023-06-06 22:10:00.030098 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/config/appdaemon/apps/batpred/predbat.py", line 2552, in run_time_loop
    self.update_pred()
  File "/config/appdaemon/apps/batpred/predbat.py", line 2065, in update_pred
    load_minutes = self.minute_data_load(now_utc)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/appdaemon/apps/batpred/predbat.py", line 715, in minute_data_load
    entity_ids = self.get_arg('load_today', indirect=False)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/appdaemon/apps/batpred/predbat.py", line 599, in get_arg
    value = self.resolve_arg(arg, value, default=default, indirect=indirect, combine=combine, attribute=attribute, index=index)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/appdaemon/apps/batpred/predbat.py", line 577, in resolve_arg
    item = self.resolve_arg(arg, item, default=default, indirect=indirect)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/appdaemon/apps/batpred/predbat.py", line 584, in resolve_arg
    value = value.format(**self.args)
            ^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'geserial'`

springfall2008, 2023-06-06:
Did you reboot appdeamon as sometimes updates cause the config to misread ?

dbrb2, 2023-06-06:
I rebooted HA - let me try an explicit reboot of appdaemon - one moment!

springfall2008, 2023-06-06:
It’s saying it can’t find geserial in the config. Either the regex didn’t match or there’s an error in the config.

dbrb2, 2023-06-06:
My apologies. I had checked the config - which looked right - but after a whole pi restart, it now seems to be working fine

springfall2008, 2023-06-08:
Closing for now

closed 2023-06-08 by springfall2008