springfall2008/batpred #11: incorrect values in predbat.rates on Agile tariff

mervin39, 2023-06-09:
Before 16:00, predbat.rates shows an incorrect value at 23:00 and 23:30. For me it shows at 38p/kWh. 
The Agile rates are published up until 22:30, and I think batpred uses yesterday's rates to extrapolate forwards? It seems to be fine from 00:00 onwards, using the previous day's values.
38 p/kWh is much higher than the average rate and it's pushing up the "average" attribute of predbat.rates

Sorry if I've explained this badly - let me know if you need anything more!

mervin39, 2023-06-09:
[agile rates - sensor.electricity_XX.txt](https://github.com/springfall2008/batpred/files/11703992/agile.rates.-.sensor.electricity_XX.txt)
[predbat.rates.txt](https://github.com/springfall2008/batpred/files/11703994/predbat.rates.txt)

springfall2008, 2023-06-09:
Now fixed in v4.26

It was using metric_house for missing values, now changed to use the last known rate instead

mervin39, 2023-06-09:
Updated and showing correct values now. Thanks!

treforsouthwell-sifive, 2023-06-09:
Keep in mind it's values not yet decided by Octopus, so it's just a guess (as in the next day until they publish it)