springfall2008/batpred #12: Unable to get 'next' discharge window to fully discharge

adamwylde, 2023-06-19:
Hi

I love the tool you've created. I'm trying to tune it to do what I want whilst I'm waiting for our export MPAN to be issued.

I've modified the config to 
- use the Flux import/export tariffs (until we actually switch)
- calculate_best_discharge: True (but turned off actually charging/discharging using the new switches)
- reduced the SOC min/keep to zero as I dont care if we run out of battery, only that we make the best use of import/export for Flux:
`  best_soc_margin: 0
  best_soc_min: 0
  best_soc_keep: 0
  best_soc_step: 0.25
  best_soc_pass_margin: 0`

As you will see, the best and best10 predictions keep a reserve of around 2kwh prior to the overnight charging window where I would have expected it to maximise discharge, and estimate hitting 0kwh just before the charging window. 

[image]

Any tips to tweak this would be great.

Thanks

springfall2008, 2023-06-19:
If you are using the latest version these options are now in home assistant as sliders/switches and so charging the config file won’t have any impact. Can you check the settings in HA directly?

adamwylde, 2023-06-19:
Ah of course. Yes that’s done something. Will let it play out tomorrow and feedback. 

Thank you

closed 2023-06-20 by adamwylde