springfall2008/batpred #13: Improvement suggestion for inter-day averaging

salewis38, 2023-06-25:
Different users have varying needs for averaging of consumption data - from previous day only to 3/5/7/14-day average, but also the same day last week appears popular for those with a weekly usage cycle. Also, giving yesterday's consumption a higher weighting than earlier days is a useful technique. 

Consider mirroring the more flexible scheme of weighted averages that I recently introduced in palm.py. It can accommodate (almost) any regime a user could dream up. Here are some examples:

# Load history is a weighted average of actual load from previous days.
    # Uncomment required settings or make your own using positive integers only. Examples:
    # Most recent day only
    # load_hist_weight = [1]
    # 3-day average
    load_hist_weight = [1, 1, 1]
    # 7-day average
    # load_hist_weight = [1, 1, 1, 1, 1, 1, 1]
    # Same day last week - useful if, say, Monday is always wash-day
    # load_hist_weight = [0, 0, 0, 0, 0, 0, 1]
    # Weighted average (a more extreme example)
    # load_hist_weight = [4, 2, 2, 1, 1, 1, 1]

springfall2008, 2023-07-03:
I'll consider adding it in future, although to be honest I didn't see much change when I moved from -7 days to the average of -7 and -14 days (which is supported) so I wonder if this makes a big impact?

springfall2008, 2023-07-18:
This is now implemented/released. Please test when you get time.