springfall2008/batpred #15: Not an issue - Advice

robbstech, 2023-07-16:
Great work by the way.
I don't have an electric vehicle, I also only have 4 1.3kw Solar panels, and to add insult to injury, I don't get paid for export. I am on the agile tariff. So what would be the best settings for for this? I have an 8.2kwh batter with a DoD of 100%

springfall2008, 2023-07-16:
It should just work, remember to set export to 0

robbstech, 2023-07-17:
Amazing thank you. Sorry for the questions. But posting on here in case it also helps other people... My setup is working exactly as it should but I am getting the following error. 2023-07-17 20:11:24.183963 INFO pred_bat: WARN: Inverter 0 Trying to write charge_start_time to 23:30:00 didn't complete got 00:00:00.

Which shows in home assistant as Changed to Warn - Inverter 0 write to charge_start_time failed. But I am not using the REST URL because I couldn't get that to work. I am not sure what the REST URL should be.

robbstech, 2023-07-17:
Unfortunately that never worked, It just threw error after error. But I assume the URL was wrong for my particular instance

robbstech, 2023-07-17:
Ah to be clear. I run home assistant in docker and also run appdaemon within docker. So my setup is quite different from your documentation

springfall2008, 2023-07-17:
Did you try changing the polling time to 20 seconds or more ? That might fix your timeout errors.

robbstech, 2023-07-18:
I can't find that setting anywhere. Sorry

robbstech, 2023-07-18:
I did and it seems to have worked. I am not using the rest url though. I don't know what the URL should be because I am not using mass.io I am using Home Assistant and GivTCP both in docker containers. Will see how it goes

springfall2008, 2023-07-18:
Sure no problem, REST isn't a must have

robbstech, 2023-07-19:
Quick update. Working great. Not sure if I have fully set it up correctly or it hasn't had enough time to predict usage etc. But My battery is always going flat lol. Solar has been terrible though. Sunny cumbria for you. I just want to thank you for your great work and support!

springfall2008, 2023-07-19:
There’s probably something not right if you aren’t getting the right charge level. Maybe post your battery chart ?

robbstech, 2023-07-19:
I need to watch the rest of your videos and set it up correctly I think. I just wanted to let you know that it's working with no errors now. And I am grateful for the work you have put into this. Which battery chart do you mean? The graph for the SOC?

robbstech, 2023-07-19:
2023-07-19 22:20:23.066932 INFO pred_bat: Evalute trigger {'name': 'large', 'minutes': 60, 'energy': 1.0} results off total_energy 0.0
2023-07-19 22:20:23.072395 INFO pred_bat: Evalute trigger {'name': 'small', 'minutes': 15, 'energy': 0.25} results off total_energy 0.0
2023-07-19 22:20:23.077241 INFO pred_bat: Best charging limit socs [0.94, 2.19, 8.19, 8.19, 8.19, 8.19, 8.19, 8.19, 8.19, 8.19, 8.19, 8.19, 8.19, 8.19] export [] gives import battery 2.31 house 0.6 export 2.15 metric 134.61 metric10 134.63
2023-07-19 22:20:23.105639 INFO pred_bat: Charge window will be: 2023-07-20 03:30:00+01:00 - 2023-07-20 04:00:00+01:00
2023-07-19 22:20:23.106515 INFO pred_bat: Not setting charging window yet as not within the window (now 07-19 22:20:00 target set_window_minutes 30 charge start time 07-20 03:30:00
2023-07-19 22:20:23.107442 INFO pred_bat: Setting ECO mode as no discharge window planned
2023-07-19 22:20:23.108351 INFO pred_bat: Inverter 0 Adjust force discharge to Eco times None - None, current mode Eco times 00:00:00 - 00:00:00
2023-07-19 22:20:23.109182 INFO pred_bat: Not setting charging SOC as we are not within the window (now 07-19 22:20:00 target set_soc_minutes 30 charge start time 07-20 03:30:00
2023-07-19 22:20:23.109925 INFO pred_bat: Adjust reserve to default (as set_reserve_enable is true)
2023-07-19 22:20:23.110727 INFO pred_bat: Inverter 0 Current reserve is 4.0 already at target
2023-07-19 22:20:23.111464 INFO pred_bat: Completed run status Idle
2023-07-19 22:20:23.118443 WARNING AppDaemon: Excessive time spent in callback 'run_time_loop() in pred_bat', Thread 'thread.thread-3' - now complete after 23.117893 seconds (limit=10)

robbstech, 2023-07-19:
[image]

springfall2008, 2023-07-19:
Sorry I mean the apex chart that predicts the battery levels in the future

robbstech, 2023-07-19:
Oh I haven't got that far yet. I am still getting my head around appdaemon lol. Will do that next

robbstech, 2023-07-21:
Quick update. All seems to be running as I would expect. I added all the charts to get an insight of what's going on. 
[image]

springfall2008, 2023-08-06:
Resolving this for now, please raise any new issues again

closed 2023-08-06 by springfall2008