springfall2008/batpred #16: Fluctuating Target SoC

macray76, 2023-07-18:
Hi Trefor.

Not sure this is an issue with the calculations or just my own specific setup (using latest v6.7) but last night the calculated target SoC was fluctuating every 5 minutes.

My Zappi was charging the car too so the battery was discharging into the EV when the target was below the actual SoC (this is something I would prefer it not to do even if it means the actual SoC ends up higher than the target SoC at the end of the final Intelligent Octopus cheap charging window at 5:30am).

But because the target SoC would move between either side of the actual SoC, the battery was charging then discharging into the EV. Solcast forecast wasn't changing - everything should be static.

This is an extract of the log showing it moving around every 5 minutes:
`2023-07-18 00:00:05.061102 INFO pred_bat: predict base end_record 07-18 23:30:00 final soc 2.05 kwh metric 160.74 p min_soc 0.38 @ 07-18 01:15:00 kwh load 33.38 pv 12.48
2023-07-18 00:00:05.064066 INFO pred_bat:          [  00:00,   01:00,   02:00,   03:00,   04:00,   05:00,   06:00,   07:00,   08:00,   09:00,   10:00,   11:00,   12:00,   13:00,   14:00,   15:00,   16:00,   17:00,   18:00,   19:00,   20:00,   21:00,   22:00,   23:00]
2023-07-18 00:00:05.066968 INFO pred_bat:     SOC: [   4.48,    1.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.42,    0.46,    0.67,     1.2,    1.78,    2.69,    3.64,    4.31,    4.74,    4.92,    5.03,    4.94,    4.65,    4.03,    3.16,    2.37]
2023-07-18 00:00:05.069773 INFO pred_bat:   STATE: [  g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g~be+,   g<be-,   g~be+,   g~be+,   g~be-,   g~be+,   g~be+,   g~be+,   g~be+,   g~be+,   g~be+,   g~be+,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-]
2023-07-18 00:00:05.073700 INFO pred_bat:    LOAD: [    0.0,    4.26,    8.59,   13.48,   17.93,   22.27,   22.82,   23.26,   23.81,   24.51,    25.2,    25.8,   26.57,    27.2,   27.77,   28.44,   29.04,   29.67,   30.12,   30.53,    30.9,    31.5,   32.32,   33.08]
2023-07-18 00:00:05.076779 INFO pred_bat:      PV: [    0.0,     0.0,     0.0,     0.0,     0.0,     0.0,    0.07,    0.37,    0.91,    1.62,    2.52,    3.68,    5.08,    6.66,    8.23,     9.6,   10.66,   11.49,   12.06,   12.38,   12.47,   12.48,   12.48,   12.48]
2023-07-18 00:00:05.079809 INFO pred_bat:  IMPORT: [    0.0,    1.32,     4.7,    9.59,   14.04,   18.37,   18.85,    19.0,   19.06,    19.1,    19.1,    19.1,    19.1,    19.1,    19.1,    19.1,    19.1,    19.1,    19.1,    19.1,    19.1,    19.1,    19.1,    19.1]
2023-07-18 00:00:05.089595 INFO pred_bat:  EXPORT: [    0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0]
2023-07-18 00:00:05.092586 INFO pred_bat:     CAR: [    0.0,    3.89,     8.8,    13.7,   18.61,   23.52,   23.52,   23.52,   23.52,   23.52,   23.52,   23.52,   23.52,   23.52,   23.52,   23.52,   23.52,   23.52,   23.52,   23.52,   23.52,   23.52,   23.52,   23.52]
2023-07-18 00:00:05.106100 INFO pred_bat:  METRIC: [    0.0,    9.87,   35.21,   71.89,  105.29,  137.81,   150.9,  156.65,  159.29,  160.74,  160.74,  160.74,  160.74,  160.74,  160.74,  160.74,  160.74,  160.74,  160.74,  160.74,  160.74,  160.74,  160.74,  160.74]
2023-07-18 00:00:05.774754 INFO pred_bat: Battery has 1.25 hours left - now at 4.48
2023-07-18 00:00:08.035205 INFO pred_bat: Inverter 0 Current charge Limit is 40.0 % and new target is 34 %

2023-07-18 00:05:03.842971 INFO pred_bat: predict base end_record 07-18 23:30:00 final soc 2.08 kwh metric 159.14 p min_soc 0.38 @ 07-18 01:20:00 kwh load 33.12 pv 12.52
2023-07-18 00:05:03.845925 INFO pred_bat:          [  00:05,   01:05,   02:05,   03:05,   04:05,   05:05,   06:05,   07:05,   08:05,   09:05,   10:05,   11:05,   12:05,   13:05,   14:05,   15:05,   16:05,   17:05,   18:05,   19:05,   20:05,   21:05,   22:05,   23:05]
2023-07-18 00:05:03.848998 INFO pred_bat:     SOC: [   4.38,    1.28,    0.38,    0.38,    0.38,    0.38,    0.38,    0.39,    0.38,    0.49,    0.75,    1.23,    1.94,    2.86,    3.75,    4.38,    4.82,     5.0,    5.09,    4.93,    4.65,     4.0,    3.12,    2.34]
2023-07-18 00:05:03.851784 INFO pred_bat:   STATE: [  g~be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g~be-,   g~be+,   g~be+,   g~be+,   g~be+,   g~be+,   g~be+,   g~be+,   g~be+,   g~be+,   g~be+,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-]
2023-07-18 00:05:03.854619 INFO pred_bat:    LOAD: [    0.0,    4.38,     8.7,   13.58,   18.04,   22.05,   22.58,   23.03,   23.68,    24.3,   24.97,   25.66,   26.34,   26.95,   27.56,   28.24,   28.84,   29.46,    29.9,   30.32,   30.66,   31.29,   32.12,   32.87]
2023-07-18 00:05:03.857382 INFO pred_bat:      PV: [    0.0,     0.0,     0.0,     0.0,     0.0,    0.01,    0.09,    0.41,    0.97,    1.71,    2.65,    3.85,    5.27,    6.86,    8.41,    9.74,    10.8,   11.63,   12.16,   12.44,   12.51,   12.52,   12.52,   12.52]
2023-07-18 00:05:03.860205 INFO pred_bat:  IMPORT: [    0.0,    1.43,     4.9,    9.78,   14.24,   18.24,   18.69,   18.83,   18.92,   18.92,   18.92,   18.92,   18.92,   18.92,   18.92,   18.92,   18.92,   18.92,   18.92,   18.92,   18.92,   18.92,   18.92,   18.92]
2023-07-18 00:05:03.863068 INFO pred_bat:  EXPORT: [    0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0]
2023-07-18 00:05:03.866077 INFO pred_bat:     CAR: [    0.0,    4.06,    8.96,   13.87,   18.78,   23.28,   23.28,   23.28,   23.28,   23.28,   23.28,   23.28,   23.28,   23.28,   23.28,   23.28,   23.28,   23.28,   23.28,   23.28,   23.28,   23.28,   23.28,   23.28]
2023-07-18 00:05:03.868986 INFO pred_bat:  METRIC: [    0.0,   10.72,   36.73,   73.38,   106.8,  136.82,  149.95,   155.4,  159.14,  159.14,  159.14,  159.14,  159.14,  159.14,  159.14,  159.14,  159.14,  159.14,  159.14,  159.14,  159.14,  159.14,  159.14,  159.14]
2023-07-18 00:05:04.111692 INFO pred_bat: Battery has 1.25 hours left - now at 4.38
2023-07-18 00:05:23.988556 INFO pred_bat: Inverter 0 Current Reserve is 34.0 % and new target is 42 %

2023-07-18 00:10:03.969237 INFO pred_bat: predict base end_record 07-18 23:30:00 final soc 2.08 kwh metric 160.49 p min_soc 0.38 @ 07-18 01:20:00 kwh load 32.92 pv 12.52
2023-07-18 00:10:03.972195 INFO pred_bat:          [  00:10,   01:10,   02:10,   03:10,   04:10,   05:10,   06:10,   07:10,   08:10,   09:10,   10:10,   11:10,   12:10,   13:10,   14:10,   15:10,   16:10,   17:10,   18:10,   19:10,   20:10,   21:10,   22:10,   23:10]
2023-07-18 00:10:03.974940 INFO pred_bat:     SOC: [   4.09,    0.93,    0.38,    0.38,    0.38,    0.38,    0.38,    0.39,     0.4,    0.51,    0.78,    1.29,    2.05,     2.9,    3.82,    4.47,    4.84,    5.01,    5.09,     4.9,    4.62,    3.95,    3.06,    2.28]
2023-07-18 00:10:03.977675 INFO pred_bat:   STATE: [  g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g~be+,   g~be+,   g~be+,   g~be+,   g~be+,   g~be+,   g~be+,   g~be+,   g~be+,   g~be+,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-]
2023-07-18 00:10:03.980556 INFO pred_bat:    LOAD: [    0.0,    4.55,    8.87,   13.75,   18.22,   21.89,   22.42,   22.88,   23.53,   24.16,   24.84,   25.51,   26.16,   26.84,   27.41,   28.04,   28.69,    29.3,   29.73,   30.17,    30.5,   31.14,   31.99,   32.73]
2023-07-18 00:10:03.983367 INFO pred_bat:      PV: [    0.0,     0.0,     0.0,     0.0,     0.0,    0.01,    0.11,    0.45,    1.03,    1.78,    2.75,    3.96,     5.4,    6.99,    8.53,    9.84,   10.88,   11.68,   12.19,   12.45,   12.51,   12.52,   12.52,   12.52]
2023-07-18 00:10:03.986270 INFO pred_bat:  IMPORT: [    0.0,    1.55,    5.35,   10.23,   14.69,   18.36,   18.78,   18.91,    19.0,    19.0,    19.0,    19.0,    19.0,    19.0,    19.0,    19.0,    19.0,    19.0,    19.0,    19.0,    19.0,    19.0,    19.0,    19.0]
2023-07-18 00:10:03.989287 INFO pred_bat:  EXPORT: [    0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0]
[... cut ...]
2023-07-18 00:15:03.993890 INFO pred_bat:  METRIC: [    3.0,   38.46,   71.04,  107.79,  141.51,  166.72,  167.34,  167.34,  167.34,  167.34,  167.34,  167.34,  167.34,  167.34,  167.34,  167.34,  167.34,  167.34,  167.34,  167.34,  167.34,  167.34,  167.34,  167.34]
2023-07-18 00:15:04.244884 INFO pred_bat: Battery has 48.0 hours left - now at 4.19
2023-07-18 00:15:38.917053 INFO pred_bat: Inverter 0 Current Reserve is 45.0 % and new target is 42 %`

Here are my full logs starting at around 11pm to about 12:15am:
https://pastebin.com/dyBC0DFD

Here is my apps.yaml:
https://pastebin.com/X7ZcSmch

Cheers!

springfall2008, 2023-07-18:
One thing that might help is if you set

'car_charging_soc' in your apps.yml to a sensor that picks up the cars current state of charge.

This is because on each prediction the state of charge is being reported as 0 and so the predictions are changing.

macray76, 2023-07-18:
Ok will add that back - it used to be selected but on an update I have forgotten to uncomment and correct the sensor.

Thanks!

springfall2008, 2023-07-18:
Let me know if that resolves the issue, it might do but I'm not totally sure

springfall2008, 2023-07-19:
I also added something to try to some extra fluctuations in the latest release

macray76, 2023-07-19:
Thank you! Will see what happens tonight and report back

macray76, 2023-07-20:
Unfortunately my target SoC was 100% last night so I didn't get to test it fully.

I'm sure the issue is just unique to me so happy to close this issue for now.

closed 2023-07-20 by macray76