springfall2008/batpred #17: 0% Error

thewhale21, 2023-07-19:
When batteries are over voltage and go from 100% to 0% SoC due to a bug in GE's reporting. BatPred misscalculates SoC targets as it thinks there is less battery capacity. 

Saw this today on a 2 * 5.2 setup, where first one battery "faulted" and only 4.2kws was used for calcs, then when my second one over voltaged it went up to 5.1 for some reason. 

I Know GE were meant to fix this at some point in the future. But not sure what update its in.

springfall2008, 2023-07-19:
I'm not sure exactly what the issue is or how to workaround it, maybe if you provide more details? Are you using REST mode as there is a battery capacity workaround in there that might work.

thewhale21, 2023-07-19:
Im not entirely sure which metric it was that was miss reporting. But when one of my 2 batteries went offline due to it being overcharged. the below was showing Actual as 5.1Kwh.
[image]

Now that i've discharged the battery down to normal voltage limits the Actual is showing correctly again.

I assume when the battery goes into this state BatPred is unable to pull something from GivTCP so it just uses the value for the single battery. 

As i said its a bug with GE's reporting on SoC when battery is to full, :) Probably not worth the time to look into as they are meant to be fixing soon.

treforsouthwell-sifive, 2023-07-19:
The data comes from GivTCP, it's not something Predbat calculates itself. 

I take you are using battery scaling to 0.8?

thewhale21, 2023-07-19:
Yup, Ah okay. Thought that might be the case :)

in theorey it should never be a problem as batteries should discharge them selves down to a safe level anyway and batpred was just recalculate. Just makes an ugly squiggle in my chart!

But thanks for checking :)

springfall2008, 2023-07-19:
The data comes from GivTCP, it's not something Predbat calculates itself. 

I take you are using battery scaling to 0.8?

closed 2023-07-19 by thewhale21