springfall2008/batpred #19: REST error

macray76, 2023-07-25:
I'm getting the following error:

```
2023-07-25 10:10:04.340105 WARNING pred_bat: ------------------------------------------------------------
2023-07-25 10:10:04.340366 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2023-07-25 10:10:04.340596 WARNING pred_bat: Worker Ags: {'id': '69099532cee14d6ca41329e600f77a94', 'name': 'pred_bat', 'objectid': '28934eeb431f40c8870d421805df2dce', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x7f42499d1d90>>, 'pin_app': True, 'pin_thread': 1, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-1'}}
2023-07-25 10:10:04.340769 WARNING pred_bat: ------------------------------------------------------------
2023-07-25 10:10:04.343343 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/config/appdaemon/apps/batpred/predbat.py", line 4044, in run_time_loop
    raise e
  File "/config/appdaemon/apps/batpred/predbat.py", line 4040, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/appdaemon/apps/batpred/predbat.py", line 3549, in update_pred
    metric, self.charge_limit_percent, import_kwh_battery, import_kwh_house, export_kwh, soc_min, soc, soc_min_minute = self.run_prediction(self.charge_limit, self.charge_window, self.discharge_window, self.discharge_limits, self.load_minutes, pv_forecast_minute, save='base', end_record=end_record)
                                                                                                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/appdaemon/apps/batpred/predbat.py", line 1854, in run_prediction
    save, self.time_abs_str(end_record + self.minutes_now), self.dp2(final_soc), self.dp2(final_metric), self.dp2(soc_min), self.time_abs_str(soc_min_minute), self.dp2(final_load_kwh), self.dp2(final_pv_kwh)))
                                                                     ^^^^^^^^^
UnboundLocalError: cannot access local variable 'final_soc' where it is not associated with a value

2023-07-25 10:10:04.343602 WARNING pred_bat: ------------------------------------------------------------
```

Config file is using the standard REST connection:
```
  givtcp_rest: 
   - 'http://homeassistant.local:6345'
```

However, when I comment out the REST comment so it uses the HA entity control, predbat starts working fine. Switching back to REST control and predbat continues to work fine.

This has happened a couple of times now.

It seems like if the communication via REST becomes broken then it doesn't try to recover that connection other than restarting predbat.

springfall2008, 2023-08-06:
I did submit a fix to master for this specific issue that will be in the next release, but I don't know what the root cause of you losing REST communication is, I've not seen it myself. I would suspect an issue with GivTCP, maybe your polling time is too short?

springfall2008, 2023-08-17:
Resolving this ticket, please raise a new one if you have any debug information and see the issue again

closed 2023-08-17 by springfall2008