springfall2008/batpred #20: New feature request - Stop battery discharge whilst EV charging

macray76, 2023-07-28:
I have a new feature request (unless it is already there and I've missed it!)

Currently when the EV is charging on grid power (in my case during an Intelligent Octopus charging slot) and the target SoC is below the battery's current SoC, the battery discharges into the EV.

To reduce unnecessary battery cycles, I'd prefer if there was an option to hold the existing battery SoC (i.e. disable discharging) whilst in an active charging slot (or for non-IO users if it detects the EV is charging from grid).

I know this might mean that by the end of the night's cheap rate the battery SoC stays higher than the target SoC meaning a full battery earlier than planned resulting in exporting solar (although I have a solar diverter to avoid this) but I would rather export solar during the day than charge my EV direct from the battery overnight.

Is this easy to do or is it a silly idea because of something I've missed?

springfall2008, 2023-08-06:
I think this is doable provided you have your EV configured with Predbat so it knows when it will be charging, I'll try to add this feature over the next week.

macray76, 2023-08-14:
Thank Trefor. I will test it next time I am due to charge the car but the battery doesn't need charging. With the rubbish weather at the moment this might take a few days!!!

springfall2008, 2023-08-23:
Now resolved, others have used this feature and it's working

closed 2023-08-23 by springfall2008