springfall2008/batpred #21: [Feature Request] Manage Export Limitation

kingjpc, 2023-08-03:
My system has an export limitation in place, I have been playing around with a simple script to try and optimize charging so that when we generate > Export we can pipe the spare into the battery. This has meant that instead of capping out at 5kw it has been able to cap closer to the total PV value so the system. 

I would be nice if batpred supported this sort of thing, it would improve generation for anyone where PV > Invertor output.

springfall2008, 2023-08-06:
I suspect the export limitation could be modelled, does it just cap your total export? What happens if you generate or try to force discharge above the limit?

kingjpc, 2023-08-07:
Test 1: battery full, solar export < 3.68AC, putting batter on full export
Uses solar first and then the battery to get to the max export limit.

Test 2: battery full, solar should be > 3.68AC
System appear to limit solar production to keep within the export limit

Test 3: battery has space, solar generation > 5AC
System will allow solar to be higher than ac output as it can send straight to battery. (You may already model this) holding onto space in the battery till hours here > AC limit will mean we can generate and export more power

It looks like they try to stay 10% below the export limit as well

springfall2008, 2023-08-29:
Okay, I think I can add something to the model for export limit, will backlog this

springfall2008, 2023-08-30:
Release v6.26 now supports this feature, please can you test?

kingjpc, 2023-09-01:
I have this running now with the limit set to 3600 It does appear to do something but Im not sure its working as expected. Although the old graph also feels wrong to me. 
Seems to show best grid as -1000(WithoutLimit) and -736(WithLimit)
[image]
[image]

treforsouthwell-sifive, 2023-09-01:
I think it had no impact as at no point in that chart do you export above 3.6kw anyhow, that would be when the red is below the line near the bottom of the chart?

kingjpc, 2023-09-01:
That the bit that confuses me as I have exported today at 3.4kw but I don't seem to see that in the graph. The PV-Load-Battery should be the export no? You can see the battery charging up then once its full the Grid Export starts but it flat lines at 1000w/736w.

springfall2008, 2023-09-01:
The chart is future looking and doesn’t show the past

kingjpc, 2023-09-01:
I did a quick roll back to v6.25 this is what I expected the graph to look like in the other pictures i sent.
[image]

springfall2008, 2023-09-01:
This does seem odd, I’ll take a look tomorrow

springfall2008, 2023-09-01:
Okay, I've made a new release - silly mistake. Please update tomorrow and try again.

kingjpc, 2023-09-02:
Looks a lot better limits at the set value correctly.

closed 2023-09-02 by kingjpc

springfall2008, 2023-09-02:
Great, can’t believe I didn’t notice I was a factor of 5 out, it’s mostly as I don’t tend to export