springfall2008/batpred #22: Option to minimise import

alfwro13, 2023-08-05:
Is there an option somewhere where you can minimise grid imports?

springfall2008, 2023-08-06:
This is what Predbat does by default, assuming import costs more than export. I'm not sure if you have a specific scenario or setting in mind?

alfwro13, 2023-08-06:
What I meant is that by minimising grid imports I want to minimise battery cycles - certain batteries have limited number of cycles and on Octopus flux the script will want the battery charged 100% as the night import rates are little bit lower then the day export rates. 
In my my current setup I have a HA template that will looks at the battery SoC, next day forecast and historic usage and only charge the minimum amount that is needed + small safety margin.
BTW,  I wish I had found your repository months ago. Batpred is quite impressive!

springfall2008, 2023-08-09:
It's doing this as the export value is higher, so it saves you money to charge up more. You could set your min_metric_improvement to something positive eg. 5 - this means it won't charge more unless you save at least 5p. You might need to tune it up and down and look at the chart to see what you get.

divenal, 2023-08-09:
Does the script implement any notion of round-trip losses on charging - are those typically 10% or so, so that there's little benefit in importing for export unless there's at least 10% profit in it ?

springfall2008, 2023-08-09:
Yes you can set inverter loss, charging loss and discharging loss in the HA Gui by changing those input_number entities

springfall2008, 2023-08-23:
Resolving for now, please open a new ticket if you have questions

closed 2023-08-23 by springfall2008