springfall2008/batpred #23: Octopus Energy update (V8)

Speshman, 2023-08-05:
Since the Octopus energy was updated https://github.com/BottlecapDave/HomeAssistant-OctopusEnergy it seems that batpred can no longer read the rates from it.
[image]

`  # Set import and export entity to point to the Octopus Energy plugin
  # automatically matches your meter number assuming you have only one (no need to edit the below)
  # Will be ignored if you don't have the sensor
  # Or manually set it to the correct sensor names if the automatic match doesn't work e.g:
  # sensor.octopus_energy_electricity_xxxxxxxxxx_xxxxxxxxxxxxx_current_rate
  # sensor.octopus_energy_electricity_xxxxxxxxxx_xxxxxxxxxxxxx_export_current_rate
  metric_octopus_import: 're:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_current_rate)'
 # metric_octopus_export: 're:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_export_current_rate)'
  
  # Or set your actual rates across time for import and export
  # If start/end is missing it's assumed to be a fixed rate
  # Gaps are filled with metric_house from above
  #rates_import:
  #  -  start: "00:30:00"
  #     end: "04:30:00"
  #     rate: 7.5
  #  -  start: "04:30:00"
  #     end: "00:30:00"
  #     rate: 40.0
  #     
  rates_export:
    -  rate: 0`

[image]

sensor.octopus_energy_electricity_22l4330885_2199998082940_current_rate is what my rate sensor is called, just in case your auto discovery is looking for something different

springfall2008, 2023-08-06:
I've submitted a fix to master, will release it once someone else has tested it

adamwylde, 2023-08-06:
I was seeing the same problem with Octopus 8.0.0… I’ve overwritten predbat.py with the one from master and my rates have returned and it looks like its correctly calculated my upcoming flex discharge window correctly, and tomorrows charge window, so I think the fix is good.

dbrb2, 2023-08-06:
I can confirm this seems to have fixed the problem - I have only been running it for a few minutes, but the missing rates have reappeared. Thanks!

springfall2008, 2023-08-06:
Thanks, I've made a release of this code now

closed 2023-08-06 by springfall2008