springfall2008/batpred #28: Lost helper card

Speshman, 2023-08-17:
More than likely something I've done but I seem to have lost the helper card from my dashboard, is there an easy way to readd the whole card?

springfall2008, 2023-08-23:
Do you mean this? https://github.com/springfall2008/batpred/blob/main/example_dashboard.yml

springfall2008, 2023-08-24:
There isn't a default card for Predbat I'm afraid.

Speshman, 2023-08-26:
Perhaps my terminolgy is off.

When first installing predbat it autopopulates several cards on the main dashboard, one of which is titled helper and has various toggles, offsets etc on it.  This is the card I have 'misplaced' and given that it was autopopulated during install I was hoping to reclaim it

springfall2008, 2023-08-29:
I think this is when you had your dashboard set to automatic rather than manual, Predbat itself doesn't have such a card but the above link maybe useful for you.

springfall2008, 2023-08-31:
Resolving this ticket, please re-open if you have any further questions

closed 2023-08-31 by springfall2008