springfall2008/batpred #30: README improvement: Adding info to help others to contribute

iainfogg, 2023-08-23:
Hi, I think it'd be good to update the README with some info to help others to contribute, if you're up for them doing that.

Things like:

- Getting a development environment set up, including with the right Python version, required libraries (and library versions) etc.
- Any preferences you have for people contributing, creating pull requests etc.
- Any standards or approaches you want people to follow
- Anything else they'd need to know to be able to work on the code

There may be other things, but they're what came to mind.

treforsouthwell-sifive, 2023-08-23:
Development wise I don't actually run the code locally, I edit the Python with VSCode directly on my on HA machine (mounted via SAMBA) and then test it there.

Anyone is welcome to create a pull request, of course I'll only accept things that are generic and not hard-wired for a particular use case.

I'll try to update the Readme soon. 

Are you thinking of contributing?

iainfogg, 2023-08-23:
Yes, I was thinking of it. I was going to have a crack at a quick README update to fix a few missing line breaks and stuff, and I was maybe going to offer to help with moving the README content to a proper documentation site if that would help.

I was also going to suggest some thoughts I had around documentation content after having just got set up with Predbat, as some aspects didn't feel as easy as others to understand.

Got one or two feature ideas to that I could chat about some time if you like around simplifying setup and usage. And also maybe around adding some automated tests.

springfall2008, 2023-08-23:
That would be fantastic, please do make a pull request with any readme updates.

springfall2008, 2023-08-26:
Thanks for your pull request, I've merged it now :)

iainfogg, 2023-08-26:
You're welcome!

closed 2023-09-25 by springfall2008