springfall2008/batpred #68: Octopus Intelligent ‘bump charge’

adamwylde, 2023-09-09:
Hi

I’m not sure if this is just too fine grained to be handled, and its not a major issue… but I noticed that when I initiated an Octopus Intelligent bump charge (which is at the ‘day’ import rate) that Predbat started to charge my giv battery (not just the car). I figure this is because it sees the planned_dispatches on the dispatching sensor but perhaps doesn’t differentiate between the source (bump-charge vs smart-charge) and presumes the dispatch window is at the cheap rate?

No big deal but thought I’d let you know.

Many thanks for your work @springfall2008 - its a great tool.

Adam

springfall2008, 2023-09-13:
Should be fixed in: https://github.com/springfall2008/batpred/releases/tag/v6.47

springfall2008, 2023-09-14:
@adamwylde so basically I use Predbat Slot to trigger an automation which:
a) Wakes up the Tesla
b) Unlocks the Wallbox (takes it out of pause)
c) Waits a few seconds
d) Tells my Tesla to start charging if it's not already charging

Then when the slot goes off the automation Pauses the Wallbox.

springfall2008, 2023-09-14:
alias: Car charging - Predbat slot on
description: ""
trigger:
  - platform: state
    entity_id:
      - binary_sensor.predbat_car_charging_slot
    to: "on"
    from: "off"
  - platform: time_pattern
    minutes: /5
condition:
  - condition: state
    entity_id: input_boolean.wallbox_solar_running
    state: "off"
  - condition: state
    entity_id: binary_sensor.predbat_car_charging_slot
    state: "on"
    for:
      hours: 0
      minutes: 0
      seconds: 0
  - type: is_not_charging
    condition: device
    device_id: xxxx
    entity_id: binary_sensor.tsunami_charging
    domain: binary_sensor
    for:
      hours: 0
      minutes: 5
      seconds: 0
  - type: is_plugged_in
    condition: device
    device_id: xxxx
    entity_id: binary_sensor.tsunami_charger
    domain: binary_sensor
  - condition: device
    device_id: xxxx
    domain: device_tracker
    entity_id: device_tracker.tsunami_location_tracker
    type: is_home
action:
  - device_id: xxxx
    domain: button
    entity_id: button.tsunami_wake_up
    type: press
  - if:
      - condition: device
        device_id: xxxx
        domain: lock
        entity_id: lock.wallbox_portal_locked_unlocked
        type: is_locked
    then:
      - device_id: xxxx
        domain: lock
        entity_id: lock.wallbox_portal_locked_unlocked
        type: unlock
  - delay:
      hours: 0
      minutes: 2
      seconds: 0
      milliseconds: 0
  - if:
      - condition: and
        conditions:
          - condition: device
            type: is_off
            device_id: xxxxx
            entity_id: switch.wallbox_portal_pause_resume
            domain: switch
    then:
      - type: turn_on
        device_id: xxxxx
        entity_id: switch.wallbox_portal_pause_resume
        domain: switch
      - service: input_boolean.turn_off
        data: {}
        target:
          entity_id: input_boolean.wallbox_had_override
      - device_id: xxxx
        domain: number
        entity_id: number.wallbox_portal_max_charging_current
        type: set_value
        value: 32
  - delay:
      hours: 0
      minutes: 1
      seconds: 0
      milliseconds: 0
  - if:
      - type: is_not_charging
        condition: device
        device_id: xxxx
        entity_id: binary_sensor.tsunami_charging
        domain: binary_sensor
    then:
      - type: turn_on
        device_id: xxxx
        entity_id: switch.tsunami_charger
        domain: switch
  - service: notify.notify
    data:
      message: Car charging - started as Predbat slot is on
mode: single

springfall2008, 2023-09-14:
alias: Car charging - Predbat slot off
description: ""
trigger:
  - platform: state
    entity_id:
      - binary_sensor.predbat_car_charging_slot
    to: "off"
  - platform: time_pattern
    minutes: /5
condition:
  - condition: state
    entity_id: binary_sensor.predbat_car_charging_slot
    state: "off"
  - condition: state
    entity_id: input_boolean.wallbox_solar_running
    state: "off"
  - type: is_plugged_in
    condition: device
    device_id: xxxx
    entity_id: binary_sensor.tsunami_charger
    domain: binary_sensor
  - condition: device
    device_id: xxxx
    domain: device_tracker
    entity_id: device_tracker.tsunami_location_tracker
    type: is_home
  - condition: or
    conditions:
      - condition: device
        type: is_on
        device_id: xxxx
        entity_id: switch.wallbox_portal_pause_resume
        domain: switch
      - type: is_charging
        condition: device
        device_id: xxxx
        entity_id: binary_sensor.tsunami_charging
        domain: binary_sensor
action:
  - if:
      - type: is_charging
        condition: device
        device_id: xxxx
        entity_id: binary_sensor.tsunami_charging
        domain: binary_sensor
    then:
      - type: turn_off
        device_id: xxxx
        entity_id: switch.tsunami_charger
        domain: switch
  - delay:
      hours: 0
      minutes: 0
      seconds: 5
      milliseconds: 0
  - if:
      - condition: device
        type: is_on
        device_id: xxxx
        entity_id: switch.wallbox_portal_pause_resume
        domain: switch
    then:
      - type: turn_off
        device_id: xxxx
        entity_id: switch.wallbox_portal_pause_resume
        domain: switch
  - service: notify.notify
    data:
      message: Car charging has been pause - Predbat slot is off
mode: single

closed 2023-09-15 by springfall2008