springfall2008/batpred #78: Multiple instances of batpred

Ed-M72, 2023-09-13:
Hi Trevor,

To run multiple instances of batpred do I just need to duplicate apps.yaml and in the second one change the name of the two following tags below?

pred_bat2:
  module: predbat
  class: PredBat

  prefix: predbat2

springfall2008, 2023-09-13:
I think both configs can go in the same apps.yml And yes a different prefix for the second one.I’d be interested to understand your use case

Ed-M72, 2023-09-13:
To clarify, do I also need to change pred_bat to pred_bat2?

A predbat for each phase.  And on the car charging, I assume I would use the total for the phase rather than the total for the charger?  I might need a bit of help on this as I test.

springfall2008, 2023-09-15:
Yes I think pred_bat2 is correct.

Yes for car charging I guess it's the total for the phase too.

Can you not combine the phases, e.g. just use one Predbat and multiple inverters?

springfall2008, 2023-09-25:
Closing for now, feel free to open a new ticket for anything new

closed 2023-09-25 by springfall2008