springfall2008/batpred #84: Time  zone issues?

rsarwar87, 2023-09-19:
Hi i noticed that something odd on the log:
@ TS 00:15:52. it said now 09-19 23:14:00.  Tim zone is Europe/London both on appdaemon.yml and apps.yaml. 
My first charge slot was at 23:30-00:00. tht was triggered correctly, but at 23:45, th aemon changed the start tim to 22:30-2300.
```
2023-09-20 00:15:52.565870 INFO pred_bat: Charge window will be: 2023-09-21 05:00:00+01:00 - 2023-09-21 05:30:00+01:00 - current soc 78.target 100
2023-09-20 00:15:52.567479 INFO pred_bat: Configuring charge window now (now 09-19 23:14:00 target set_window_minutes 30 charge start time 09-20 05:00:00
2023-09-20 00:15:52.569196 INFO pred_bat: Inverter 0 charge window is 05:00:00 - 05:30:00, being changed to 05:00:00 - 05:30:00
**2023-09-20 00:15:52.570634 INFO pred_bat: Not setting charging SOC as we are not within the window (now 09-19 23:14:00 target set_soc_minutes 30 charge start time 09-20 05:00:00)**
2023-09-20 00:15:52.572087 INFO pred_bat: Adjust reserve to default (as set_reserve_enable is true)
2023-09-20 00:15:52.573503 INFO pred_bat: Inverter 0 Current reserve is 4.0 already at target
2023-09-20 00:15:52.574932 INFO pred_bat: Completed run status Idle
2023-09-20 00:15:52.586617 INFO AppDaemon: Previous message repeated 1 times
2023-09-20 00:15:52.587008 WARNING AppDaemon: Excessive time spent in callback 'update_time_loop() in pred_bat', Thread 'thread.thread-1' - now complete after 82.585794 seconds (limit=60.0)
```
also plots have gone weird after midnight.
[image]

rsarwar87, 2023-09-19:
also sometimes i notice that the inverter oes not stop charging at the end of a slot, most likely if the soc target has not been met. would it be worth issuing a enableChargeTarget(false) at the end of the window just to be sure?

springfall2008, 2023-09-20:
Does the time in the left hand side of the log match the correct time, and is the inverter time aligned correctly to this too?

springfall2008, 2023-09-20:
Charts going wrong at midnight I've seen before, I must get around to debugging this but I'm always in bed!

rsarwar87, 2023-09-20:
```
2023-09-20 09:53:11.138063 INFO pred_bat: Invertor time 2023-09-20 09:52:54+01:00 AppDeamon time 2023-09-20 09:53:08.003787+01:00 difference -0.25 minutes
2023-09-20 09:53:30.200572 INFO pred_bat: Configuring charge window now (now 09-20 08:53:00 target set_window_minutes 30 charge start time 09-21 05:00:00
```
inverter times looks okay. target seems off. I can have a look in the code when i get off work - but you may be able to track it down more quickly.

rsarwar87, 2023-09-20:
[image]
charge time set correctly at 2246 of 2330, then goes into a state if "not setting charge time" (still 2330) and then finally decides to set it to 2230 @ 2350

again at 0258, setting at 0130
[image]

and 0333, setting at 0230
[image]

[image]

this is what the battery actually ended up doing.  
[image]

treforsouthwell-sifive, 2023-09-20:
I don't see anything wrong as such, other than it can't make it mind up. Was the end result okay?

rsarwar87, 2023-09-20:
> I don't see anything wrong as such, other than it can't make it mind up. Was the end result okay?

both times the charging halted when it decided to move things by 1 hour. i had to pause the appdaemon container manually set the window 
[image]

springfall2008, 2023-09-20:
Sorry posted on the wrong account.
"both times the charging halted when it decided to move things by 1 hour."

Why is this a problem as such?

springfall2008, 2023-09-20:
I'm still really struggling to understand exactly the issue, are you saying the Predbat chart shows it should charge at say 2am but it's actually charging at 3am? 

Or are you saying Predbat just changed its mind and decided to charge later?

springfall2008, 2023-09-20:
Can you snapshot the history of predbat.status?

rsarwar87, 2023-09-21:
the problem was on AppDaemon,when run as docker, despite setting tz on appdaemon.yaml, it it does not seem to want to work all the time, either because of the way the times are retrieved, or app daemon/python issue. 

Sorted by passing /etc/localtime from the rpi4

closed 2023-09-21 by rsarwar87