springfall2008/batpred #85: Odd discharge pattern with IO

adamwylde, 2023-09-20:
Hi @springfall2008 

I followed your guide and video on setting up IO having switched from Flux, and have therefore set my settings as per https://github.com/springfall2008/batpred#cheap-night-rate-with-a-good-export-rate-eg-intelligent-with-octopus-outgoing

I kinda expected that predbat would schedule export at the end of the window thus minimising the risk of running out, but it seems to be scheduling a few random tiny discharge_best drops throughout the window with the final large one ending half an hour before the low rate kicks in:

[image]

Does that seem right?

Thanks

Adam

springfall2008, 2023-09-20:
If the export slot line is above the red line then it's export freeze, where it stops the battery charging and instead exports all solar to the grid. It's actually more efficient to do it this way as you avoid charging and discharging losses from the battery. You can turn this on and off with the export freeze option in HA.

springfall2008, 2023-09-20:
BTW: You might want to check you have turned off 'combine discharge slots'

adamwylde, 2023-09-20:
Ah I see so avoiding charging only to discharge it later! Genius.

No further questions. Thanks again!

(combine discharge slots is off already)

closed 2023-09-20 by adamwylde