springfall2008/batpred #88: Missing Data

chris626w, 2023-09-20:
Originally opened on Facebook: I no longer have Load, PV or Import data appearing in the pred_bat log and also not in the Data or Power charts. This data is available from GivTCP. As requested logs are attached

[home-assistant_2023-09-20T19-42-32.646Z.log](https://github.com/springfall2008/batpred/files/12678008/home-assistant_2023-09-20T19-42-32.646Z.log)
[appdaemon.log](https://github.com/springfall2008/batpred/files/12678011/appdaemon.log)
[a0d7b954_appdaemon_2023-09-20T19-33-41.581Z.log](https://github.com/springfall2008/batpred/files/12678012/a0d7b954_appdaemon_2023-09-20T19-33-41.581Z.log)
[image]

springfall2008, 2023-09-21:
So the error in your logfile

  File "/config/appdaemon/apps/batpred/predbat.py", line 3418, in clip_discharge_slots
    soc_end = predict_soc[predict_minute]
              ~~~~~~~~~~~^^^^^^^^^^^^^^^^

I believe has been fixed in the latest release, can you update and see what you get?

springfall2008, 2023-09-21:
The fix was here: https://github.com/springfall2008/batpred/releases/tag/v7.0.1

chris626w, 2023-09-21:
Hi Trefor

I have updated Predbat to v7.0.2 and made a full HA reboot. No change to
Predbat output.
In the attached Predbat log it looks like Predbat is not/can't fetch data
from Solcast

2023-09-21 13:20:03.991444 INFO pred_bat: PV Data for pv_forecast_today
total 0 kWh
2023-09-21 13:20:03.998322 INFO pred_bat: PV Data for pv_forecast_tomorrow
total 0 kWh
2023-09-21 13:20:04.004770 INFO pred_bat: PV Data for pv_forecast_d3 total
0 kWh
2023-09-21 13:20:04.010751 INFO pred_bat: PV Data for pv_forecast_d4 total
0 kWh

I triggered Solcast with debug on and the output is available in the HA
Core log attached. I think Octopus Energy is having some issues with the
gas meter (which I don't need) but I don't see how that would
affect anything else.

Predbat has worked correctly for weeks, but from around the time of the
Socast update, plus a host of other updates around the same time, something
seems to have gone wrong. Earlier this week I reinstalled Solcast,
GivTCP and MQTT. My next step would be to reinstall Predbat but I don't
know it if deleting it from the custom repositories will give a clean
delete, or if I need to remove files as well.

Predbat is a great app and it is still controlling battery SOC, but I would
like to see all the data back!

Many thanks for your support

Chris

chris626w, 2023-09-21:
Must have been stripped out! Here they are
[home-assistant_2023-09-21T12-33-13.993Z.log](https://github.com/springfall2008/batpred/files/12687435/home-assistant_2023-09-21T12-33-13.993Z.log)
[appdaemon.log](https://github.com/springfall2008/batpred/files/12687439/appdaemon.log)
[a0d7b954_appdaemon_2023-09-21T12-20-28.570Z.log](https://github.com/springfall2008/batpred/files/12687442/a0d7b954_appdaemon_2023-09-21T12-20-28.570Z.log)
[home-assistant_2023-09-21T12-20-13.346Z.log](https://github.com/springfall2008/batpred/files/12687444/home-assistant_2023-09-21T12-20-13.346Z.log)

springfall2008, 2023-09-21:
Okay, so what do you have in apps.yaml for Solcast, it should look like this:

  # Set these to match solcast sensor names
  # The regular expression (re:) makes the solcast bit optional
  # If these don't match find your own names in Home Assistant
  pv_forecast_today: re:(sensor.(solcast_|)(pv_forecast_|)forecast_today)
  pv_forecast_tomorrow: re:(sensor.(solcast_|)(pv_forecast_|)forecast_tomorrow)
  pv_forecast_d3: re:(sensor.(solcast_|)(pv_forecast_|)forecast_(day_3|d3))
  pv_forecast_d4: re:(sensor.(solcast_|)(pv_forecast_|)forecast_(day_4|d4))

chris626w, 2023-09-21:
Sorry.......bad spelling on my part. Missed the 'e' out or forecast! I've looked at it a hundred times but not seen it. PV is now coming through. 
Still no load or power in Data chart and Power chart. My app.yaml for this is
geserial: 're:sensor.givtcp_(.+)_soc_kwh'
  geserial2: 're:sensor.givtcp2_(.+)_soc_kwh'

  #
  # Sensors, currently more than one can be specified and they will be summed up automatically
  # however if you have two inverters only set one of them as they will both read the same.
  #
  load_today: 
    - sensor.givtcp_{geserial}_load_energy_today_kwh
  import_today: 
    - sensor.givtcp_{geserial}_import_energy_today_kwh
  export_today: 
    - sensor.givtcp_{geserial}_export_energy_today_kwh

which I don't think has been changed. Data is coming through from GivTCP

springfall2008, 2023-09-21:
Do you maybe have the _2 versions of these sensors after the time GivTCP renamed everything?

e.g. can you find this one?

[image]

chris626w, 2023-09-21:
I did have multiple entities which was the reason I uninstalled GivTCP and MQTT so I could delete them. This is what I have now
[image]

chris626w, 2023-09-21:
Just found how to get to the view you posted and this is what I have, which is not the same

[image]

springfall2008, 2023-09-21:
How long ago did the update to GivTCP happen, so you actually have 7 days of days on that sensor?

chris626w, 2023-09-21:
GivTCP was reinstalled on Monday/Tuesday this week. However looking in history for Load Energy Today it only goes back to yesterday. Either way, not 7 days. So I guess wait and see?

springfall2008, 2023-09-22:
Just change 'days_previous' in apps.yaml to 1 and it should work again

chris626w, 2023-09-22:
Predbat is running again. Many thanks for a great app and top support! Beer on its way.....

springfall2008, 2023-09-22:
Great :)

closed 2023-09-22 by springfall2008