springfall2008/batpred #90: Chart Data

zubrusaurus11, 2023-09-21:
I think the data behind a few of my charts might not be correct. I know i'm an above average user for energy but surely these figures aren't correct?
Background - 16kWp in solar panels, 2 x AC Coupled Inverters and 2 x 9.5kWh batteries.
[image]
[image]
[image]
Log file - 
[a0d7b954_appdaemon_2023-09-21T16-00-18.004Z.log](https://github.com/springfall2008/batpred/files/12686810/a0d7b954_appdaemon_2023-09-21T16-00-18.004Z.log)

Any help would be very much appreciated :)

springfall2008, 2023-09-21:
your load data looks crazy, what sensor are you using and can you show it in HA?

springfall2008, 2023-09-21:
As in do you have this in apps.yaml:

  load_today: 
    - sensor.givtcp_{geserial}_load_energy_today_kwh

zubrusaurus11, 2023-09-21:
> your load data looks crazy, what sensor are you using and can you show it in HA?

Using: predbat.best_load_energy
[image]

springfall2008, 2023-09-21:
You can't set load_today in apps.yaml to point to a predbat sensor, it needs to point to the GivTCP load energy today in kWh sensor

springfall2008, 2023-09-21:
[image]
This is the correct one for me

zubrusaurus11, 2023-09-21:
> You can't set load_today in apps.yaml to point to a predbat sensor, it needs to point to the GivTCP load energy today in kWh sensor

ah okay. as I have 2 inverters, I should just choose the master inverter?

springfall2008, 2023-09-21:
Yes, the master inverter - they both see the same import/export/load anyhow.

zubrusaurus11, 2023-09-21:
> Yes, the master inverter - they both see the same import/export/load anyhow.
Gotcha. Should all entities be GivTCP?
[image]

For the imort/export entities, the GivTCP entities seem to be way off?
[image]

springfall2008, 2023-09-21:
This is the template:

  load_today: 
    - sensor.givtcp_{geserial}_load_energy_today_kwh
  import_today: 
    - sensor.givtcp_{geserial}_import_energy_today_kwh
  export_today: 
    - sensor.givtcp_{geserial}_export_energy_today_kwh

These three should be GivTCP entities

springfall2008, 2023-09-21:
That said your Import today looks corrupted, that big spike will confuse things massively.

springfall2008, 2023-09-25:
Closing this ticket, mostly as it's GivTCP related I'm afraid

closed 2023-09-25 by springfall2008