springfall2008/batpred #91: Discharge with Octopus Intelligent and Export tariff

SheriffOfNotts, 2023-09-22:
Hi Trefor,

As per FB, last night there was 69% battery remains at 21:43, with the Intelligent off-peak rate starting at 23:30.

Any idea why it wasn't discharging? Logs from today enclosed. I will update when it happens again.

[image]

[appdaemon.log](https://github.com/springfall2008/batpred/files/12698821/appdaemon.log)

springfall2008, 2023-09-22:
Looking at your log you had a few errors with some data timeouts, but they seemed to go away.

You don't have much discharge planned now as you are due to run out of battery.

If this evening you have spare battery please snapshot the battery chart and the log again and post it here

SheriffOfNotts, 2023-09-23:
Hi Trefor, same again last night - see enclosed logs and photos.. Appreciate you looking into it..

[image]
[image]
[appdaemon3.log](https://github.com/springfall2008/batpred/files/12706445/appdaemon3.log)
[appdaemon2.log](https://github.com/springfall2008/batpred/files/12706446/appdaemon2.log)
[appdaemon1.log](https://github.com/springfall2008/batpred/files/12706447/appdaemon1.log)
[appdaemon.log](https://github.com/springfall2008/batpred/files/12706448/appdaemon.log)

springfall2008, 2023-09-23:
Unfortunately the logs only show from this morning onwards.

For tonight i see a planned discharge in the log, can you post the Predbat battery chart?

Next discharge window will be: 2023-09-23 22:55:00+01:00 - 2023-09-23 23:00:00+01:00 at reserve 39.0

SheriffOfNotts, 2023-09-23:
Hi, current chart enclosed..
[image]

springfall2008, 2023-09-23:
So now it's not predicting a discharge today as the batter level expected must be below your minimum. What do you have metric_soc_min and metric_soc_keep set to?

SheriffOfNotts, 2023-09-24:
See enclosed..

[image]

springfall2008, 2023-09-24:
So your plan seemed correct, that is the keep was 2.7 and it was ending around 2.7

If you can capture the log for the hour before the charge window I can check further

SheriffOfNotts, 2023-09-25:
I had a forced discharge period last night which discharged the battery to 27%. Do you know why it wouldn't be discharging to the defined 4%?

Unfortunately the logs look to have been reset to this morning - can I check my config is correct? I have tried log_generations, but this just seems to spread the logs across multiple files rather that chronologically?

logs:
  error_log:
    filename: /config/appdaemon/logs/error.log
  main_log:
    filename: /config/appdaemon/logs/appdaemon.log
    log_size: 8192
  access_log:
    filename: /config/appdaemon/logs/access.log
  diag_log:
    filename: /config/appdaemon/logs/diag.log

treforsouthwell-sifive, 2023-09-25:
You have Best SOC Keep set to 2.7 so that would be the target discharge level, if you reduce this you should get more.

One thing I did wonder about adding as a feature is some sort of scaling of SOC keep so as it gets closer to a charge slot it scales down .

springfall2008, 2023-09-25:
You have Best SOC Keep set to 2.7 so that would be the target discharge level, if you reduce this you should get more.

One thing I did wonder about adding as a feature is some sort of scaling of SOC keep so as it gets closer to a charge slot it scales down .

SheriffOfNotts, 2023-09-25:
Thank you - I'll update it. The approach sounds spot on :)

Can I check the log configuration is correct?

springfall2008, 2023-09-25:
I just set the main_log, this contains all the Predbat stuff. Then if there is an error I check in System/Logs/AppDeamon in the HA gui to see any more details.

logs:
  main_log: 
    filename: /config/appdaemon.log

springfall2008, 2023-09-25:
Closing for now, feel free to open a new ticket if you get further issues

closed 2023-09-25 by springfall2008