springfall2008/batpred #94: Scheduling re-evaluation at the start of next window

sparks1372, 2023-09-23:
This is a suggestion rather than specifically an issue.

I have noticed that for due to the periodic scheduling of batpred, you can miss out on part of your discharge window because both the window and the force discharge need to be set. The window can be set in advance but the force discharge can only be set by the first scheduled run within the discharging window, so in the worst case you will miss out on discharging for period equal to your recalc period length.

What would be nice is if the next scheduled recalc is after the start of the discharge window, that a recalc is specifically scheduled to run at the start of the discharge window, or potentially a specific task to enable force discharge.

springfall2008, 2023-09-23:
I’m a little confused what you are trying to say here as normally predbat runs every 5 minutes aligned to 5 minute intervals. Charge and discharge slots are also aligned to 5 minute intervals.

are you referring to the recalculation run time? For me it’s around 5-10 seconds.

sparks1372, 2023-09-24:
So the schedule of predbat doesn't appear to always align with the 0, 5, 10, 15.... past the hour. Yesterday mine was running at 3, 8, 13, 18... past the hour. This means that I can't reduce the frequency of the recalc to every 30 minutes as there is a risk that it will run on 29, 59 past the hour and therefore miss most of the first 30 mins of a discharge window.

sparks1372, 2023-09-24:
[image]
The schedule today has also fallen off the 0, 5, 10, 15... past the hour and onto 13, 18 again. With the next discharge session calculated at 17:35 it will mean that the Force Discharge setting won't be activated until 17:38. Three minutes is probably not something to be too concerned over but does feel like an unfortunate effect.

sparks1372, 2023-10-16:
I am on 5 minute intervals and the calculation thread takes ~20 seconds. My point is more that I expect in many situations the output of the calculation is unlikely to change very often so re-optimising every 30 minutes/an hour may be sufficient. However, the frequency needs to be kept high to ensure that configuring the invertor is done in a timely manner given some settings cannot be scheduled, the Force Discharge being one.

springfall2008, 2023-10-16:
The last person who had this issue found it was resolved by fixing the appdeamon time zone, you can see in your logs if it reports the correct time or not

springfall2008, 2023-10-16:
But yes, I plan to separate the calculations from the control at some point soon

springfall2008, 2023-11-09:
This is now implemented and released

closed 2023-11-09 by springfall2008