springfall2008/batpred #95: Setting discharge window is resetting Force Discharge to False

sparks1372, 2023-09-23:
I have noticed that if it is currently within the discharge window, and the recalc wants to update the start or end time, sending that instruction to GivTCP appears to disable Force Discharge. This means the invertor stops exporting until the next recalc cycle at which point the Force Discharge setting is re-enabled.
This is similar to something I recall happening when I was hand rolling automations for charging/discharging so it could be the result of a GivTCP bug but if batpred checked, and updated if necessary, the Force Discharge setting after the discharge window has been set the impact would be minimised.

sparks1372, 2023-09-23:
Thinking about it, the split of the optimisation and action of setting values would also address #94

springfall2008, 2023-09-23:
I don't see this issue myself, are you using REST mode or not?

springfall2008, 2023-09-23:
e.g.here's my log of inverter mode: 
[image]

sparks1372, 2023-09-24:
No I am not using REST mode, it is using the entities from MQTT. I will provide a similar screenshot later today, my inverter also ends up in Unknown state which I believe is as a result of the disabling of Force Discharge which I mention.
You may want to obscure your inverter serial number in your screenshot

sparks1372, 2023-09-24:
As you can see the status is flicking between Timed Export and Unknown, and this appears to be related to when the end of the discharge cycle is being set, sometime successfully to 1900 and other times failing and resetting to 0700. I had similar issues with my custom automation when setting the charging/discharging windows so had to set the start and end with a 30 second gap in between.
Just to confirm, there are no automations active so the only process which is setting these values is batpred.

[image]
[image]

sparks1372, 2023-09-24:
I have moved over to using REST and will report back if that avoids these issues

springfall2008, 2023-09-24:
I don’t get the issue with rest, see if it solves it. If so we should probably report this as a bug to GivTCP

springfall2008, 2023-10-05:
Did that solve it?

sparks1372, 2023-10-16:
Yeah rest has been working fine

closed 2023-10-16 by sparks1372