springfall2008/batpred #99: Add option to chart % rather than just kWh

springfall2008, 2023-09-25:
Some users would like charts in battery % rather than kWh

gcoan, 2023-09-26:
I have (up to starting with batpred) thought of everything in % terms, which with different sized batteries is awkward.  And in terms of kW-to-empty I much prefer using the batpred _soc_kw_h0 current battery soc in kwh (although the entity name is a bit obscure)

mpartington, 2023-12-17:
Just wondering if there is an easy way for me to implement this locally outside of predbat. Realise it's it's a nice to have and wouldn't be used by all. I did take a look at template sensors, but no idea how I would convert all the attribute readings

springfall2008, 2023-12-17:
Sure in the chart code just divide by the battery size and multiply by 100

mpartington, 2023-12-17:
Thanks, hadn't thought of that!  Got this far now, is there a way to change kWh into % in the chart code, and also change resolution to 0dp?  

[image]

mpartington, 2023-12-17:
Well after a lot of googling and trial + error, I managed it! Apex charts is still a black art to me
This is the code if useful to anyone else

`type: custom:apexcharts-card
header:
  show: true
  title: Home Battery Prediction %
  show_states: true
  colorize_states: true
graph_span: 60h
span:
  start: minute
  offset: '-12h'
now:
  show: true
yaxis:
  - min: 0
    max: 100
    decimals: 0
series:
  - entity: sensor.givtcp_SERIAL_NUMBER_soc
    stroke_width: 1
    curve: stepline
    name: actual
    extend_to: now
    show:
      in_header: true
  - entity: predbat.soc_kw
    stroke_width: 1
    curve: smooth
    name: base
    unit: '%'
    float_precision: 0
    data_generator: >
      let res = []; for (const [key, value] of
      Object.entries(entity.attributes.results)) { res.push([new
      Date(key).getTime(), value*100/8.36]); } return res.sort((a, b) => {
      return a[0] - b[0]  })
    show:
      in_header: true
  - entity: predbat.soc_kw_best
    stroke_width: 3
    curve: smooth
    name: best
    unit: '%'
    float_precision: 0
    data_generator: >
      let res = []; for (const [key, value] of
      Object.entries(entity.attributes.results)) { res.push([new
      Date(key).getTime(), value*100/8.36]); } return res.sort((a, b) => {
      return a[0] - b[0]  })
    show:
      in_header: true
  - entity: predbat.soc_kw_best10
    stroke_width: 1
    curve: smooth
    name: best10
    unit: '%'
    float_precision: 0
    data_generator: >
      let res = []; for (const [key, value] of
      Object.entries(entity.attributes.results)) { res.push([new
      Date(key).getTime(), value*100/8.36]); } return res.sort((a, b) => {
      return a[0] - b[0]  })
    show:
      in_header: true
  - entity: predbat.best_charge_limit_kw
    stroke_width: 4
    curve: stepline
    name: charge_limit_best
    unit: '%'
    float_precision: 0
    type: area
    opacity: 0.2
    data_generator: >
      let res = []; for (const [key, value] of
      Object.entries(entity.attributes.results)) { res.push([new
      Date(key).getTime(), value*100/8.36]); } return res.sort((a, b) => {
      return a[0] - b[0]  })
    show:
      in_header: true
  - entity: predbat.charge_limit_kw
    stroke_width: 1
    curve: stepline
    name: charge_limit_base
    unit: '%'
    float_precision: 0
    data_generator: >
      let res = []; for (const [key, value] of
      Object.entries(entity.attributes.results)) { res.push([new
      Date(key).getTime(), value*100/8.36]); } return res.sort((a, b) => {
      return a[0] - b[0]  })
    show:
      in_header: true
  - entity: predbat.best_discharge_limit_kw
    stroke_width: 2
    curve: stepline
    name: discharge_best
    unit: '%'
    float_precision: 0
    data_generator: >
      let res = []; for (const [key, value] of
      Object.entries(entity.attributes.results)) { res.push([new
      Date(key).getTime(), value*100/8.36]); } return res.sort((a, b) => {
      return a[0] - b[0]  })
    show:
      in_header: true
  - entity: predbat.record
    stroke_width: 3
    curve: stepline
    name: record
    type: area
    opacity: 0.5
    color: black
    data_generator: >
      let res = []; for (const [key, value] of
      Object.entries(entity.attributes.results)) { res.push([new
      Date(key).getTime(), value*100/8.36]); } return res.sort((a, b) => {
      return a[0] - b[0]  })
    show:
      in_header: true
  - entity: predbat.soc_kw_base10
    stroke_width: 1
    curve: smooth
    name: base10
    unit: '%'
    float_precision: 0
    data_generator: >
      let res = []; for (const [key, value] of
      Object.entries(entity.attributes.results)) { res.push([new
      Date(key).getTime(), value*100/8.36]); } return res.sort((a, b) => {
      return a[0] - b[0]  })
    show:
      in_header: true`

[image]

springfall2008, 2024-08-06:
Closing, as Matt says can be done in the Apex chart

closed 2024-08-06 by springfall2008