springfall2008/batpred #105: Enhance predbat status with time information

gcoan, 2023-09-26:
The predbat status would be more useful if it additionally showed time information in it, eg "Idle, last run at hh:mm" or "Charging, started at hh:mm to xx:yy"
This would give a simple status confirmation that predbat is still running ok and when major activities started
[image]

springfall2008, 2023-09-26:
Have you see the display of status inside HA as it gives all the times and states? In fact I was thinking the 'at xxx' bit shouldn't be there on the messages so the same command shows in the same colour more than once.

For Charging I could add the end time maybe, e.g. Charging until 12:31?

[image]

gcoan, 2023-09-26:
Yes I've seen that, thank you.  

The problem is that in the example above, Idle was set 4 hours ago and of course predbat has been running ever since, its just the status hasn't changed.

It was where I was fiddling with the settings/apps.yaml and waiting for predbat to run, and found there wasn't an easy way of seeing when it last ran without opening the file editor, logfile, scrolling to the end.  Then repeating when it hadnt yet run.

Trade-off of putting too much information in the status that the simple history line will get very cluttered as time based things change.  Even with 'charging to' I've noticed that it charges to a level, stops charging then 5 minutes later charges a bit more so its going to get very cluttered 
[image]

springfall2008, 2023-09-26:
How about I add the last updated time to attributes of status e.g:

[image]

gcoan, 2023-09-26:
Yes that would be useful and would mean the history bar graph remained clear of clutter

springfall2008, 2023-09-27:
Now added

closed 2023-09-27 by springfall2008