springfall2008/batpred #119: Clarify Octopus Intelligent Plugin requirement in README

iainfogg, 2023-09-28:
Hi @springfall2008  - I think it's worth doing some rewording in the README around the use of the Octopus Intelligent Plugin, as it's left me a bit unsure of whether it's needed or not.

This text appears in the Energy Rates section:

> The Intelligent Plugin is no longer required as the Octopus Energy Plugin now gives this information, however if it is installed it will be used.

But the next section, Car Charge Planning, says that for IO-led planning, you should:

> Enable Octopus Intelligent plugin - in which case Predbat will use the slots allocated by Intelligent in battery prediction 

Then the point after says:

> Ensure octopus_intelligent_slot points to the Intelligent Slot sensor in the Octopus Plugin or the Octopus Intelligent Plugin

2 out of the 3 suggest that the Intelligent plugin isn't needed, but the other one doesn't - hopefully you can clarify which one's right.

Again, happy to fix if you like, just not sure what the state of play is.

springfall2008, 2023-09-29:
I missed one of the updates, the newer versions of the Octopus Plugin does everything so the Intelligent plug is no longer needed. If you can fix that's great :)

springfall2008, 2023-10-05:
I think this is now done

closed 2023-10-05 by springfall2008