springfall2008/batpred #125: Merge notifications from multiple inverter activity together

gcoan, 2023-09-29:
When predbat is managing multiple inverters, each notification is sent separately for each inverter.

If possible can these be combined, e.g. “Inverter 0 and Inverter 1 Target Reserve changed to 46”

It would also cut down on notifications if there was some “buffer window” on the change as to whether a new notification was sent or not.  E.g. notifications are currently set when reserve changed to 44 then 45 then 46

[image]

springfall2008, 2023-09-29:
Good point, I wonder if it can just say Inverter N as it does the same thing to all of them anyhow

gcoan, 2023-09-29:
I don't have any particular preference. If the same activity is applied for all inverters then it doesn't make any benefit to notify per inverter

gcoan, 2024-04-29:
changes to notifications have delivered this

closed 2024-04-29 by gcoan