springfall2008/batpred #130: Using predbat with battery only setup?

pootle, 2023-10-01:
I have a battery only setup with heat pump heating, This looks really good, but just wondering if it will cope with a battery only setup.

Could I run without solcast? clearly much of that is irrelevant, but picking up some forecast info to give an idea of additional power needed in cold weather would be good.

springfall2008, 2023-10-01:
You can certainly run Predbat without Solcast, just comment out those config lines.

There isn't anything to forecast heat pump usage, it would be based on historical usage. If you have a way to forecast heat pumps it might be something that could be added but I wouldn't know what formula to use.

springfall2008, 2023-10-08:
Closing for now, feel free to open new questions

closed 2023-10-08 by springfall2008