springfall2008/batpred #143: Discharge planned finishing 30 minutes too late (possibly)

iainfogg, 2023-10-05:
Hi @springfall2008 - we chatted on a Facebook issue the other day about my discharge that takes place immediately before the Intelligent cheap rate slot was finishing 30 minutes later than I thought it should.

It's on this post: https://www.facebook.com/groups/615579009972782/permalink/843257357204945/

We discussed in the issue whether there was something in the code causing it to finish later than started, as I said I'd expect / prefer it to finish discharging by 23:30 to then go into charge mode through the whole charge period.

Tonight I've plugged in my car, and Intelligent is planning to start charging it from 22:30 tonight, an hour earlier than usual.

And Predbat has come up with its own discharge then charge plan as usual, but tonight the discharge is finishing at 23:00 rather than the 00:00 it's been doing, in other words it's still finishing discharging 30 mins later than when the cheap period starts.

My guess is there's an issue with the code to do with array indexes or a rounding issue or something which is meaning that the discharge is always running for one more period than it should do. I don't yet understand the code enough to point to a specific issue, I'm just making guesses based on the behaviour of stopping discharging and starting charging 30 mins (one period) later than when the cheap period starts.

If you've got any idea why that'd be great :)

Thanks, Iain

iainfogg, 2023-10-06:
PS just to add, a side effect of this is that for the first 30 mins of charging my car, the battery discharge is actually going to my car rather than to the grid at my normal export rate, so there is an impact on the value of those 30 mins.

Here's an example of the two overlapping on an Intelligent slot during the day:

2023-10-06 10:05:18.798529 INFO pred_bat: Charge window will be: 2023-10-06 11:00:00+01:00 - 2023-10-06 12:00:00+01:00 - current soc 91.target 100
2023-10-06 10:05:18.801688 INFO pred_bat: Not setting charging window yet as not within the window (now 10-06 10:05:00 target set_window_minutes 30 charge start time 10-06 11:00:00
2023-10-06 10:05:18.804764 INFO pred_bat: Next discharge window will be: 2023-10-06 10:30:00+01:00 - 2023-10-06 11:00:00+01:00 at reserve 79.0
2023-10-06 10:05:18.808090 INFO pred_bat: Inverter 0 Adjust force discharge to False, change times from 10:30:00 - 11:00:00 to 10:30:00 - 11:00:00
2023-10-06 10:05:18.811200 INFO pred_bat: Car charging from battery is off, next slot for car 0 is 10-06 10:30:00 - 10-06 12:00:00

treforsouthwell-sifive, 2023-10-06:
I did add a toggle to try to prevent this

switch.predbat_calculate_discharge_oncharge

In theory if you turn this off then it won't discharge during this time, can you give it a try?

iainfogg, 2023-10-06:
Thanks, I've upgraded and I'm trying that out now, I'll let you know how it goes.

However, I do still wonder if there's a logic error, I don't understand why the discharge always runs precisely 30 mins into the cheap slot, it feels like there's something not right about that.

springfall2008, 2023-10-06:
>However, I do still wonder if there's a logic error, I don't understand why the discharge always runs precisely 30 mins into the cheap slot, it feels like there's something not right about that.

My gut feeling is because when you run out of battery in the cheap period it doesn't cost extra to start importing. If you run flat before that time then you have the import issue. It maybe it just works out a little better this way, assuming Predbat has no knowledge of your car charging.

I think there maybe a bug with car charging from battery when it's 'off' the simulation might ignore the forced discharge, I'm going to look at that.

springfall2008, 2023-10-06:
I don't obviously see the issue in the code, does the export predicted model the car charging against this discharge correctly?

iainfogg, 2023-10-12:
> My gut feeling is because when you run out of battery in the cheap period it doesn't cost extra to start importing. If you run flat before that time then you have the import issue. It maybe it just works out a little better this way, assuming Predbat has no knowledge of your car charging.

If it's doing an export, I'd expect it to do it in the last possible slot, i.e. running up to 23:30. Say it decided to do 22:00-23:30 then during that time I started using more, it would just cancel the plan part way through. Also, it does have the minimum keep value as a safety buffer.

That doesn't feel like a reason why it would calculate to run exactly 30 minutes past 23:30, but I am just guessing as I still don't properly understand the code.

I've been running for a week or so with that flag to avoid discharging in cheap slots, and it's doing exactly what I'd expect, so I'll close this off. Thanks for adding that switch to control this.

closed 2023-10-12 by iainfogg