springfall2008/batpred #146: Freeze discharge doesn’t completely freeze

adamwylde, 2023-10-06:
More of an observation, and I think this must be a Giv problem not Predbat, but I’m finding when in freeze discharge mode that up to 300w still goes into the battery rather than export? 

[image]

here you can see the Predbat status is freeze discharge, and that the inverter is in eco and has been set to 0w charge rate. The gauge shows 295w going into battery and the rest to export. You can also see that although it’s been in this state since 7am the battery Soc has been climbing all day. (Graph is today and tomorrow).

springfall2008, 2023-10-06:
Yes, I believe it's a bug in the current battery firmware. I suppose I could model it if it makes a significant difference?

adamwylde, 2023-10-06:
No don’t worry. At least I’ve not got something configured incorrectly! Thanks

closed 2023-10-06 by adamwylde