springfall2008/batpred #172: 7.8.2 combined slots different results to previously

adamwylde, 2023-10-12:
Hi @springfall2008 

I’ve updated (and reverted) and as of 7.8.2 I’m not getting the correct discharge calculated. Tariff is IO so the discharge/charge benefit is fairly clear - usually predbat dumps whatever is left leading up to the overnight charge but now its not (for today or tomorrow):

After update:
[image]

After revert to previous ver:
[image]

Thanks for all your efforts. It’s a remarkable tool.

adamwylde, 2023-10-12:
PS - it runs bloody fast now though!

springfall2008, 2023-10-12:
You probably need to try putting the max windows back up to 96, but I’ll take a proper look why tomorrow

adamwylde, 2023-10-12:
Thanks - wasn’t worried as reverting is easy, but tried the 96 and yes indeed that reverts the behaviour, and results in the following - 53 windows I guess.

2023-10-12 21:49:19.299615 INFO pred_bat: Calculate Best options: calculate_best_charge(True) calculate_best_discharge(True) calculate_discharge_first(True) calculate_discharge_oncharge(True) set_discharge_freeze(True) set_discharge_freeze_only(False) set_discharge_during_charge(True) combine_charge_slots(True) combine_discharge_slots(False) best_soc_min(0.0 kWh) best_soc_max(0.0 kWh) best_soc_keep(0.0 kWh) inverter_loss(5 %) battery_loss(5 %) battery_loss_discharge (5 %) inverter_hybrid(True) metric_min_improvement(0.0 p) metric_min_improvement_discharge(0.0 p) metric_battery_cycle(0.0 p/kWh)
2023-10-12 21:49:19.307128 INFO pred_bat: Optimise all windows, total charge 1 discharge 52

adamwylde, 2023-10-13:
Looking better but it is not right in 7.8.4 - the discharges are scheduled earlier than expected.

[image]

springfall2008, 2023-10-14:
I did also put in a fix for this issue, so the latest 7.8.5 should be better. That said less windows is likely to give a slightly worse plan if there are less options available

adamwylde, 2023-10-15:
Still creates gappy discharge blocks and doesn’t butt up to the charge window. As you say, it has less options available - so I’ve just kept it at 96 windows.

springfall2008, 2023-10-15:
Okay thanks. Resolving as it's kind of as expected.

closed 2023-10-15 by springfall2008