springfall2008/batpred #173: Charge slots incorrect since upgrade to 7.8.2

AGLAISTER, 2023-10-12:
SInce upgrade to 7.8.2 charging at higher price times and not occurring at times where the battery is lower and the price is much lower than the actual set charge times.
It is also not charging the battery anywhere near as high as pre 7.8.1 leaving me at risk of running out of battery during peak rate times.
[appdaemon.log.1.txt](https://github.com/springfall2008/batpred/files/12886967/appdaemon.log.1.txt)
[appdaemon.log.2.txt](https://github.com/springfall2008/batpred/files/12886969/appdaemon.log.2.txt)
[appdaemon.log.3.txt](https://github.com/springfall2008/batpred/files/12886970/appdaemon.log.3.txt)
[image]
[image]
[image]

springfall2008, 2023-10-12:
It maybe worth putting metric_max_windows back to 96 and see if you get similar results to before 

Also check calculate_discharge_incharge and make sure it’s on

AGLAISTER, 2023-10-12:
where is metric_max_windows???? I can't find it either trying to add it to a card or in the apps.yaml

also despite restarting home assistant and appdaemon I cannot turn calculate_discharg_incharge on, I click it on but the switch turns off on its own.

springfall2008, 2023-10-13:
Sorry I had the name wrong: input_number.predbat_calculate_max_windows

Sometimes when you change a toggle you have to wait a few minutes, it might toggle back in the meantime before updating again.

springfall2008, 2023-10-28:
Did this help?

AGLAISTER, 2023-10-28:
Hi since I've dropped the windows back to 32 things seem to be working ok
again

springfall2008, 2023-11-13:
Closing this for now, I think things have moved on.

closed 2023-11-13 by springfall2008