springfall2008/batpred #177: Excessive time in thread

AGLAISTER, 2023-10-13:
Following 7.8.1 update and changing the charge windows to > 32 i experience issues including
Slow/no response
Unable to change settings
Slow change of settings
I eventually reduced the windows to 32 but there are errors in the logs showing excessive time in thread

Logs uploaded
[appdaemon.log 13_10_23.txt](https://github.com/springfall2008/batpred/files/12898422/appdaemon.log.13_10_23.txt)
[appdaemon
[appdaemon.log.2 13_10_23.txt](https://github.com/springfall2008/batpred/files/12898425/appdaemon.log.2.13_10_23.txt)
.log.1 13_10_23.txt](https://github.com/springfall2008/batpred/files/12898424/appdaemon.log.1.13_10_23.txt)
[appdaemon.log.3 13_10_23.txt](https://github.com/springfall2008/batpred/files/12898427/appdaemon.log.3.13_10_23.txt)

springfall2008, 2023-10-13:
Can I suggest turning off debug (debug enable) and also increasing best_soc_step to 0.5 which should double the speed

slopemad, 2023-10-14:
Just having a think about runtimes.  My HA environment is an Odroid N2+, so pretty low spec.  The last pass of Predbat for me took 99 seconds.

`2023-10-14 10:11:02.025544 WARNING AppDaemon: Excessive time spent in callback 'run_time_loop() in pred_bat', Thread 'thread.thread-1' - now complete after 99.024748 seconds (limit=10)`

The thread was started at 09:09:23

`2023-10-14 10:09:23.476869 INFO pred_bat: --------------- PredBat - update at 2023-10-14 09:09:23.475374+01:00 with clock skew 0 minutes`

and finished at 10:11:02.

But it seems, from the logs, that it's the changing of settings in GivTCP that takes the time.  This bit started at 10:10:21.

```
2023-10-14 10:10:21.912255 INFO pred_bat: Combine window with next window 10-14 09:30:00-10-14 10:00:00
2023-10-14 10:10:21.917394 INFO pred_bat: Charge window will be: 2023-10-14 09:00:00+01:00 - 2023-10-14 10:00:00+01:00 - current soc 100.target 100
2023-10-14 10:10:40.077268 INFO pred_bat: Set inverter 0 charge schedule False via REST successful on retry 0
2023-10-14 10:10:40.164809 INFO pred_bat: Inverter 0 Turning off scheduled charge
2023-10-14 10:10:40.170626 INFO pred_bat: Holding current charge level using reserve: 100
2023-10-14 10:10:40.176542 INFO pred_bat: Next discharge window will be: 2023-10-15 13:45:00+01:00 - 2023-10-15 14:01:00+01:00 at reserve 95.0
2023-10-14 10:10:40.182705 INFO pred_bat: Setting ECO mode as we are not yet within the discharge window - next time is 10-15 13:45:00 - 10-15 14:00:00
2023-10-14 10:10:40.189172 INFO pred_bat: Inverter 0 Current SOC is 100.0 already at target
2023-10-14 10:10:40.197002 INFO pred_bat: Adjust reserve to target charge % (set_reserve_enable is true)
2023-10-14 10:10:40.203690 INFO pred_bat: Inverter 0 Current Reserve is 4.0 % and new target is 100 %
2023-10-14 10:11:01.957967 INFO pred_bat: Set inverter 0 reserve 100 via REST successful on retry 0
2023-10-14 10:11:02.008519 INFO pred_bat: Completed run status Hold charging
```

Looking at a specific bit, for example, this bit:

```
2023-10-14 10:10:40.203690 INFO pred_bat: Inverter 0 Current Reserve is 4.0 % and new target is 100 %
2023-10-14 10:11:01.957967 INFO pred_bat: Set inverter 0 reserve 100 via REST successful on retry 0
```

was 21 seconds from initiating the settings change

But the GivTCP logs suggest the whole transaction was done and dusted in 4 seconds.

```
2023-10-14 09:10:40,208 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 100
2023-10-14 09:10:44,965 - Inv1 - write       -  [INFO    ] - Setting shallow charge 100 was a success
```

So why the lag between 'setting shallow charge 100 was a success' at 09:10:44 and the 'successful on retry 0' message in AppDaemon a good 14 seconds later?

(There is a 1 hour difference between AppDaemon and GivTCP logs).

I have tried best_soc_step at 0.5 rather than 0.25, but it makes no notable difference, mainly because the lag seems to be in changing inverter settings rather than making calculations.  The more settings to change, the longer this takes..

But the issue here is that the thread seems to start just before the 5 minutes mark, and this means that at the beginning of a window (say the car charge window, or a change from charge to discharge to idle), this may not actually take effect until 6 or 7 minutes into the window (because the thread starting at 08:59:43, for example, relates to the 08:30 to 09:00 slot, not the 09:00 to 09:30 slot), so the changes being made in PB for the 09:00 to 09:30 slot don't actually even start until, for example, 09:04:43 and the changes start being pushed a short while after that.

So two issues here...  Some kind of lag between PB sending a message to givTCP and the response from givTCP, and the threads starting at 5 minute intervals since the previous thread start, rather than at xx:00:00 and xx:05:00, etc.  I did note this in the logs from a recent AppDaemon restart:

```
2023-10-14 09:59:23.467478 INFO pred_bat: Predbat: Next run time will be 2023-10-14 09:00:00 and then every 300 seconds
2023-10-14 09:59:23.471515 INFO pred_bat: Balance inverters will run every 60 seconds (if enabled)
2023-10-14 09:59:23.474111 INFO AppDaemon: App initialization complete
2023-10-14 09:59:38.477100 INFO pred_bat: --------------- PredBat - update at 2023-10-14 08:59:38*474953+01:00 with clock skew 0 minutes
```

i.e. the Appdaemon log says Predbat will run at 09:00:00, but it actually starts 37 seconds earlier (again, weird timezone in the log timestamp), introducing a massive lag between the start and end of an event being pushed into the inverter.

slopemad, 2023-10-14:
The timestamp in the log literally only impacts the log itself, no practical impact on the application.  The issue I'm seeing is more about when the thread actually starts (so rather than at xx:00:00 and xx:05:00, xx:10:00, etc, it starts when AppDaemon starts, and then every five minutes after that.  

I can workaround by stopping and restarting AppDaemon at a more appropriate appropriate time so the thread starts slightly after at the correct time, that feels more like a fudge (but I don't know if it is even possible to run on a given schedule, rather than simply 300 seconds since the last run?)

run_loop in GivTCP was 10 seconds, I changed it to 5, which made no difference to the lag.

I'm not sure what the bottleneck in my system is.  Whether replacing the N2+ with an M1 would make a notable difference, whether the issue is CPU, memory or or disk I/O.

springfall2008, 2023-10-14:
The appdeamon time stamp does make a difference as the immediate actions such as discharge will not happen at the right time unless this is fixed.

the GivTCP run loop time should really be at least 15 seconds otherwise you will overload the mod bus and make things run slower. Remember to restart GivTCP when you update it.

I don’t understand what the thread doesn’t run at the right time. It will run once on startup or when you change a setting but afterwards it should be exactly aligned to the 5 minute mark. Maybe fixing the timezone could fix this

slopemad, 2023-10-14:
So timezone in batpred/config/apps.yaml is
  # Timezone to work in
  timezone: Europe/London

appdaemon.yaml was set to time_zone: Europe/Amsterdam (the default?), set this to Europe/London...

Now the logs look more sane from a timestamp point of view:
2023-10-14 11:46:57.632598 INFO AppDaemon: AppDaemon is stopped.
2023-10-14 10:47:09.053211 INFO AppDaemon: AppDaemon Version 4.4.2 starting

Lo and behold, the thread now starts correctly!

2023-10-14 10:49:04.454834 WARNING AppDaemon: Excessive time spent in callback 'update_time_loop() in pred_bat', Thread 'thread.thread-1' - now complete after 96.454113 seconds (limit=10)
date
2023-10-14 10:50:00.008071 INFO pred_bat: --------------- PredBat - update at 2023-10-14 10:50:00.005336+01:00 with clock skew 0 minutes

GivTCP run_loop set to 15

Still a ~20 second lag in sending a command to GivTCP REST and moving on, possibly improved by interacting directly with MQTT, but that's almost certainly a huge piece of work and might just fix a corner case rather than a general performance issue.

Thank you for all your ongoing help with this.

```
2023-10-14 11:05:59.852051 INFO pred_bat: Charge window will be: 2023-10-14 11:30:00+01:00 - 2023-10-14 16:00:00+01:00 - current soc 100.target 100
2023-10-14 11:05:59.857237 INFO pred_bat: Configuring charge window now (now 10-14 11:05:00 target set_window_minutes 30 charge start time 10-14 11:30:00
2023-10-14 11:05:59.862649 INFO pred_bat: Inverter 0 charge window is 10:30:00 - 16:00:00, being changed to 11:30:00 - 16:00:00
2023-10-14 11:06:16.817780 INFO pred_bat: Inverter 0 set charge slot 1 {'start': '11:30', 'finish': '16:00'} via REST successful after retry 0
2023-10-14 11:06:16.896041 INFO pred_bat: Inverter 0 Updated start and end charge window to 11:30:00 - 16:00:00 (old 10:30:00 - 16:00:00)
2023-10-14 11:06:33.773054 INFO pred_bat: Set inverter 0 charge schedule True via REST successful on retry 0
2023-10-14 11:06:33.833597 INFO pred_bat: Inverter 0 Turning on scheduled charge
2023-10-14 11:06:33.839260 INFO pred_bat: Setting ECO mode as no discharge window planned
2023-10-14 11:06:54.746126 INFO pred_bat: Set inverter 0 mode Eco via REST successful on retry 0
2023-10-14 11:06:54.867674 INFO pred_bat: Inverter 0 set force discharge to False
2023-10-14 11:06:54.884252 INFO pred_bat: Inverter 0 Current SOC is 100.0 already at target
2023-10-14 11:06:54.912331 INFO pred_bat: Adjust reserve to default (as set_reserve_enable is true)
2023-10-14 11:06:54.932453 INFO pred_bat: Inverter 0 Current reserve is 4.0 already at target
2023-10-14 11:06:54.946873 INFO pred_bat: Completed run status Idle with Errors reported (check log)
2023-10-14 11:06:54.951980 WARNING AppDaemon: Excessive time spent in callback 'run_time_loop() in pred_bat', Thread 'thread.thread-1' - now complete after 114.950365 seconds (limit=10)
```

springfall2008, 2023-10-14:
That’s good news

so the other thing is it says finished with errors, what is the error (check Predbat.status)?

slopemad, 2023-10-14:
The Fordpass isn't working since this morning and so not reporting a car SOC into the appropriate sensor, I think that's what's causing that.

2023-10-14 11:35:01.575371 INFO pred_bat: WARN: Return bad float value unknown from car_charging_soc using default 0.0

springfall2008, 2023-10-14:
Ah okay, no worries then.

AGLAISTER, 2023-10-14:
I'm still getting the warning in my logs, i've changed soc step to 0.5 and second pass is turned off, attaching fresh logs 

WARNING AppDaemon: Excessive time spent in callback 'run_time_loop() in pred_bat', Thread 'thread.thread-1' - now complete after 164.10826 seconds (limit=10)
[appdaemon.log.1 (2).txt](https://github.com/springfall2008/batpred/files/12905369/appdaemon.log.1.2.txt)
[appdaemon.log.txt](https://github.com/springfall2008/batpred/files/12905370/appdaemon.log.txt)
[appdaemon.log.3 (2).txt](https://github.com/springfall2008/batpred/files/12905371/appdaemon.log.3.2.txt)
[appdaemon.log.2 (2).txt](https://github.com/springfall2008/batpred/files/12905372/appdaemon.log.2.2.txt)

springfall2008, 2023-10-14:
The warning you can ignore, it's just your machine is slower at it takes longer than the 10 second default limit. As it's under 5 minutes it's okay.

That said, on 'main' I put a version where I removed some 10 second waits that I don't think are required for REST mode, you can give it a try if you like.

springfall2008, 2023-10-28:
Did the new version help any?

gcoan, 2023-10-28:
I've recently (mid week) changed from Octopus Flux to Octopus Agile for my import (although still seem to be getting Flux Outgoing :-) )
batpred works remarkably well, very impressed with the way it calculates all the slots and best schedule (although I am getting overwhelmed by the dozens of alerts I get now #125 and may need to turn off some alerts.

But since moving to agile I too am seeing timeouts in my logs.  I understand that predbat is got more to calculate, but whole runs are now taking an average of 2-3 minutes and I think its sometimes even longer as I get alerts saying 'predbat changed xxx at 16:00' but the alert comes through at 16:07.

I thought it might be related to having to process more slots around 3-4pm, but I don't at the moment have any logs from that far back.
[appdaemon (2).log](https://github.com/springfall2008/batpred/files/13195554/appdaemon.2.log)

Turned off second pass and changed calculate_max_windows down from 96 to 32 (had turned to 96 per advice you'd given on #173), but still see the timeouts and occasional delayed alerts.
Running 7.9.1, not the patch on main.   I'll be putting 7.9.2 in now

AGLAISTER, 2023-10-28:
Hi Trefor,

I've just checked the logs - still seeing the warning with times
anywhere from 37s up to 97s

However generally predbat is behaving OK.

Just upped my SOC max to 0 and set load scaling to 2 as I wasn't getting
enough charge to my liking when i've got some low rates.
The planned charge is much lower than recent days but i'm guessing that is
because tomorrows solcast looks high nearly up to 3kw

gcoan, 2023-10-28:
Running with the latest 7.10.1 beta which has the performance improvements and calculate_plan_every set to 10, I'm still seeing timeouts and long run times
[appdaemon (3).log](https://github.com/springfall2008/batpred/files/13196584/appdaemon.3.log)

springfall2008, 2023-10-28:
This won’t solve the longer runtimes, but what do you mean by timeouts? The idea of this change is you don’t recalculate so often and the inverter controls are done first so they aren’t late

gcoan, 2023-10-28:
I thought some of this fix was to resolve the excessive time errors being seen, vis:
```
2023-10-28 20:48:18.999722 WARNING AppDaemon: Excessive time spent in callback 'update_time_loop() in pred_bat', Thread 'thread.thread-2' - now complete after 162.998876 seconds (limit=120.0)
```

springfall2008, 2023-10-28:
For the specific wanting you can just increase the limit, although it’s harmless. The fix was to sort out the side effects of things taking too long

gcoan, 2023-10-28:
OK, I found the thread_duration_warning_threshold set to 120 already in appdaemon.yaml.  2 minutes is quite a while.
I could set it higher but the last warning I got was that the loop tool 241 seconds so its almost getting to the point where its going to be running continuously not every 5 minutes.

I also see a different warning that's occurring after about 2.5 seconds in the cost evaluation, this doesn't refer to the 120 second limit:
```yaml
2023-10-28 22:30:26.319795 WARNING AppDaemon: Excessive time spent in utility loop: 2741.0ms, 2741.0ms in check_app_updates(), 0.0ms in other
2023-10-28 22:30:31.842250 INFO pred_bat: Best charge windows in price group 26 best_metric 216.16 best_cost 88.27 metric_keep 899.11 windows [ 30-10 14:00:00 - 30-10 15:00:00 @ 23.81p 95.0%, 30-10 13:30:00 - 30-10 14:00:00 @ 23.59p 80.0%, 29-10 14:00:00 - 29-10 15:00:00 @ 23.81p 100.0%, 29-10 13:30:00 - 29-10 14:00:00 @ 23.59p 97.0% ]
2023-10-28 22:30:34.144518 WARNING AppDaemon: Excessive time spent in utility loop: 2825.0ms, 2824.0ms in check_app_updates(), 1.0ms in other
2023-10-28 22:30:39.049793 INFO pred_bat: Best charge windows in price group 25 best_metric 216.16 best_cost 88.27 metric_keep 899.11 windows [  ]
2023-10-28 22:30:39.052568 INFO pred_bat: Best discharge windows in price group 25 best_metric 216.16 best_cost 88.27 metric_keep 899.11 windows [  ]
2023-10-28 22:30:40.095867 WARNING AppDaemon: Excessive time spent in utility loop: 2855.0ms, 2854.0ms in check_app_updates(), 1.0ms in other
2023-10-28 22:30:41.317686 INFO pred_bat: Best charge windows in price group 24 best_metric 216.16 best_cost 88.27 metric_keep 899.11 windows [  ]
2023-10-28 22:30:44.771210 INFO pred_bat: Best charge windows in price group 23 best_metric 216.16 best_cost 88.27 metric_keep 899.11 windows [  ]
2023-10-28 22:30:47.570731 INFO pred_bat: Best charge windows in price group 22 best_metric 207.93 best_cost 84.7 metric_keep 899.11 windows [ 30-10 13:00:00 - 30-10 13:30:00 @ 19.4p 100.0%, 29-10 13:00:00 - 29-10 13:30:00 @ 19.4p 100.0% ]
````

Is this OK, supressable or an issue?

gcoan, 2023-10-29:
Hi Trefor, minor formatting bug in the log file, can you round the minutes logged?

``
Will recompute the plan as it is now 9.983333333333333 minutes old and will exceed the max age of 10 before the next run
``

closed 2023-12-29 by springfall2008