springfall2008/batpred #180: Suggestion: Recode 'set reserve enable' to toggle eco mode on/off to accommodate new inverter firmware

mpartington, 2023-10-14:
Unfortunately the new faster firmware that is being/will be rolled out means that the previous method of setting the battery reserve to the soc target no longer results in zero charge / discharge. Infant rather odd things are happening.

The new firmware still has a small charge even when paused, which appears to be driving fluctuations. The only way I have identified to truly pause the battery is by turning off eco mode. Therefore if possible, could this method be used instead? It should be compatible with all inverter types.

Example of the issue (set reserve enable is on). Soc target is 70%

[image]

[image]

Understand if not possible and this should really be fixed in firmware, which I will continue to push for

mpartington, 2023-10-14:
This is an example from this morning, where set reserve enable was switched off. The fluctuations were still present, but less frequent
[image]
[image]

springfall2008, 2023-10-14:
Are you suggesting to use Eco (Paused) mode?

[image]

mpartington, 2023-10-14:
I normally select from the portal,.which completely turns off charge /discharge and sets the mode to Eco (Paused). Bizarrely, doing it the other way around (selecting in GivTCP, doesn't seem to have the same effect. There is still a 200W charge.

I'll check Willy's balancing scripts, as sure he uses this method
[image]

gcoan, 2023-10-14:
Before I started using predbat, if I wanted to hold the battery level in my own HA scripts I just set discharge or charge rate to zero.  I never used target soc.
Would this work?

mpartington, 2023-10-14:
If you're on the 1st gen original.firmware, that works perfectly. On the new faster firmware, it doesn't fully shut off (for reasons still unexplained by Giv).  We then got around this by setting the reserve to the soc target.

However the firmware has evolved again breaking that work around, and now they have added a battery pause register, which ironically still doesn't fully pause the battery.

I'm on the AC3 beta firmware,.which is.meant to be the final version. I keep asking for a true pause to be coded in firmware, but not happened yet :-(. Turning off eco is.about the only way to do 8t, but it looks like the portal does it differently to GivTCP

mpartington, 2023-10-14:
Got it, it's 
switch.givtcp_xxxx_battery_power_mode
Turn it off and battery power drops to zero

gcoan, 2023-10-14:
Thanks for the explanation. Crazy that setting the battery still charges and discharges when the rate is set to zero.  Definitely sounds like a GivEnergy bug/feature.

Yes I’m on 1st gen original firmware so it works for me.

springfall2008, 2023-10-14:
Matt that's odd, as reading the code setting battery power mode to 0 is forced export mode - maybe it's 0 outside the window and force export inside?

mpartington, 2023-10-14:
Not sure, but this is what it currently does. I've added the toggle to my dashboard, so you can see the mode changes 

[image]
[image]
[image]

reidjr, 2023-10-15:
> Matt that's odd, as reading the code setting battery power mode to 0 is forced export mode - maybe it's 0 outside the window and force export inside?

Its not quite that:
You need to look at the pair.
Battery mode 0/1 ECO off or On outside the window
Enable discharge 0/1 Discharge fully in window.

So you have 4 options:
Battery Power Mode, Enable Discharge
1,0 ECO
1,1 Timed Discharge
0,1 Timed Export
0,0 Idle ( no specific mode in the Apps )

Trouble is Gen1 and gen2, or at least firmware to firmware behaviour to those settings differ.

mpartington, 2023-10-15:
Richards given some good insight into the logic (with an assumption GivTCP switches 2 registers- as suggested above).

It might be easier for me to knock up a.local.automation, but I'm not sure if predbat would see the mode changes and change them back.  I'll give it a go tonight

[image]

[image]

mpartington, 2023-10-15:
This will differ system to system, but this is the residual charge amounts for the different methods (AC3 new firmware)

Pause charge: 28W
Active power set to zero : 86W
Battery charge rate set to zero: 316W
Pause during charging (firmware logic) : approx 90W
Eco mode off: zero

mpartington, 2023-10-16:
Possibly a semi success. I ran a simple automation the turned off battery mode when soc= soc target and back on whenever soc target changed in the charge period. That bit worked perfectly, however the battery still had a residual charge, unlike my earlier experiment.

I need to understand this, so will disable predbat tonight and set a soc target manually, to rule.in or out predbat changing something back, or if it just works differently in a charge window. The charts do have a few very small areas of zero charge. Will also see if anything in the logs.

[image]

[image]

closed 2023-10-17 by mpartington