springfall2008/batpred #181: Solar Edge Integration

Londonlad95, 2023-10-14:
Hi Trefor, 

As discussed, here are the list of entities etc of the solar edge mod bus multi-integration.

`<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/benjt/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/benjt/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Name | Full Name
-- | --
Solaredge B1 Maximum Energy | sensor.solaredge_b1_maximum_energy
Solaredge B1 State of Energy | sensor.solaredge_b1_state_of_energy
Solaredge B1 State of Health | sensor.solaredge_b1_state_of_health
Solaredge B1 Status | sensor.solaredge_b1_status
Solaredge B1 Version | sensor.solaredge_b1_version
Solaredge I2 AC Charge Limit | number.solaredge_i2_ac_charge_limit
Solaredge I2 AC Charge Policy | select.solaredge_i2_ac_charge_policy
Solaredge I2 AC Current | sensor.solaredge_i2_ac_current
Solaredge I2 AC Energy kWh | sensor.solaredge_i2_ac_energy_kwh
Solaredge I2 AC Frequency | sensor.solaredge_i2_ac_frequency
Solaredge I2 AC Power | sensor.solaredge_i2_ac_power
Solaredge I2 AC Voltage AB | sensor.solaredge_i2_ac_voltage_ab
Solaredge I2 Active Power Limit | sensor.solaredge_i2_active_power_limit
Solaredge I2 Active Power Limit | number.solaredge_i2_active_power_limit
Solaredge I2 Advanced Power Control | binary_sensor.solaredge_i2_advanced_power_control
Solaredge I2 Backup Reserve | number.solaredge_i2_backup_reserve
Solaredge I2 DC Current | sensor.solaredge_i2_dc_current
Solaredge I2 DC Power | sensor.solaredge_i2_dc_power
Solaredge I2 DC Voltage | sensor.solaredge_i2_dc_voltage
Solaredge I2 Device | sensor.solaredge_i2_device
Solaredge I2 Grid Control | switch.solaredge_i2_grid_control
Solaredge I2 Refresh | button.solaredge_i2_refresh
Solaredge I2 RRCR Status | sensor.solaredge_i2_rrcr_status
Solaredge I2 Status | sensor.solaredge_i2_status
Solaredge I2 Status Vendor | sensor.solaredge_i2_status_vendor
Solaredge I2 Storage Charge Limit | number.solaredge_i2_storage_charge_limit
Solaredge I2 Storage Command Mode | select.solaredge_i2_storage_command_mode
Solaredge I2 Storage Command Timeout | number.solaredge_i2_storage_command_timeout
Solaredge I2 Storage Control Mode | select.solaredge_i2_storage_control_mode
Solaredge I2 Storage Default Mode | select.solaredge_i2_storage_default_mode
Solaredge I2 Storage Discharge Limit | number.solaredge_i2_storage_discharge_limit
Solaredge I2 Temp Sink | sensor.solaredge_i2_temp_sink
Solaredge I2 Version | sensor.solaredge_i2_version
Solaredge M1 AC Current | sensor.solaredge_m1_ac_current
Solaredge M1 AC Current A | sensor.solaredge_m1_ac_current_a
Solaredge M1 AC Current B | sensor.solaredge_m1_ac_current_b
Solaredge M1 AC Current C | sensor.solaredge_m1_ac_current_c
Solaredge M1 AC Frequency | sensor.solaredge_m1_ac_frequency
Solaredge M1 AC Power | sensor.solaredge_m1_ac_power
Solaredge M1 AC Power A | sensor.solaredge_m1_ac_power_a
Solaredge M1 AC Power B | sensor.solaredge_m1_ac_power_b
Solaredge M1 AC Power C | sensor.solaredge_m1_ac_power_c
Solaredge M1 AC Voltage AB | sensor.solaredge_m1_ac_voltage_ab
Solaredge M1 AC Voltage AN | sensor.solaredge_m1_ac_voltage_an
Solaredge M1 AC Voltage BC | sensor.solaredge_m1_ac_voltage_bc
Solaredge M1 AC Voltage BN | sensor.solaredge_m1_ac_voltage_bn
Solaredge M1 AC Voltage CA | sensor.solaredge_m1_ac_voltage_ca
Solaredge M1 AC Voltage CN | sensor.solaredge_m1_ac_voltage_cn
Solaredge M1 AC Voltage LL | sensor.solaredge_m1_ac_voltage_ll
Solaredge M1 AC Voltage LN | sensor.solaredge_m1_ac_voltage_ln
Solaredge M1 Device | sensor.solaredge_m1_device
Solaredge M1 Exported A kWh | sensor.solaredge_m1_exported_a_kwh
Solaredge M1 Exported B kWh | sensor.solaredge_m1_exported_b_kwh
Solaredge M1 Exported C kWh | sensor.solaredge_m1_exported_c_kwh
Solaredge M1 Exported kWh | sensor.solaredge_m1_exported_kwh
Solaredge M1 Imported A kWh | sensor.solaredge_m1_imported_a_kwh
Solaredge M1 Imported B kWh | sensor.solaredge_m1_imported_b_kwh
Solaredge M1 Imported C kWh | sensor.solaredge_m1_imported_c_kwh
Solaredge M1 Imported kWh | sensor.solaredge_m1_imported_kwh
Solaredge M1 Meter Events | sensor.solaredge_m1_meter_events
Solaredge M1 Version | sensor.solaredge_m1_version

</body>

</html>
`

[image]
[image]
[image]

Londonlad95, 2023-10-14:
I forgot to include the modbus device entities on the front end; here they are. 

[image]
[image]

springfall2008, 2023-10-14:
Thanks, I'll try to work up a configuration for a read-only version at first

Do you know how you set the charge time?

Londonlad95, 2023-10-14:
In order to charge from the Grid, I have to set the AC charge policy to "Always Allowed", and then set the storage command mode to "Charge from Solar Power and Grid" Then it charges from the grid. I don't have a way of setting a time on that apart from the automation running and looking at the octopus target and then updating those statuses.

springfall2008, 2023-10-15:
I'm having trouble tallying some of the pictures to the table, not all seem to match.

Do you have the name for 'Available Energy' as shown in the picture?

I also need two sensors I can't see here:

1. The house consumption in kWh - Givenergy calls that 'load'. This is used to predict future consumption in Predbat.
2. The amount of Solar energy produced in kWh so far.

If you can find these I'll do the updates and we can give it a try.

I've created a branch here with a few minor updates to Predbat and an example configuration (solaredge.yaml) which would be used instead of the default apps.yaml

The code from Predbat with the update and the template configuration is on this branch:

https://github.com/springfall2008/batpred/tree/solar_edge

Londonlad95, 2023-10-15:
Hi,

So the available energy as seen above, is from: sensor.solaredge_b1_available_energy

As for the house load, I believe it's sensor.solaredge_i2_ac_power

Now as for the Solar Energy produced, that's a tricky one, I think. I had to create custom sensor of that from memory, as I use this sensor for that which is an accumulative total ever produced; solar_panel_production_kwh. I got that from one of the guides on how to use the Solare edge multi-integration on the home assistant forums.

springfall2008, 2023-10-16:
Okay great, I think I have something to test.

Can you go through the normal Predbat install as if it was a GE inverter (install AppDaemon, enable AppDaemon in HACS, installed Predbat). Then rather than changing the template configuration it installs:

- Copy the the solaredge.yaml from here: https://github.com/springfall2008/batpred/blob/solar_edge/solaredge.yaml and overwrite config/appdaemon/apps/batpred/config/apps.yaml with this version.
- Copy the beta version of Predbat from here: https://github.com/springfall2008/batpred/blob/solar_edge/apps/predbat/predbat.py into config/appdaemon/apps/batpred/predbat.py

And have a look at the appdaemon log file to see what happens.

I'd recommended in appdaemon.yaml to point your logs to a separate file so that you can copy/paste them easily, just add this to the bottom on appdaemon.yaml:

```
logs:
  main_log: 
    filename: /config/appdaemon.log
    log_size: 10000000
```

RobinXe, 2023-11-05:
For info, the B1 'available energy' sensor doesn't seem to be indicative of the live state of charge. Instead, it appears to be an indicator of what the BMS thinks 100% is, so the live charge level of the battery, in kWh, would be B1 available energy * ( B1 state of energy/100).

M1's 'AC power', positive or negative, indicates the flow out/in at the grid feed (normally). It does not distinguish, on its own, whether the import is to the house or to the battery, if it's being charged from the grid. I1's 'AC power' sensor would be negative in that case, however.

Hope that all makes sense.

fboundy, 2023-11-07:
Hi - I'm helping Trefor out with adding additional inverter brands. In addition to the above can you answer the following please?

- What format are the charge/discharge start/end entities? (i.e to they take a string as HH:MM(Givenerrgy) or are there separate entities for Hours and Minutes (Solis)
- Once you have set the start/end time entities, do they automatically get written to the inverter or do you have to press a button?
- Do you set Power or Current during a timed Charge/Discharge?
- Can you set a Target SOC for a timed charge or discharge period?
- Does the inverter report SOC in kWh or Percent?
- Is there a specific boolean entity to toggle to enable Timed Charge and/or Discharge?
- Can you set a Reserve or Backup SOC?
- Does the inverter have a specific Mode that needs setting for Timed Charge, Discharge or Backup?
-

RobinXe, 2023-11-07:
Hi! Let's see:

There isn't the ability to write timed slots to the inverter. The process for triggering a grid charge/discharge is as follows:

One time only, unless reverted:
* Set 'Storage Control Mode' to 'Remote Control'.

At slot time:
* Set 'Storage Command Timeout' to length of slot, in seconds.
* Set 'Storage Command Mode' to 'Charge from Solar Power and Grid' or 'Discharge to Maximise Export'.

You can set 'Storage Charge Limit'/'Storage Discharge Limit', both in watts. These are not specific to any slot, and will persist until midnight-ish, when they'll revert to 11400W, which I believe is just the maximum value of the register, as the max for the battery is 5000W.

You cannot set a target SoC, just the power limit and duration.

The inverter reports SoC as a %age, but an energy can be derived using that and the 'available energy'.

You can set a backup reserve, in %age.

Hopefully that covers all your queries, happy to expand/clarify as required. It sounds like, if other inverters manage timed charging/discharging onboard, then SolarEdge might require an additional control component too keep track of slots within HA, and trigger them at the appropriate time.

fboundy, 2023-11-07:
Thanks! Every inverter really does do things differently…

I’ll have a dig through what you’ve sent and see it I can set up a test setup.

Arachnid, 2024-01-17:
Any progress on this? I'll soon have an installation with a SolarEdge inverter, and would be happy to help debug this and try it out.

springfall2008, 2024-02-04:
@Arachnid if you want to try then maybe get together a predbat config using one of the templates from something like Sofar and see what you can get working, then come back with the bits you are stuck on?

LupoLoopy, 2024-02-12:
Hi

I've tried to set this up on my install, renaming sensors and creating the required Solar_Panel_Energy sensor.

I'm currently facing this error in the AppDaemon log.

2024-02-12 23:43:53.879437 WARNING pred_bat: ------------------------------------------------------------
2024-02-12 23:45:09.618585 WARNING pred_bat: ------------------------------------------------------------
2024-02-12 23:45:09.618916 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-02-12 23:45:09.619220 WARNING pred_bat: Worker Ags: {'id': 'ef2f4d5f90f64a03b5644ab5fb85f01d', 'name': 'pred_bat', 'objectid': '33dccc4036124d8fba34c5e3b1505d40', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x7f04675b5a50>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-02-12 23:45:09.619443 WARNING pred_bat: ------------------------------------------------------------
2024-02-12 23:45:09.621130 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 12119, in run_time_loop
    raise e
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 12115, in run_time_loop
    self.update_pred(scheduled=True)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 11122, in update_pred
    self.fetch_inverter_data()
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 10694, in fetch_inverter_data
    inverter = Inverter(self, id)
               ^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 1081, in __init__
    self.inv_has_rest_api = INVERTER_DEF[self.inverter_type]["has_rest_api"]
                            ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
KeyError: 'SE'

The log file seems to be demonstrating generally good discoveries being performed, but ultimately no plan is being generated, the plan card outputting NULL.

I'm happy to keep plugging away at this over the next few weeks as a platform tester, as I'm 99% sure my config file and install are correct. 

I'll disable AppDaemon for tonight (revert to crude NodeRed battery management). If I can get past this bit, I'm sure I can help you guys generalise your code.

LupoLoopy, 2024-02-12:
> You can set a backup reserve, in %age.
> 
You can but it doesn't appear to anything on my ModBus install! If you set the value, it invariably resets to its default. This makes me question if it's a universibly reliable element.

RobinXe, 2024-02-13:
> > You can set a backup reserve, in %age.
> 
> You can but it doesn't appear to anything on my ModBus install! If you set the value, it invariably resets to its default. This makes me question if it's a universibly reliable element.

Yes, I've found that my SE inverter resets itself at least once a day, restoring defaults to all the settings. My solution to this has been a watchdog component to any automations, that is triggered by a change in that setting, to ensure it gets reset as desired following a reversion.

duckfullstop, 2024-03-04:
I have a semi-working, read-only installation up and running - I'll post the relevant config snippets below for those that it helps. Note that some of the sensors (namely `sensor.solar_panel_production_w`, `sensor.solar_panel_production_daily`, and `sensor.solar_house_consumption_w`) are calculated using template sensors, available here: https://github.com/ryanm101/hasolarcfg. Note that the aforementioned beta branch has been removed, so this uses the current stable build.

I'm only classing this as semi-working as for some reason my 24h battery depletion forecast is woefully pessimistic - I'm presuming I've probably just screwed up somewhere and not implemented either my iboost or EV charging scenario properly, but if you see similar (past 24h is fine!) please do chime in.

```yaml
load_today:
    - sensor.solaredge_i1_ac_energy_kwh
  import_today:
    - sensor.solaredge_m1_imported_kwh
  export_today:
    - sensor.solaredge_m1_exported_kwh
  pv_today:
    - sensor.solar_panel_production_daily  # Custom template sensor
  
  num_inverters: 1
  # Inverter type stubbed to SX4 to allow execution - SE is an invalid type in current head version
  inverter_type: "SX4"
  # inverter_type: "SE"

  # auto_restart:
  # - service: homeassistant.reload_config_entry
  #   entity_id: sensor.solaredge_i1_status

  battery_rate_max:
    - sensor.solaredge_b1_max_charge_power
  charge_rate:
    - number.solaredge_i1_storage_charge_limit
  discharge_rate:
    - number.solaredge_i1_storage_discharge_limit
  battery_power:
    - sensor.solaredge_b1_dc_power
  pv_power:
    - sensor.solar_panel_production_w # Custom template sensor
  load_power:
    - sensor.solar_house_consumption_w # Custom template sensor
  soc_kw:
    - sensor.solaredge_b1_available_energy
  soc_percent:
    - sensor.solaredge_b1_state_of_energy
  soc_max:
    - sensor.solaredge_b1_maximum_energy
  reserve:
    - number.solaredge_i1_backup_reserve
```
Control will require changes to predbat's codebase, but as previously discussed, should be relatively trivial to implement and can probably (I'm guessing!) be implemented by extending off the Home Assistant entity control that already exists.

marlon89br, 2024-03-04:
I'm having a SolarEdge system installed in the next few weeks, so I'll follow this thread and I'm also happy to help once I have the system in place.

springfall2008, 2024-03-04:
Looking at the template here:

https://github.com/springfall2008/batpred/blob/main/templates/solaredge.yaml

There's a couple of differences, this is probably another template:

```
  pv_today:
    - sensor.solar_panel_production_kwh   # This is a custom sensor - documentation TBD
 ```

And this stuff you may want to add:

```
  charge_start_time:
   - "00:00:00"
  charge_end_time:
   - "00:01:00"
  charge_limit:
   - number.solaredge_i2_ac_charge_limit
  scheduled_charge_enable:
   - off
  scheduled_discharge_enable:
   - off
  discharge_start_time:
   - "00:00:00"
  discharge_end_time:
   - "00:01:00"
```

I've re-created the branch : https://github.com/springfall2008/batpred/tree/solar_edge and added the missing 'SE' configuration table, however I don't know how to control the inverter. How can Predbat start a charge and how can it start a discharge? Can you provide an example automation in HA to show how this is done and then I can add it to the code?

duckfullstop, 2024-03-04:
> I've re-created the branch : https://github.com/springfall2008/batpred/tree/solar_edge and added the missing 'SE' configuration table, however I don't know how to control the inverter. How can [...]

Here you go!

```yaml
choose:
  - conditions:
      - type: is_on
        condition: device
        device_id: 19f5b7526346682eb7ea1c8fd071a8ac
        entity_id: 8d26dee3ef53d7a39fe7dd5e8f37c23e
        domain: binary_sensor
    sequence:
      - device_id: a0f1da08c6f00368315c3c7bed85f2db
        domain: select
        entity_id: bb33d53355844f55e69380b2f11b4ed5
        type: select_option
        option: Charge from Solar Power and Grid
        alias: Charge from Grid
    alias: Off-Peak
  - conditions:
        - condition: state
          entity_id: binary_sensor.octopus_energy_a_<redacted>_intelligent_dispatching
          state: "on"
          alias: Intelligent Octopus block / Off-Peak
    sequence:
      - device_id: a0f1da08c6f00368315c3c7bed85f2db
        domain: select
        entity_id: bb33d53355844f55e69380b2f11b4ed5
        type: select_option
        option: Charge from Solar Power and Grid
        alias: Charge from Grid
    alias: Intelligent Boost
  - conditions:
      - alias: Zappi charging from PV
        condition: state
        entity_id: sensor.myenergi_zappi_status
        state: Charging
        enabled: true
    sequence:
      - device_id: a0f1da08c6f00368315c3c7bed85f2db
        domain: select
        entity_id: bb33d53355844f55e69380b2f11b4ed5
        type: select_option
        option: Charge from Solar Power
        alias: Charge from Solar only
default:
  - device_id: a0f1da08c6f00368315c3c7bed85f2db
    domain: select
    entity_id: bb33d53355844f55e69380b2f11b4ed5
    type: select_option
    option: Maximize Self Consumption
    alias: Maximise Self Consumption
```

All commands are targetted at the `i1` device (the inverter).

Here are the valid modes for `select.solaredge_i1_storage_control_mode` / `select.solaredge_i1_storage_default_mode`:

```yaml
options:
  - Solar Power Only (Off)
  - Charge from Clipped Solar Power
  - Charge from Solar Power
  - Charge from Solar Power and Grid
  - Discharge to Maximize Export
  - Discharge to Minimize Import
  - Maximize Self Consumption
friendly_name: Solaredge I1 Storage Command Mode
```

The control flow as discussed in https://github.com/springfall2008/batpred/issues/181#issuecomment-1798864049 is correct as far as I'm aware. `storage_default_mode` stays active until otherwise changed, `storage_control_mode` applies for `number.solaredge_i1_storage_command_timeout` seconds.

One other thing: your extra config calls out `number.solaredge_i2_ac_charge_limit` - it may be pertinent to change to `i1` as most setups use one inverter.

Londonlad95, 2024-03-05:
The reason the i2 is called out is because my inverter for some bizarre reason that I cannot resolve is that it’s i2 and not i1.  I have tried changing it to i1 but to no avail.

springfall2008, 2024-03-05:
Please find on the branch an updated template with the charge/discharge service settings and an updated predbat.py to support this. Can you give it a try please?

duckfullstop, 2024-03-05:
I've just switched to the branch and command seems to be working as expected: I'll observe it over the next 24-48h and report back if there's any serious issues.

Initial observations:
* `INFO pred_bat: WARN: Inverter 0 Trying to write 4.0 to reserve didn't complete got 0` (where reserve is `number.solaredge_i1_backup_reserve`)
  * presumably this is an issue with the backup reserve setting as previously discussed, if I had to guess this is because we don't have a backup interface available on the modbus, so the inverter just ignores related commands. 
  * Disabled by setting `has_reserve_soc` on the appropriate object to `False` for the interim.
* `discharge_stop_service` - should we be setting this in config?
* Services call `storage_command_mode` by default, which is probably a good move as after the timeout it'll revert to the default mode - which means if predbat stops talking to the inverter for some reason it can use a sane default (like _Maximise Self Consumption_).
  * May be worth calling out an appropriate change to `number.solaredge_i1_storage_command_timeout`(3600 seconds is default, sounds sane?) in the documentation if this behaviour is indeed desired.
* _Power Control Options_, as well as _Enable Battery Control_, must be enabled in the _Solaredge Modbus Multi_ integration configuration.

Happy to contribute / flesh out setup documentation if it'd help!

springfall2008, 2024-03-06:
Great news, please review my pull request that is now updated: https://github.com/springfall2008/batpred/pull/817/files

- I change the i2 to i1 in the template and cleaned it up a little
- I added documentation, please review and comment
- I changed the stop method to Maximise Self Consumption
- I also updated predbat so the inverter configuration options inside predbat.py can be overridden inside apps.yaml if need be, this can allow others to create new integrations quickly

Once we are happy with this and its merged I can look at charge and discharge freeze modes.

duckfullstop, 2024-03-12:
Sorry for the delay, life's been happening 😵‍💫 The functionality in #817 seems to be working perfectly after a week's testing (still getting strange expected load but that's out of scope for this issue and I'll tackle it another time) 🎉

I guess just charge / discharge freeze to consider now?

Noodleyman, 2024-03-14:
> charge_rate and discharge_rate should be configured in apps.yaml to an HA control that allows you to control the inverter charge and discharge rate.
> 
> e.g. on my givenergy inverter it points to a control that can be set to a value between 2600 and zero, representing full rate charge/discharge down to no charge/discharge.
> 
> Predbat uses this to control the rate of battery charge and discharge.
> 
> So yes, it does sound like a question for the Solaredge modbus integration. If the answer is that there is no equivalent of charge/discharge rate for your inverter type then will need to find a different way of turning the inverter charge/discharge on/off and Trefor will have to adapt the code. But if you can ask the question first

Good Morning, thank you for the feedback, very helpful.  as you hit reply I had just finished opening a bug for this here:
https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/561

I am suspecting that the entities are correct, but the modbus integration seems to not populate them. The debug log shows it is pulling the values which the entities should be populated with. It may be an issue due to me using a prefix rather than standard config. I shall see what feedback that gets.

I am curious what values others have for the entities from their SE inverter via modbus

Noodleyman, 2024-03-14:
I figured it out. if the mode isnt set to "Remote Control" it won't populate those values. I had it still set to a time based plan. once it was changed it worked correctly. 

I did notice however a miss in the templates for the SE inverter. the following should be added to allow Octopus saving sessions to be used

```
  # Octopus saving session points to the saving session Sensor in the Octopus plugin, when enabled saving sessions will be at the assumed
  # Rate is read automatically from the add-in and converted to pence using the conversion rate below (default is 8)
  octopus_saving_session: 're:(binary_sensor.octopus_energy([0-9a-z_]+|)_saving_session(s|))'
  octopus_saving_session_octopoints_per_penny:8
```

gcoan, 2024-03-14:
> I figured it out. if the mode isnt set to "Remote Control" it won't populate those values. I had it still set to a time based plan. once it was changed it worked correctly.
>
>
>
> I did notice however [...]

Ah great. If there's extra guidance needed to the documentation please let me know and I'll add it in.

Yes the apps.yaml for the different inverter varieties is a problem because new features are not always copied into all the different configuration files.
As part of documentation overhaul I'm planning on restructuring apps.yaml to better follow the documentation and also try to merge them into just one version that can be edited to be suitable for all the different inverter types that predbat supports

Geoffrey

Noodleyman, 2024-03-14:
it appears there is a depenency on the option being "Remote Control", but for me, I wanted things to be essentially in read only mode, and left alone until my config is dialed in and I know it is good to cut over. the entities are only populated when it is in read only mode, thus... one or the other.  It may be worth at least keeping the content in the file but commented out with a note to review, and the docs for SE may need a referecne to say that remote control mode is needed. it wasn't clear to me this was required, I only figured it out with a source code review. 

as I am porting from the SE online platform, I'm coming from several seasonal profiles setup by date range, so it makes sense that this setting needs to be changed.  I suspect this would be common of a lot of SE users.

Noodleyman, 2024-03-22:
Thought I would follow up on my efforts as I hit a couple more snags. I'm likely doing something wrong but wanted to share the experience in case there is a gap in documentation.

Firstly, I am aware that the updates i #843  caused an issue for SolarEdge config and I have applied the changes in #875 to my install.  It's possible I've missed some other changes required from other threads?

I am still seeing the following error within the PredBat logs on startup:

```
2024-03-22 11:33:09.784235 WARNING pred_bat: ------------------------------------------------------------
2024-03-22 11:33:09.784724 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-03-22 11:33:09.785112 WARNING pred_bat: Worker Ags: {'id': 'a67685a1ec8a4358b968f0d8e781eff0', 'name': 'pred_bat', 'objectid': 'e63d4f43bbf3455087e866ad97f58d82', 'type': 'scheduler', 'function': <bound method PredBat.update_time_loop of <predbat.PredBat object at 0xb4700bb0>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 15, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-03-22 11:33:09.785478 WARNING pred_bat: ------------------------------------------------------------
2024-03-22 11:33:09.792103 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 12650, in update_time_loop
    raise e
  File "/config/apps/predbat.py", line 12645, in update_time_loop
    self.update_pred(scheduled=False)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 11665, in update_pred
    status, status_extra = self.execute_plan()
                           ^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 10466, in execute_plan
    inverter.adjust_charge_window(charge_start_time, charge_end_time, self.minutes_now)
  File "/config/apps/predbat.py", line 3610, in adjust_charge_window
    self.press_and_poll_button(entity_id)
  File "/config/apps/predbat.py", line 3643, in press_and_poll_button
    entity = self.base.get_entity(entity_id)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/adapi.py", line 3188, in get_entity
    self._check_entity(namespace, entity)
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 231, in inner_sync_wrapper
    f = run_coroutine_threadsafe(self, coro(self, *args, **kwargs))
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 313, in run_coroutine_threadsafe
    result = future.result(self.AD.internal_function_timeout)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 456, in result
    return self.__get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 401, in __get_result
    raise self._exception
  File "/usr/lib/python3.11/site-packages/appdaemon/adapi.py", line 588, in _check_entity
    if "." not in entity:
       ^^^^^^^^^^^^^^^^^
TypeError: argument of type 'NoneType' is not iterable

2024-03-22 11:33:09.792627 WARNING pred_bat: ------------------------------------------------------------
```

I've rolled back to Time of Use config for now

duckfullstop, 2024-03-22:
Can you copy your relevant config yaml? If I had to guess you've got an entity either set wrong or not providing correct data (either way we might need documentation updates...)

Noodleyman, 2024-03-22:
Sure, thank you. here is my apps.yaml config file.  I've likely got someting wrong, but do appreciate another pair of more experienced eyes who may be able to spot the issue without hours of trial and error :)

```
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # XXX: This is a configuration template, delete this line once you edit your configuration
  # template: True

  # Timezone to work in
  timezone: Europe/London

  #
  # Sensors, currently more than one can be specified and they will be summed up automatically
  # however if you have two inverters only set one of them as they will both read the same.
  #
  load_today:
    - sensor.solaredge_i1_ac_energy
    #- sensor.solaredge_i1_ac_energy_kwh
  import_today:
    - sensor.solaredge_m1_imported
    #- sensor.solaredge_m1_imported_kwh
  export_today:
    - sensor.solaredge_m1_exported
    #- sensor.solaredge_m1_exported_kwh
  pv_today:
    - sensor.solar_panel_production_kwh   # This is a custom sensor - see Readme on how to create or comment out
  #
  # Controls/status - must by 1 per inverter
  #
  num_inverters: 1
  # Inverter type stubbed to SX4 to allow execution - SE is an invalid type in current head version
  inverter_type: "SX4"
  #inverter_type: "SE"

  # Can be enabled if you have a backup interface
  #inverter:
  #  has_reserve_soc: True

  battery_rate_max:
    - sensor.solaredge_b1_max_charge_power
  charge_rate:
    - number.solaredge_i1_storage_charge_limit
  discharge_rate:
    - number.solaredge_i1_storage_discharge_limit
  battery_power:
    - sensor.solaredge_b1_dc_power
  pv_power:
    - sensor.solar_panel_production_w # This is a custom sensor - see Readme on how to create or comment out
  load_power:
    - sensor.solar_house_consumption_w # This is a custom sensor - see Readme on how to create or comment out
  soc_percent:
    - sensor.solaredge_b1_state_of_energy
  soc_max:
    - sensor.solaredge_b1_maximum_energy
  # Can be enabled you have a backup interface
  #reserve:
  #  - number.solaredge_i1_backup_reserve
  charge_start_time:
   - "02:00:00"
  charge_end_time:
   - "05:00:00"
  charge_limit:
   - number.solaredge_i1_ac_charge_limit
  scheduled_charge_enable:
   - off
[... 224 lines cut ...]
  # If you have an 80% DoD battery that falsely reports it's kwh then set it to 0.8 to report the real figures
  battery_scaling: 1.0

  # Can be used to scale import and export data, used for workarounds
  import_export_scaling: 1.0

  # Export triggers:
  # For each trigger give a name, the minutes of export needed and the energy required in that time
  # Multiple triggers can be set at once so in total you could use too much energy if all run
  # Creates an entity called 'binary_sensor.predbat_export_trigger_<name>' which will be turned On when the condition is valid
  # connect this to your automation to start whatever you want to trigger
  export_triggers:
     - name: 'large'
       minutes: 60
       energy: 1.0
     - name: 'small'
       minutes: 15
       energy: 0.25

  # If you have a sensor that gives the energy consumed by your solar diverter then add it here
  # this will make the predictions more accurate. It should be an incrementing sensor, it can reset at midnight or not
  # It's assumed to be in Kwh but scaling can be applied if need be
  #iboost_energy_today: 'sensor.xxxxx'
  #iboost_energy_scaling: 1.0
```

gcoan, 2024-03-22:
>   charge_start_time:
>    - "02:00:00"
>   charge_end_time:
>    - "05:00:00"
>   charge_limit:
>    - number.solaredge_i1_ac_charge_limit
>   scheduled_charge_enable:
>    - off
>   scheduled_discharge_enable:
>    - off
>   discharge_start_time:
>    - "18:30:00"
>   discharge_end_time:
>    - "19:00:00"
> 
>   # Export limit is a software limit set on your inverter that prevents exporting above a given level
>   # When enabled Predbat will model this limit
>   export_limit:
>    - 3670
>   #  - 3670
>   #
>   # The maximum rate the inverter can charge and discharge the battery can be overwritten, this will change
>   # the register programming and thus cap the max rates. The default is to use the maximum supported rates (recommended)
>   #
>   inverter_limit_charge:
>    - 5000
>   inverter_limit_discharge:
>    - 5000

The above bits are not valid YAML.  You need two spaces indented in the line before the dash

i.e.
```yaml
  inverter_limit_charge:
    - 5000
```

not

```yaml
  inverter_limit_charge:
   - 5000
```

By the way, you may want to extend your days_previous at the moment its weighting the same day this week, the previous week and the previous week (assuming you have enough history in home assistant).  You may want to blend that with some days of the current week to get a more even average

Noodleyman, 2024-03-22:
Thank you for calling out those indent issues. I did run the code through a couple of online YAML validators and neither of them reported that was an issue so I thought it was in good shape.   I have fixed those two items and shall see if things are looking healthier. 

I will 100% review days_previous_weight further, thank you for the advice.  I had left it at the current default value from the copy/paste template.

It would be nice if days of the week could be grouped, as week day usage will be very different to weekend usage. so grouping week days and weekends would (in our case) make more sense. I've not read about this being an option yet, but that's something I'll research more.  If I still have an issue I'll report back. thank you for taking the time to help a n00bie :)

Noodleyman, 2024-03-22:
still seeing some oddities. I suspect it is because number.solaredge_i1_storage_discharge_limit returns "unavailable" from the modbus, but their documentation says this is expected in some cases:

https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Template-Design-Notes

```
2024-03-22 13:57:03.238696 WARNING pred_bat: ------------------------------------------------------------
2024-03-22 14:40:36.313949 WARNING pred_bat: ------------------------------------------------------------
2024-03-22 14:40:36.315066 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-03-22 14:40:36.317212 WARNING pred_bat: Worker Ags: {'id': 'b03316b5ffbd4be58166c4d675b847b2', 'name': 'pred_bat', 'objectid': '12a07a489c9b4d6d960e342bf2e280ca', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0xb314aa70>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-03-22 14:40:36.318294 WARNING pred_bat: ------------------------------------------------------------
2024-03-22 14:40:36.325933 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 12672, in run_time_loop
    raise e
  File "/config/apps/predbat.py", line 12668, in run_time_loop
    self.update_pred(scheduled=True)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 11665, in update_pred
    status, status_extra = self.execute_plan()
                           ^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 10600, in execute_plan
    inverter.adjust_discharge_rate(inverter.battery_rate_max_discharge * MINUTE_WATT)
  File "/config/apps/predbat.py", line 2925, in adjust_discharge_rate
    self.write_and_poll_value("discharge_rate", entity, new_rate, fuzzy=(self.battery_rate_max_discharge * MINUTE_WATT / 25))
  File "/config/apps/predbat.py", line 3036, in write_and_poll_value
    matched = abs(float(current_state) - new_value) <= fuzzy
                  ^^^^^^^^^^^^^^^^^^^^
ValueError: could not convert string to float: 'unavailable'

2024-03-22 14:40:36.328391 WARNING pred_bat: ------------------------------------------------------------
```

Noodleyman, 2024-03-22:
ok, so here's a silly thing to add to the docs... If you've previously been using the Solar Edge monitoring platform, and/or you manage your own storage profiles then you need to make sure the only applicable profile is "Max Self Consumption". 

the SE platform will overwrite the config in place. I've a variety of seasonal profiles with schedules setup which seems to modify the ModBus module back to "time of use" from "remote control", thus why the entities are now unavailable.

Noodleyman, 2024-03-22:
Ah ha! inverter type. I hadn't realised that had changed between versions. Thank you, will try that as well.  I was still getting issues with the earlier changes. I ran out of time to experiment today, so will likely play with this again next week. Thank you for your feedback, I really do appreciate you taking the time :)

gcoan, 2024-03-22:
> I think some better setup docs (probably with a walk-through) specifically for SE users would be a good idea because there's seemingly tonnes of little gotchas / pitfalls that are easy to fall down.

I've been working to improve and deepen the Predbat documentation and am happy to help merge what ever text is produced into the documentation and sample apps.yaml, but it needs to be written by someone who has a Solax inverter as mine are GivEnergy!

A step by step would be a good idea.  The install instructions are at the moment centred on GivEnergy, they make a nod to the different dependent inverter control software but there's no real detail on these as no-one has written it.

As much as possible I am aiming to have just one set of comprehensive Predbat documentation (and ultimately apps.yaml file) rather than having different forks and variants for different setups and inverters. Not to say that we can't have specific documentation for specific inverters but this is a clear 'side street' rather than a different 'main road'.
This should make it much easier to maintain and control going forward and ensures that things don't get missed out - having 3 or 4 different template apps.yaml's for example means new features are easy to miss getting added.  If we can have 1 consolidated file with sections for the different inverters it'll make it much clearer.

Noodleyman, 2024-03-23:
Good Morning,

I managed to find a little time this morning over a coffee to review the inverter_type change. It does appear this was the last piece of the puzzle. with this changed things now seem to be behaving (at least so far today). I'll keep an eye on it but after a few hours it does appear to be working without errors. Exciting times, special thanks to you all for helping out and checking my config. I doubt I would have got it working without your input. I really do appreciate it.

Noodleyman, 2024-03-23:
I thought I would share another "gotcha" when setting up for SolarEdge, in case anybody else wonders or has the same concern.

I use various seasonal profiles to manage the SE system from the online monitoring platform. When setting up the Modbus module it is required to be in "Remote Control" mode to update some of the entities PredBat requires. If in any other mode, those entities have "undefined" values. 

So, in my case I kept it as "time of use" and left PredBat to learn my usage. However, this has caused issues with the data it collected as dueing battery charge events, or export battery to grid events the "undefined" entities are a problem. IE, PredBat doesn't see the event... so currently, all of my overnight battery charge windows (somewhere between 2-5AM for Octopus Flux) are reporting 0kWh load predicted. 

in my case, I didn't want to run the system until it had learnt my usage patterns, but to learn the usage pattersn you have to give up your existing profiles..  So there will be a period of disruption to normal usage whilst the system gets up to speed. It might be a good idea to have SE systems use a different config for days_previous for the first week or so (learn from previous day etc), then evolve it into a more deep config as data is accumulated. 

The other oddity, is if you do not change "SE online platform -> Site Admin -> Energy Manager: set to Maximize self-consumption" then the online monitor platform will change your modbus mode from remote control to "time of use" at the time of your next profile scheduled event, which in turn makes those entities unavaialble again and predbat can't learn your usage... (URG!). 

So you have to give PredBat full control of the system (with read only enabled) at the very start and give up the schedules to do a nice cutover.

LupoLoopy, 2024-03-24:
To feedback, I've got this working on my setup nicely.

I have however seen a new issue. Specifically, the minimum battery charge variables are not being honoured for my SE setup, it having driven the battery down to zero.

This error was showing in the logbook at the time- not sure if it's relevant:

Status changed to ERROR: Exception raised argument of type 'NoneType' is not iterable

Where do the richer logs live, so I can take a look so I might feedback in more fully please.

gcoan, 2024-03-24:
yes that is relevant, Predbat crashed

Logfile location is covered in the documentation:

https://springfall2008.github.io/batpred/output-data/#predbat-logfile

Noodleyman, 2024-03-25:
I also wanted to follow up to confirm after the recent recommended changes, things see to be working correctly. thank you!

time to tune the forecasts :)

springfall2008, 2024-03-25:
@Noodleyman are there any changes I need to roll back into Predbat?

Can you share your final config.yaml so I can put it into Predbat as a template? Some words with this on how to set things up would be very useful for other users.

Noodleyman, 2024-03-25:
Good morning @springfall2008 ,  below is my existing apps.yaml config from my working configuration on version 7.16.8 of PredBat, with the change noted in #875 manually applied to predbat.py.  Sharing what I can now, but have limited time this week for any further details.

**I am using:**

- Octopus Energy addon in Home Assistant, and we are on Octopus Flux
- Solcast for PV forecast
- Using an SE6000H inverer (upgrading to SE8000H shortly), with 11.07kWp (upgrading to 15 shortly) and a single 10kWh battery
- I have an enforced export limit of 3.68kWh (legal in allowed by DNO in the UK)
- I do not have an electric vehicle so have EV related stuff disabled

I will screen dump some screen shots of my other config below (I am still adjusting these, but they are working for now). I have very limited time to play with it this week :(

**Config:**
```
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # XXX: This is a configuration template, delete this line once you edit your configuration
  # template: True

  # Timezone to work in
  timezone: Europe/London

  #
  # Sensors, currently more than one can be specified and they will be summed up automatically
  # however if you have two inverters only set one of them as they will both read the same.
  #
  load_today:
    - sensor.solaredge_i1_ac_energy
    #- sensor.solaredge_i1_ac_energy_kwh
  import_today:
    - sensor.solaredge_m1_imported
    #- sensor.solaredge_m1_imported_kwh
  export_today:
    - sensor.solaredge_m1_exported
    #- sensor.solaredge_m1_exported_kwh
  pv_today:
    - sensor.solar_panel_production_kwh   # This is a custom sensor - see Readme on how to create or comment out
  #
  # Controls/status - must by 1 per inverter
  #
  num_inverters: 1
  # Inverter type stubbed to SX4 to allow execution - SE is an invalid type in current head version
  inverter_type: "SE"

  # Can be enabled if you have a backup interface
  #inverter:
  #  has_reserve_soc: True

  battery_rate_max:
    - sensor.solaredge_b1_max_charge_power
  charge_rate:
    - number.solaredge_i1_storage_charge_limit
  discharge_rate:
    - number.solaredge_i1_storage_discharge_limit
  battery_power:
    - sensor.solaredge_b1_dc_power
  pv_power:
    - sensor.solar_panel_production_w # This is a custom sensor - see Readme on how to create or comment out
  load_power:
    - sensor.solar_house_consumption_w # This is a custom sensor - see Readme on how to create or comment out
  soc_percent:
    - sensor.solaredge_b1_state_of_energy
  soc_max:
    - sensor.solaredge_b1_maximum_energy
  # Can be enabled you have a backup interface
[... 253 lines cut ...]
       minutes: 15
       energy: 0.25

  # If you have a sensor that gives the energy consumed by your solar diverter then add it here
  # this will make the predictions more accurate. It should be an incrementing sensor, it can reset at midnight or not
  # It's assumed to be in Kwh but scaling can be applied if need be
  #iboost_energy_today: 'sensor.xxxxx'
  #iboost_energy_scaling: 1.0
```

**SolarEdge Monitoring Platform Config**
_Note: I have an installer account whic may open up more options than others. If you need one, raise a support case with SolarEdge support and tell them you are a self-installer who wants full control of your own system_
[image]

**SolarEdge Modbus Multi:**
MUST BE IN REMOTE CONTROL MODE (See my notes earlier in this thread for GOTCHAS)
[image]

**PredBat Switches:**
[image]

**PredBat Input Variables:**
Note: SE kit is very effecient compared to others. If you check out their effeciency charts you can expect over 99.x% effeicnecy from PV. battery however is around 94.5 - 95% because of the 400v. I have added 10% load scaling (I think I eat power, so have upscaled it to be cautious), and I have scaled my PV by 10%, although I will likely change this to about 3-5% later. It will all change when I upgrade the system within the next few weeks.

[image]

Noodleyman, 2024-04-02:
Thank you @marlon89br , I was just looking into the same kind of issue and think you're suggestion will fix it. I've been seeing my expected daily usage over estimate, because the exports are going up. 

I had to start over, because my home assistant instance entirely died a week ago, so had to restore which meant I lost some data :/ so am playing catchup now.

marlon89br, 2024-04-02:
Hopefully that fixes it for you. I’ll also keep an eye out when I come back
home if it tracks the actual house load, but looking at the historical data
that looks like the right sensor.

I guess if you confirm it working as well we can update the template.

Noodleyman, 2024-04-02:
I'm also adding some content I got from solar edge support earlier this year when quizzing their dashboards. this may be useful for comparisons.

for their online platform:
𝑏𝑎𝑡𝑡𝑒𝑟𝑦 𝑐𝑜𝑛𝑠𝑢𝑚𝑝𝑡𝑖𝑜𝑛 − 𝑏𝑎𝑡𝑡𝑒𝑟𝑦 𝑐ℎ𝑎𝑟𝑔𝑒 𝑓𝑟𝑜𝑚 𝑠𝑜𝑙𝑎𝑟 = 𝑏𝑎𝑡𝑡𝑒𝑟𝑦 𝑔𝑟𝑖𝑑𝑐ℎ𝑎𝑟𝑔𝑒

Once we have this we can then use:
𝐵𝑎𝑡𝑡𝑒𝑟𝑦 𝑔𝑟𝑖𝑑𝑐ℎ𝑎𝑟𝑔𝑒 + 𝑐𝑜𝑛𝑠𝑢𝑚𝑝𝑡𝑖𝑜𝑛 𝑓𝑟𝑜𝑚 𝑔𝑟𝑖𝑑 = 𝐼𝑚𝑝𝑜𝑟𝑡

Checking the bespoke sensor today _solar_house_consumption_kwh_ reports 14.82kWh between midnight and now, and the online monitoring tool reports 15.1kWh across grid/battery/PV, so it's about right.

Noodleyman, 2024-04-02:
and to follow up, the forecasted consumption is now where it should be. about 350% lower, IE not inflated anymore. I think you were right on the money @marlon89br

duckfullstop, 2024-04-05:
> @duckfullstop mentioned their predictions were very pessimistic and I was observing the same thing on my setup. I think I figured it out… the load_today variable is set to the inverter AC load, and what I observe in my setup is that it includes both import and export, so my load was massively overestimated now that I’m away on holiday. I changed it to the template sensor solar_house_consumption_kwh that I got with the other custom sensors and it looks sensible now.

Yup, you're totally right, sorry for missing this one! I think that's the only relevant change in my config, estimations seem _far_ more on target now (bar some weirdness with Intelligent but I think that's because there's two EVs in play).

Noodleyman, 2024-04-05:
I'm currently tracking what I think is a bug, but I've yet to figure it out. I've seen this happen twice so far. Since my last update I've now got new equipment. running na SE8000H and 15.5kWp.

My estimated load is about correct for the day, and my PV forecasts are about right at around 3kWh per 30 min bloc, But, I do see an odd pattern. Today showed a 20% battery charge, but PredBat wanted to Hold Charge at 41%, but once it starts to charge it seems to get stuck in a charge loop until battery is at 100%.  Hold Charge seems to set Charging mode as active and keep it active, which results in unnessary charging from grid to make up any shortfall in generation. in my case today why it stood out is I was production 9kWh, thus it was in "charging" mode, yet PV generation was fully supporting the house load, max charging the battery and it was exporting to grid... 

I had roughly a 12kWh forecast covering the next 2 hours, and a forecasted usage of around 3kWh. Every cycle of PredBat it would simply walk up the "Hold Charge" by a few % until at 100%. This seems to align with some warnings. I really don't know why it is doing that at the moment. 

I'm unsure if this is in any way related to the SolarEdge config. In a week or so I'll be bevolving the config further to include the Off Grid (backup) module as well.

Noodleyman, 2024-04-08:
Had the same issue again yesterday. It wanted to hold power, then the charge cycle to 100% starts.  for example, battery could be at 35%, but it wants to hold at 40%. it goes into charge mode, predbat updates and suggests it wants to hold at 43%, it charges, predbat updates and suggests it wants to hold at 45%, it charges (continue cycle to 100% battery). most of the time  the value it sets it within a few % of existing battery charge, and it charges from grid and PV.

I do have errors from around this time, but they only showed up once. I suspect if the issue of being stuck in a charge cycle where "hold" = charge until 100% is unrelated as I would expect them to show up every processing loop if it was root cause, (is this an incorrect assumption?)

I did have a thought, does PretBat handle SolarEdge oversising correctly?  SolarEdge allows up to 200% oversizing, I am currently running at 185% oversized.  The inverter can be supplying 8kW, but in parallel the battery can also be charging at 5kW. I don't think this would be an issue, but best to ask.

I noted the following warnings when this happened:
```
Changed to Warn - Return bad float value unavailable from discharge_rate
16:45:14 - 14 hours ago - Supervisor

Changed to Warn - Return bad float value unavailable from battery_power
16:45:11 - 14 hours ago - Supervisor

Changed to Warn - Return bad float value unavailable from soc_percent
16:45:11 - 14 hours ago - Supervisor

Changed to Warn - Return bad float value unavailable from discharge_rate
16:45:11 - 14 hours ago - Supervisor

hanged to Warn - Return bad float value unavailable from charge_rate
16:45:11 - 14 hours ago - Supervisor

Changed to Warn - Return bad float value unavailable from battery_rate_max
16:45:11 - 14 hours ago - Supervisor
```

Just updated to 7.16.11, so will see if anything changes

Noodleyman, 2024-04-09:
following up, I think it was a bug, and is fixed via #852. 

Hopefully that's the last issue.

I've revised my config yesterday to add the backup module as I now have the off grid backup option installed, thus added the following in apps.yaml

```
  inverter:
    has_reserve_soc: True

  reserve:
    - number.solaredge_i1_backup_reserve
```

Not seen a single error since updating and setting the above, hopefully that will continue. :) 

Next up, need to add an automation to dynamically manage the battery reserve percentage based on bad weather conditions, as I'll want to hold power in reserve during storms/bad weather but allow a higher battery usage during nicer weather. we live in the arse end of nowhere, so often get power outages in storms. Looks like I can just directly update the value via the modbus  integration, which due to the config above will feed back into Predbat.

Unless something new crops up, I think we're there! Weeeeeee. Thank you all who took the time to read my updates and provide feedback, your time is appreciated and valued! :)

Noodleyman, 2024-04-09:
Ah, the joys of deciding to share an update, and then shortly after it decides to play up hah.

I guess the bug being discussed above in #852 isn't fixed.   from a moment ago:
[image]

However, "Hold charging" status per PredBat doesn't actually pause charging on the inveter. instead it's put it into "Charge from Solar Power and Grid" mode, which just pulls whatever power it needs from the grid to make up any PV shortfall to charge the battery. I caught this whilst the battery was at 20% and still charging. 
[image]

[image]

I flipped Predbat into read only mode, then changed the battery to max self consumption mode instead which broke the cycle.

Noodleyman, 2024-04-09:
Further updates. would appreciate if anybody could offer some guidance on what else I should review?

From the Predbat AppDaemon v4.4.2 engine (on port 5051), this is likley showing the issue with the never ending charge issue.

```
2024-04-09 11:36:59.633857 WARNING HASS: Code: 500, error: 500 Internal Server Error Server got itself in trouble
2024-04-09 11:36:59.632962 WARNING HASS: Error calling Home Assistant service default/number/set_value
2024-04-09 11:36:56.632681 WARNING HASS: Code: 500, error: 500 Internal Server Error Server got itself in trouble
2024-04-09 11:36:56.630839 WARNING HASS: Error calling Home Assistant service default/number/set_value
2024-04-09 11:36:53.632666 WARNING HASS: Code: 500, error: 500 Internal Server Error Server got itself in trouble
2024-04-09 11:36:53.631869 WARNING HASS: Error calling Home Assistant service default/number/set_value
2024-04-09 11:36:53.326859 INFO pred_bat: Creating predbat dashboard at /homeassistant/predbat_dashboard.yaml
2024-04-09 11:36:53.299003 INFO pred_bat: Completed run status Hold charging with Errors reported (check log)
2024-04-09 11:36:53.229479 INFO pred_bat: WARN: Inverter 0 Trying to write 86 to reserve didn't complete got 83
2024-04-09 11:36:50.625042 WARNING HASS: Code: 500, error: 500 Internal Server Error Server got itself in trouble
2024-04-09 11:36:50.624229 WARNING HASS: Error calling Home Assistant service default/number/set_value
2024-04-09 11:36:47.656922 WARNING HASS: Code: 500, error: 500 Internal Server Error Server got itself in trouble
2024-04-09 11:36:47.645699 WARNING HASS: Error calling Home Assistant service default/number/set_value
2024-04-09 11:36:46.817616 INFO AppDaemon: New client Admin Client connected
Server got itself in trouble
2024-04-09 11:36:44.618788 WARNING HASS: Code: 500, error: 500 Internal Server Error
2024-04-09 11:36:44.616337 WARNING HASS: Error calling Home Assistant service default/number/set_value
2024-04-09 11:36:41.041656 INFO pred_bat: Inverter 0 Current Reserve is 83.0 % and new target is 86 %
2024-04-09 11:36:41.034707 INFO pred_bat: Adjust reserve to target charge 85 % (set_reserve_enable is true)
**2024-04-09 11:36:41.029908 INFO pred_bat: Current SOC 84% is greater than Target SOC 0. Grid Charge disabled.**
2024-04-09 11:36:41.022590 INFO pred_bat: Adjust idle time computed idle is 11:35:00-23:59:00
2024-04-09 11:36:41.015792 INFO pred_bat: Adjust idle time, charge 00:00:00-00:00:00 discharge 00:00:00-00:00:00
2024-04-09 11:36:41.009629 INFO pred_bat: Inverter 0 Adjust force discharge to False, change times from 00:00:00 - 00:00:00 to 00:00:00 - 00:00:00
2024-04-09 11:36:40.986641 INFO pred_bat: Setting ECO mode as we are not yet within the discharge window - next time is 04-09 16:05:00 - 04-09 16:30:00
2024-04-09 11:36:40.981558 INFO pred_bat: Next discharge window will be: 2024-04-09 16:05:00+01:00 - 2024-04-09 16:31:00+01:00 at reserve 87
2024-04-09 11:36:40.973800 INFO pred_bat: Inverter 0 Call service select/select_option with data {'entity_id': 'select.solaredge_i1_storage_command_mode', 'option': 'Charge from Solar Power and Grid'}
**2024-04-09 11:36:40.969558 INFO pred_bat: Inverter 0 Call service template charge_start_service = {'service': 'select.select_option', 'entity_id': 'select.solaredge_i1_storage_command_mode', 'option': 'Charge from Solar Power and Grid'}**
2024-04-09 11:36:40.965996 INFO pred_bat: Adjust idle time computed idle is 11:35:00-23:59:00
2024-04-09 11:36:40.961851 INFO pred_bat: Adjust idle time, charge 00:00:00-00:00:00 discharge 00:00:00-00:00:00
2024-04-09 11:36:40.956257 INFO pred_bat: Charge window will be: 2024-04-09 11:00:00+01:00 - 2024-04-09 13:00:00+01:00 - current soc 84 target 85
2024-04-09 11:36:40.952503 INFO pred_bat: Base discharge window [ 09-04 00:00:00 - 09-04 00:00:00 @ 0p 100.0%, 10-04 00:00:00 - 10-04 00:00:00 @ 0p 100.0% ]
2024-04-09 11:36:40.948059 INFO pred_bat: Base charge window [ ]
2024-04-09 11:36:40.846901 INFO pred_bat: Found 1 inverters totals: min reserve 0.39 current reserve 8.05 soc_max 9.7 soc 8.17 charge rate 5.0 kW discharge rate 5.0 kW battery_rate_min 0.0 w ac limit 450000.0 export limit 3.67 kW loss charge 3 % loss discharge 4 % inverter loss 4 %
**2024-04-09 11:36:40.843547 INFO pred_bat: Note: Inverter does not support charge freeze - disabled
2024-04-09 11:36:40.840616 INFO pred_bat: Note: Inverter does not support discharge freeze - disabled**
2024-04-09 11:36:40.837606 INFO pred_bat: Inverter 0 discharge windows currently [{'start': 0, 'end': 0, 'average': 0}, {'start': 1440, 'end': 1440, 'average': 0}]
2024-04-09 11:36:40.832373 INFO pred_bat: Inverter 0 scheduled discharge enable is False
2024-04-09 11:36:40.827136 INFO pred_bat: Inverter 0 sensor.predbat_SE_0_scheduled_discharge_enable set to False
2024-04-09 11:36:40.822232 INFO pred_bat: Inverter 0 Wrote scheduled_discharge_enable to False successfully and got off
2024-04-09 11:36:40.813950 INFO pred_bat: Inverter 0 Charge settings: timed charged is disabled, power 5.0 kW
2024-04-09 11:36:40.809884 INFO pred_bat: Inverter 0 charge windows currently []
2024-04-09 11:36:40.805807 INFO pred_bat: Inverter 0 scheduled charge enable is False
2024-04-09 11:36:40.800409 INFO pred_bat: Inverter 0 scheduled_charge_enable set to False
2024-04-09 11:36:40.795426 INFO pred_bat: Inverter 0 Wrote scheduled_charge_enable to False successfully and got off
[... cut ...]
Firstly, the charge_rate and discharge_rate entities always show 5000w or 11400w, never any other numerical value. I am curios if this is part of the issue. I suspect this should be showing more real time data about the actual rate of charge, or rate of discharge of the battery? 

The other bit that stands out is the charge & discharge freeze is listed as not supported thus disabled.I'm assuming this is why it won't actually hold properly. In this scenario I would assume it should set Max Self Consumption mode instead, but perhaps it isn't able to do that?

duckfullstop, 2024-04-09:
Could you check the logs on the Home Assistant side? A 500 response implies something's going wrong on that side.

Noodleyman, 2024-04-09:
Ah, good shout. Checking those logs and there are a few issues. 

Failed to connect errors to the modbus around the same time as the 500 errors

[image]

This is an interesting one, although all of those entities are avilable and have values. 
```
Logger: homeassistant.helpers.service
Source: helpers/service.py:251
First occurred: 8 April 2024 at 19:37:31 (15 occurrences)
Last logged: 11:35:51

Referenced entities input_number.predbat_pv_scaling are missing or not currently available
Referenced entities input_number.predbat_set_reserve_min are missing or not currently available
Referenced entities input_number.predbat_load_scaling are missing or not currently available
Referenced entities switch.predbat_set_read_only are missing or not currently available
Referenced entities switch.predbat_debug_enable are missing or not currently available
```

It seems the comms to the inverter reports as unstable, although cross checking the UniFi network logs I can see the connection looks stable so I don't think it's a network issue.  Given that _"SolarEdge inverters only support one Modbus/TCP connection at a time. Additional attempts to connect to the inverter will fail until the last connection is closed"_, it could just be unfortunate timing with the online platform checking in. I'm tempted to block the inverters internet access to see if that improves things on this front.

duckfullstop, 2024-04-09:
> cross checking the UniFi network logs I can see the connection looks stable so I don't think it's a network issue

This is going to sound like I'm going crazy, but are you using a PoE switch (unifi or otherwise), and is the inverter connected directly to it?

Noodleyman, 2024-04-09:
Happy to explore any and all options :) No PoE switches in my network, although I am using PoE injectors for 3 access points which are in the roof of the building.  I am connecting directly over WiFi to the Inverter. Ah, good to know about theonline platform, thank you. I can cross that thought off the list.

duckfullstop, 2024-04-09:
Sounds like your issue is more with the Modbus implementation than batpred in that case, so you might want to try jumping into their support channels for further thoughts. Do triple check that you don't have any other software running that might be trying to talk to the inverter locally (EVCC, for example).

If you're connecting to the inverter over WiFi, is there any way you can hardwire for a while just to rule out wireless issues?

Noodleyman, 2024-04-09:
Unfortunately there is no method to connect via ethernet. I'll do some more monitoring and see if I can gather some more useful information from logs so I can make sure whatever story they tell ends up in the appropriate place. I don't have anything else on the network that will talk to the SE kit.

Thanks for the feedback, appreciated as always. I'm still learning how the solution works, and thus where to gather various data and logs from the combined solutions, so will get there eventually

[21 more replies]

closed 2024-08-06 by springfall2008