springfall2008/batpred #189: Easy way to disable predbat from making changes?

gcoan, 2023-10-17:
99% of the time predbat does an excellent job of managing my battery and charging activities, but there are times, like in the Octopus power-up events where predbat's not able to do the kind of close optimisation I'd like (see #144 ) and I'd like an easy way of turning predbat off or into "manual control mode".

This would also be of benefit to new users of predbat, when I first installed it it started controlling the battery before I'd really understood all the different options.  A read-only master switch (default on for new installs) could help these new users.

St the moment I believe I have to change several switches to stop the different aspects of predbat, and neither of my approaches work as expected.   Plus it requires coding the state of the different predbat switches into a script and keeping that script in line with the way predbat is manually configured to ensure when turning off and on.

Attempt 1: automation script to flip the appropriate switches:
```
alias: Octopus Power Up, turn predbat off and on
description: >-
  On an Octopus Power-Up day, at sunrise, turn off predbat from changing
  inverter settings, then turn back on at 4pm (power-up end)
trigger:
  - platform: sun
    event: sunrise
    offset: 0
    id: power-up-day-start
  - platform: time
    at:
      - "16:00:20"
    id: power-up-end
condition:
  - condition: template
    value_template: >-
      {{ (now().date()) | string ==
      states('input_datetime.octopus_power_up_date') }}
action:
  - choose:
      - conditions:
          - condition: trigger
            id:
              - power-up-day-start
        sequence:
          - service: switch.turn_off
            data: {}
            target:
              entity_id:
                - switch.predbat_set_charge_window
                - switch.predbat_set_discharge_window
                - switch.predbat_set_discharge_freeze
                - switch.predbat_set_reserve_enable
                - switch.predbat_set_reserve_hold
                - switch.predbat_set_soc_enable
                - switch.predbat_balance_inverters_enable
      - conditions:
          - condition: trigger
            id:
              - power-up-end
        sequence:
          - service: switch.turn_on
            data: {}
            target:
              entity_id:
                - switch.predbat_.....
mode: single
```

Attempt 2: put all the required switches in a helper group, and flip that single group switch:
[image]

[image]
[image]
(tried with 'all entities' turned on and off within the group)

With both attempts, the request to change all the predbat switches appears to work, but only the first listed switch (set charge window) actually changes. All of the other switch changes get lost somehow and the switches don't flip over.
May be a workaround to script changing one switch, waiting 1 minute for predbat to run, changing the next, waiting a minute, etc,  but this means it'll take 6 minutes or so to turn predbat on or off!

Approach 3:  Shutdown the appdaemon add-on.   
This works OK and is the approach I have been using, but of course it stops any other python scripts running

Approach 4: Change template to true in apps.yaml
Now I think about it, un-commenting out the # Template: True line in predbat's apps.yaml would have the desired effect of stopping predbat from making any changes.
But how to automate that from a script?  Something along the lines of writing a shell script to use 'sed' (or similar) to edit the apps.yaml file then the shell command integration to invoke the shell script from HA https://www.home-assistant.io/integrations/shell_command/

gcoan, 2023-10-17:
wow that was quick, thanks, I'll give it a go

gcoan, 2023-10-17:
Installed and can see the new switch OK.  Toggled it on and can see it in the logs and the status. 
I'll have to try it out when predbat is trying to 'do something' - probably peak discharge tomorrow evening

springfall2008, 2023-10-17:
Just be careful it won’t reset the controls it will just stop controlling your battery

mpartington, 2023-10-19:
Tried this and works great. One thing I'd like to do (locally) is when I set read only mode, I'd like to restore my settings to 'my standard'. However predbat turns off this register entry, and I cannot for the life of me find a way to turn it back on via GivTCP. I have to go to the portal each time and toggle there.

The one circled in green is toggled by switch.givtcp_ce2146g269_enable_charge_schedule

Any ideas what code to use to turn on the one circled in red?

Thanks

[image]

springfall2008, 2023-10-19:
I don't know how you change it directly, but if you change the charge % in GivTCP it will enable this toggle.

gcoan, 2023-10-19:
The read-only switch worked perfectly for me today, did exactly what I wanted.  predbat left running last night to do an overnight cheap-rate charge (which ironically didn't work but that appeared to be GivTCP getting connection refused errors when talking to the inverters.  Fixed by restarting GivTCP this morning).
Today was an Octopus power-up event.  At sunrise an automation turned off predbat via the read only switch and my own automations managed the battery, holding one empty and letting the other half charge up so that at 2pm they then switched to full charge through to 4pm when the automation cancelled the charge (both batteries now fully charged) and turned predbat back on.
Perfect.
As you say Trefor, do need to make sure you fully control what the batteries and inverters do when predbat is turned off as it may have left things in place like charging or discharging windows that might not be what you want.

springfall2008, 2023-10-19:
Glad it’s working out for you :)

closed 2023-10-22 by gcoan