springfall2008/batpred #195: Repeatedly turning AC charge off and on

slopemad, 2023-10-20:
I spotted in Home Assistant that the "Enable Charge Schedule" was repeatedly being turned on and off (this example was between 03.30 and 04.00 this morning, i.e. the cheapest of the overnight slots, but there were more)
[image]

This screenshot shows it happening quite a bit overnight
[image]

Unfortunately at this point the appdaemon logs don't go far back enough to show what Predbat was actually doing in each of these events, but it's clear that the Enable Charge Schedule was being turned off, and then back on again a matter of seconds later (i.e. in the same Predbat thread).  So there's something that's causing Predbat to turn the charge schedule off then on again when it is already on and in a charge window.

The charge schedule start and finish times, and the charge rate were not constantly changing in that window though, just the 'Enable Charge Schedule' setting as far as I can see in HA.

springfall2008, 2023-10-25:
If you look at predbat.status the history should be there in HA?

slopemad, 2023-10-27:
OK, so I've just 'caught' it happening.

The predbat.status doesn't show anything:

[image]

but the log file does...

```
2023-10-28 00:08:04.552837 INFO pred_bat: Include original charge start 10-28 00:00:00, keeping this instead of new start 10-28 00:00:00
2023-10-28 00:08:04.560110 INFO pred_bat: Charge window will be: 2023-10-28 00:00:00+01:00 - 2023-10-28 06:30:00+01:00 - current soc 14.target 16
2023-10-28 00:08:04.566224 INFO pred_bat: Configuring charge window now (now 10-28 00:05:00 target set_window_minutes 30 charge start time 10-28 00:00:00
2023-10-28 00:08:04.573236 INFO pred_bat: Inverter 0 charge window is 00:00:00 - 00:30:00, being changed to 00:00:00 - 06:30:00
2023-10-28 00:08:26.803608 INFO pred_bat: Set inverter 0 charge schedule False via REST successful on retry 1
2023-10-28 00:08:26.822515 INFO pred_bat: Inverter 0 Turning off scheduled charge
2023-10-28 00:08:36.823811 INFO pred_bat: Inverter 0 set charge slot 1 {'start': '00:00', 'finish': '06:30'} via REST successful after retry 0
2023-10-28 00:08:36.896732 INFO pred_bat: Inverter 0 Updated start and end charge window to 00:00:00 - 06:30:00 (old 00:00:00 - 00:30:00)
2023-10-28 00:08:49.939237 INFO pred_bat: Set inverter 0 charge schedule True via REST successful on retry 0
2023-10-28 00:08:49.997051 INFO pred_bat: Next discharge window will be: 2023-10-28 07:20:00+01:00 - 2023-10-28 07:31:00+01:00 at reserve 78.0
2023-10-28 00:08:50.005084 INFO pred_bat: Setting ECO mode as we are not yet within the discharge window - next time is 10-28 07:20:00 - 10-28 07:30:00
2023-10-28 00:08:50.014948 INFO pred_bat: Inverter 0 Current SOC is 16.0 already at target
2023-10-28 00:08:50.024930 INFO pred_bat: Adjust reserve to target charge % (set_reserve_enable is true)
2023-10-28 00:08:50.031763 INFO pred_bat: Inverter 0 Current Reserve is 4.0 % and new target is 16 %
2023-10-28 00:08:58.997546 INFO pred_bat: Set inverter 0 reserve 16 via REST successful on retry 0
2023-10-28 00:08:59.111023 INFO pred_bat: Completed run status Charging
```

The inverter had the charge schedule enabled with a window of 00:00:00 to 00:30:00, and Predbat wanted to change that to 00:00:00 to 06:30:00.  So it turned off the charge schedule, changed the window, and then turned it back on.  Whilst in that window, and during the same run.

I also note it turned the charge schedule off again at the 00:10 run

```
2023-10-28 00:10:50.741542 INFO pred_bat: Include original charge start 10-28 00:00:00, keeping this instead of new start 10-28 00:00:00
2023-10-28 00:10:50.747103 INFO pred_bat: Charge window will be: 2023-10-28 00:00:00+01:00 - 2023-10-28 06:30:00+01:00 - current soc 15.target 16
2023-10-28 00:11:02.641905 INFO pred_bat: Set inverter 0 charge schedule False via REST successful on retry 0
2023-10-28 00:11:02.714609 INFO pred_bat: Inverter 0 Turning off scheduled charge
2023-10-28 00:11:02.720479 INFO pred_bat: Holding current charge level using reserve: 16
2023-10-28 00:11:02.730468 INFO pred_bat: Next discharge window will be: 2023-10-29 09:00:00+01:00 - 2023-10-29 09:31:00+01:00 at reserve 75.0
2023-10-28 00:11:02.736167 INFO pred_bat: Setting ECO mode as we are not yet within the discharge window - next time is 10-29 09:00:00 - 10-29 09:30:00
2023-10-28 00:11:02.742144 INFO pred_bat: Inverter 0 Current SOC is 16.0 already at target
2023-10-28 00:11:02.748045 INFO pred_bat: Adjust reserve to target charge % (set_reserve_enable is true)
2023-10-28 00:11:02.754297 INFO pred_bat: Inverter 0 Current reserve is 16.0 already at target
2023-10-28 00:11:02.804098 INFO pred_bat: Completed run status Hold charging
```

and back on again at 00:15

```
2023-10-28 00:15:50.878009 INFO pred_bat: Charge window will be: 2023-10-28 00:00:00+01:00 - 2023-10-28 06:30:00+01:00 - current soc 14.target 16
2023-10-28 00:15:50.883423 INFO pred_bat: Configuring charge window now (now 10-28 00:15:00 target set_window_minutes 30 charge start time 10-28 00:00:00
2023-10-28 00:15:50.888967 INFO pred_bat: Inverter 0 charge window is 00:00:00 - 06:30:00, being changed to 00:00:00 - 06:30:00
2023-10-28 00:15:59.671605 INFO pred_bat: Set inverter 0 charge schedule True via REST successful on retry 0
2023-10-28 00:15:59.756056 INFO pred_bat: Inverter 0 Turning on scheduled charge
2023-10-28 00:15:59.770502 INFO pred_bat: Next discharge window will be: 2023-10-29 09:00:00+01:00 - 2023-10-29 09:31:00+01:00 at reserve 75.0
2023-10-28 00:15:59.785500 INFO pred_bat: Setting ECO mode as we are not yet within the discharge window - next time is 10-29 09:00:00 - 10-29 09:30:00
2023-10-28 00:15:59.792557 INFO pred_bat: Inverter 0 Current SOC is 16.0 already at target
2023-10-28 00:15:59.798588 INFO pred_bat: Adjust reserve to target charge % (set_reserve_enable is true)
2023-10-28 00:15:59.804748 INFO pred_bat: Inverter 0 Current reserve is 16.0 already at target
2023-10-28 00:15:59.835966 INFO pred_bat: Completed run status Charging
```

But nothing had changed.  The charge windows remained the same, as did the target SOC.  (And I can't imagine it's particularly efficient to hold the charge in this way?)

springfall2008, 2023-10-28:
I've release a fix now

closed 2023-12-29 by springfall2008