springfall2008/batpred #197: Inverter ignores instructions to grid charge

gcoan, 2023-10-22:
This is a self-inflicted issue, but nevertheless I think its worth logging to (a) see if there is anything more that predbat could do, and (b) add some more to the predbat FAQ's / how-to setup instructions as this issue has caught me out twice in two days.

Yesterday my home assistant got corrupted and I spent most of the evening recovering it, eventually having to recover from a backup on the 16th, so lost several days of history.

Whilst I was recovering HA and predbat I manually set an inverter overnight charge (Grid+Solar) using the app from 2-5am to 50%.  Finally got HA and most of the history back so turned off the manual overnight charge (Grid mode) and left predbat to run overnight.

This morning quite surprised to find that neither battery had charged, despite predbat setting the power reserve, charge rate, etc:
[image]

When I set the overnight charge schedule using the app the following settings in red were changed:
[image]

But when I turned off the scheduled charge via the app, the blue setting (AC charge enable) was turned off [image]

I don't think Enable AC Charge Upper % Limit or AC Charge Upper % Limit is exposed by GivTCP,
[image]

Enable Charge Schedule is exposed and this did toggle on for the charge period:
[image]

There's something here that is causing the inverters not to charge up despite being instructed to do so by predbat.  Both inverters didn't charge overnight.   The previous night (and a few nights before) only 1 inverter charged up which I traced to AC Charge Upper Limit % being set to 4% on that inverter.  Doing a 'reset to defaults' cured this.

Both inverters are today charging fine off solar, and the day before were both charging from grid when I set Force Charge=120 (2 hour) force charge in the Octopus Power Up event.   The issue seems to be overnight with predbat not being able to cause a charge by setting the reserve %,
I can do a reset to defaults and see if this fixes the issue again this time,  but is there anything more predbat can do/check to cause the charge to work, or is there more that would be useful to add to the FAQs etc

Logfile:
[appdaemon.log.2.txt](https://github.com/springfall2008/batpred/files/13063036/appdaemon.log.2.txt)

gcoan, 2023-10-23:
Inverter 1 charged ok overnight from the grid but inverter 0 which has the larger battery didn’t.  I was up until nearly 4am trying to work out why it wouldn’t charge, but the inverter was just flashing the green Icon and saying “waiting “ on the GivEnergy portal.  Just wouldn’t ac charge but charged from solar ok during the day.
So looks like an inverter issue.  Talking to GivEnergy about it

springfall2008, 2023-10-25:
That's frustrating, I'm not sure why. The AC Charge upper limit % is the target SOC in GivTCP, while AC Charge enable is the Enable Charge schedule. The enableAC charge upper % is enabled and disabled by GivTCP based on the target % soc setting.

I hope GE help with your issue.

gcoan, 2023-10-28:
I did get it resolved, but not immediately by GivEnergy.  According to the console log they read a number of settings (including some I don't normally see like 115 Meter Direction, CT Direction, On/Off State and more), and they changed Enable AC Charge Upper % Limit from False to True, changed the AC Charge 1 Start and End Time and set AC Charge Enable to True.
I'm trying to remember precisely what happened now but I think this still didn't work properly, the inverters charged but one of the inverters got stuck after a while not discharging to meet house load when it should do.
I reset to defaults (twice? because one of them timed out) and restarted the inverters again and since then its been behaving perfectly.
Strange behaviour.  Hard to know how to prevent this and its infrequent and not predictable that writing automations to detect "inverter stuck" is hard

closed 2023-10-30 by gcoan