springfall2008/batpred #204: Solis Inveters

fboundy, 2023-10-27:
Just found this having got some of the way to writing my own version for Solis inverters using very similar logic and sources. 

Given that Solis use TCP ModBus I presume it should be straightforward enough to add support?

I'll take a look at what would be needed.

springfall2008, 2023-10-29:
Have a look at this ticket: https://github.com/springfall2008/batpred/issues/181

I think if you can find an existing integration of Solis with HA then we can try to configure Predbat for it.

fboundy, 2023-10-29:
I’ve already got the Solis talking to HA (https://github.com/fboundy/ha_solis_modbus) so I’ll go ahead and fork the repo with a branch for a Solis config. Might be a while before I can look at it properly but I think we can get it working.

Some people use the Solax integration with Solis inverters. I haven’t so far - that might be the ne t step

springfall2008, 2023-10-29:
The first step is to find the controls in HA, then map them to the apps.yaml for Predbat. Then see what's missing or different.

Predbat already has an inverter type setting so you can use that for any customisations, it's best to do in a way it can be merged back afterwards

fboundy, 2023-10-29:
Have it running but not sure it’s making sense yet. Have set the type to GS for Ginlong Solis and started a ginlong-solis.yaml cong file. All in a separate branch for now. PR in due course.

Bart39, 2023-10-30:
Interested in your progress, I have a solax inverter and using https://github.com/wills106/homeassistant-solax-modbus (which also supports solis amongst others)

closed 2023-11-05 by fboundy