springfall2008/batpred #207: Feature request: visual representation of 30 minute slots where predbat will charge/discharge

gavintillott, 2023-10-28:
I find it quite difficult to see the individual 30 minute Agile slots where predbat will charge / discharge; is there a way to add to predbat so that this could be visualised, ideally integrated into the Octopus energy rates card:

[image]

gcoan, 2023-10-28:
I've been thinking the same. The information is in the log files per hour, but it's quite cryptic, g+<, bf- etc

I was thinking something that could be simply graphed, charging, Eco, force discharging and holding charge being the sort of states over time we need to see. Possibly with soc (kw or %) as an optional attribute.
But I like the idea of integrating it into the agile price dashboard card as this is I find the most useful way of seeing the forward agile plan

Redding43, 2023-10-28:
I agree. I am still getting my head around much of this but it would be great to combine the battery graph with tariff details to more easily see what associated tariff is being used for forced charge and discharge

springfall2008, 2023-10-28:
I've released a beta of this in the latest version, please see the release notes for details:

https://github.com/springfall2008/batpred/releases/tag/v7.10

[image]

gcoan, 2023-10-28:
I've just been fiddling around with mashing up some of the Apex charts which could give an alternative view of what predbat is planning on doing.
I've taken the home battery prediction graph, reduced it down to just showing 2 hours of history and the forward 10 hours of predicted activity, and then mashed in some of the energy rates chart

Here's the YAML code for this card:
```
type: custom:apexcharts-card
header:
  show: true
  title: Home Battery Prediction (10 hours)
  show_states: true
  colorize_states: true
graph_span: 12h
span:
  start: minute
  offset: '-2h'
now:
  show: true
yaxis:
  - min: 0
series:
  - entity: predbat.soc_kw_h0
    stroke_width: 1
    curve: smooth
    name: actual
    extend_to: now
    show:
      in_header: raw
  - entity: predbat.soc_kw_best
    stroke_width: 3
    curve: smooth
    name: best
    data_generator: >
      let res = []; for (const [key, value] of
      Object.entries(entity.attributes.results)) { res.push([new
      Date(key).getTime(), value]); } return res.sort((a, b) => { return a[0] -
      b[0]  })
    show:
      in_header: raw
  - entity: predbat.soc_kw_best10
    stroke_width: 1
    curve: smooth
    name: best10
    data_generator: >
      let res = []; for (const [key, value] of
      Object.entries(entity.attributes.results)) { res.push([new
      Date(key).getTime(), value]); } return res.sort((a, b) => { return a[0] -
      b[0]  })
    show:
      in_header: raw
  - entity: predbat.best_charge_limit_kw
    stroke_width: 3
    curve: stepline
    name: charge_limit_best
    type: area
    opacity: 0.2
    data_generator: >
      let res = []; for (const [key, value] of
      Object.entries(entity.attributes.results)) { res.push([new
      Date(key).getTime(), value]); } return res.sort((a, b) => { return a[0] -
      b[0]  })
    show:
      in_header: raw
  - entity: predbat.best_discharge_limit_kw
    stroke_width: 1
    curve: stepline
    name: discharge_best
    data_generator: >
      let res = []; for (const [key, value] of
      Object.entries(entity.attributes.results)) { res.push([new
      Date(key).getTime(), value]); } return res.sort((a, b) => { return a[0] -
      b[0]  })
    show:
      in_header: raw
  - entity: predbat.grid_power_best
    name: power
    stroke_width: 1
    type: area
    opacity: 0.2
    data_generator: >
      let res = []; for (const [key, value] of
      Object.entries(entity.attributes.results)) { res.push([new
      Date(key).getTime(), value*5]); } return res.sort((a, b) => { return a[0]
      - b[0]  })
    show:
      in_header: raw
  - entity: predbat.rates
    stroke_width: 2
    curve: stepline
    name: import
    data_generator: >
      let res = []; for (const [key, value] of
      Object.entries(entity.attributes.results)) { res.push([new
      Date(key).getTime(), value]); } return res.sort((a, b) => { return a[0] -
      b[0]  })
    show:
      in_header: raw
```

And the resultant apex chart, you can see the import rates dropping (in dark green), charging periods in (purple), the soc rising (blue for best and red for best10, but this is overnight so they're the same, so hence appear as a purple line), with charging spikes (light green):
[image]

gcoan, 2023-10-28:
I installed the latest beta, 7.10.1, and added the custom charging chart per the readme (neat the way you have fed it with your own custom series and it all *just works*), but its not displaying right.
1. No 'charge' rates displaying, despite them being in the entity attributes 
2. I think because I am forecasting 48 hours ahead as per the advice for agile rates (forecast_plan_hours - 48 ) I am getting all the rates duplicated:

[image]
[image]

here's the card setup:
[image]

part of the attributes from predbat_current_rates_import 👍 
[image]

springfall2008, 2023-10-28:
Did you uninstall the original card?

gcoan, 2023-10-28:
fixed it and re-edited my comment above.  hadn't appreciated that you had forked the code!

gcoan, 2023-10-28:
Just realised that as the clocks go back, the Octopus rates API is giving TWO 1am and 1:30am slots!
[image]

Most confusing for scheduling things

gavintillott, 2023-10-28:
@springfall2008, this is exactly what I wanted, awesome work; I'll test it over the next few days. @gcoan you can add <showday: true> to the new card, that'll bring the day back to the display

gcoan, 2023-10-28:
Ah yes, brilliant, got the dates back.

This does show why I’m seeing so many rows in the predbat output vs the standard display driven off the Octopus API.

The octopus api is giving rates to Sunday 10:30 whereas predbat is predicting up to Tuesday 10pm! Of course those rates don’t exist.  That’s 3 days ahead?

Personally would still like to see import and export on the same dataset, alongside each other. 

And another thing just noticed, my phone must have changed to night mode as all the unpopulated activities are white boxes, not background colour showing through.  Looks very weird

Thanks

gcoan, 2023-10-28:
Looking at it just now, there's something not quite right.

Maybe the clocks changing at 1am and there being duplicated time slots might explain the second 1:30, but doesn't explain the 5:30 activity - it appears that predbat wants to both charge and discharge in the same 30 minute timeslot.
Doesn't sound right to me?

[image]
[image]

gavintillott, 2023-10-29:
This might be correct if you have switch.predbat_calculate_discharge_oncharge set to true - this allows charge and discharge in the same time slot I think.

springfall2008, 2023-10-29:
Predbat slots go down to 5 minutes so it doesn’t mean they overlap as such but are just within the same 30 minute timeframe

gcoan, 2023-10-29:
Yes I do have calculate discharge on charge slots set to true, and if predbat is planning down to 5 minutes, that’d explain it.

Thanks

gavintillott, 2023-10-29:
Off topic formatting Q - why does my card not look like like yours @gcoan? 

[image]

springfall2008, 2023-10-29:
I've thought about making my own instead, what do you think of this?

[image]

gavintillott, 2023-10-29:
That's really nice @springfall2008. As I'm on Agile, I have days' worth of slots on my card so it would be good to flag those rates that are not yet published, with a crystal ball icon or similar. Also, what does it look like in dark mode?

springfall2008, 2023-10-29:
I released a first version here: https://github.com/springfall2008/batpred/releases/tag/v7.10.2

Yes I think it's a good idea to flag unknown rates, maybe put them in Itallic or similar.

springfall2008, 2023-10-29:
I've put unknown rates in italics and with a ? now
I've removed the octopus rates card as I'd prefer not support both.

https://github.com/springfall2008/batpred/releases/tag/v7.10.3

Redding43, 2023-10-29:
[image]

looks good

Only takes up half the page width but that’s more to do that I can’t get the Lovelace layout card option field to show for dashboards, so can’t use it to render the whole dashboard in a easy way

gcoan, 2023-10-29:
Sorry late in replying, I think this looks great, more information and takes up less room than the octopus rate card, plus as you say Trefor, one thing less to support.

I’ll install it myself and give it a look

Not forgotten your Q about formatting Gavin

Geoffrey

Redding43, 2023-10-29:
@springfall2008 , dark mode will need some work

[image]

slopemad, 2023-10-30:
Suggestion for the Predbat plan card - to perhaps add a column for whether there is car charging planned for that slot?

springfall2008, 2023-10-30:
Good idea

I think I fixed dark mode in the last release.

Redding43, 2023-10-30:
@springfall2008 - I see this in dark mode after taking the latest update? Am I doing something wrong?

[image]

gavintillott, 2023-10-30:
Yeah I get the below in v7.10.5 when in dark mode:
[image]
[image]

gcoan, 2023-10-30:
> Off topic formatting Q - why does my card not look like like yours @gcoan?

I suspect its because of the view type I have on my dashboard.
I've got it set to "Grid (layout card)" which comes from the layout card HACS front end integrations https://github.com/thomasloven/lovelace-layout-card
Provides 4 excellent new dashboard layout styles.  Grid is particularly useful I find, even if you don't define any formatting of the grid dimensions or columns:
```
  - title: Agile Rates
    path: octopus-agile-rates
    icon: ''
    type: custom:grid-layout
    badges: []
    cards:
      - title: Import rates
        type: custom:octopus-energy-rates-card
        entity: predbat.current_rates_import
        cols: 2
        showday: true
        showpast: false
      - title: Export rates
        type: custom:octopus-energy-rates-card
        entity: predbat.current_rates_export
        cols: 2
        showday: true
        showpast: false
```

Redding43, 2023-10-30:
Mine is Masonary (Layout-card)

springfall2008, 2023-10-30:
I've just done a new release that should fix the text colour and tweaks the formatting a little

Redding43, 2023-10-30:
Looks better Trefor 
[image]

gcoan, 2023-10-30:
I had only just installed it, but have put  the newer version in to give the latest feedback
[image]

The green of charge and the rates seems very lurid, can it be toned down a bit?
The different import rates are not colouring on mine, all bar Tues 16:30 46.81 are bright green with prices as high as 46.11 and as low as 12.81.  Could the colours/banding more match the octopus rate card?  Export rates (flux outgoing) are white for 5.76 and red for 16.93 and 28.1 
I like the SOC direction arrows, but the staying constant is quite different in style from the up and down?
And the Charge up arrow looks very spindly

gcoan, 2023-10-30:
but it is very useful, all in one central place.

And I am still having an absolutely massive forward plan from now (Mon 7pm) to Wed 18:30. I'm guessing that's because I have forecast_hours: 48 in apps.yaml.  I know that's the recommendation but wonder if I should decrease it to say 36 as the battery will have been through a full charge and discharge cycle in that period anyway

gcoan, 2023-10-30:
I've dropped the forecast_hours down to 36 now and that has reduced the data shown in the predbat plan to match.  At 9:45pm Monday its now planning out to 9:00am Wednesday.  I may reduce it further, but this is better.

Looking at the PV data, I'm not sure I understand the significance of the asterisk on most rows, and the colour looks strange.  Grey=no PV is fine but then it goes yellow at the start and end of the day and red in the middle.  I can see from the figures that its a low PV day but not sure why some is yellow and some red?
[image]

Redding43, 2023-10-30:
PV as I read it

grey = no sun
Yellow = some sun
Red = lots of sun

the asterisks are Suns

[image]

gcoan, 2023-10-30:
Yes your sun's look more like sun's than mine which are asterisks on my Windows 10 Edge browser, but on my iPad they are indeed suns.
The colour coding just doesn't make sense to me for the PV.  If we are following green=good, red=bad, yellow in the middle then the shape (in my example) red then yellow then maybe green then back to yellow and red across the day depending on what the thresholds are.  Red for more sun than yellow seems counter-intuitive to me.

Sorry, and another request, can the car kwh column be suppressed if num_cars=0 in apps.yaml?  No point in showing car charging if you don't have a car

gcoan, 2023-11-02:
Another observation with the new predbat plan, an inconsistency in the limit %.  Looks like the charges all are a to a whole number % but the discharge is to %.0

[image]

gcoan, 2023-11-07:
Can we close this issue now with the dashboard @springfall2008 created and the tweaks I did in #281 ?

gavintillott, 2023-11-08:
Happy to close, although I haven't seen your tweaks as part of #281 - are they in a release? I have the latest version of Predbat, 7.11.10.

closed 2023-11-08 by gavintillott

gcoan, 2023-11-08:
My tweaks were accepted into base so will be in the next release