springfall2008/batpred #211: Feature Request: Octopus energy Octoplus capability

Redding43, 2023-10-28:
With Octoplus being announced again it would be good to have some of this pricing changes in the app.

Not quite sure what that could be but typically last year I would be working manually to:

- Charge battery fully on the ‘free electric’ period
- Force discharge of battery to maximise income outside of free period
- If saver period (period when demand should be limited) ensure battery does not force charge
- Prior to saver period, can a Predbat entity identify the period to potentially use my higher demand appliances?

I may have messed up the terminology of the above but hopefully it makes sense. Thanks again Trefor

springfall2008, 2023-10-28:
I think the rates override features covers this, you can enter one off import or export rates in your config file and then Predbat should do the right thing. This would be to cover power up sessions and saver sessions.

  # Import rates can be overriden with rate_import_override
  # Export rates can be overriden with rate_export_override
  # Use the same format as above, but a date can be included if it just applies for a set day (e.g. Octopus power ups)
  # This will override even the Octopus plugin rates if enabled
  #
  #rates_import_override:
  # -  date: '2023-09-10'
  #    start: '14:00:00'
  #    end: '14:30:00'
  #    rate: 5

Redding43, 2023-10-28:
So, manually change these dates and values each time the saver and power up sessions happen?

Redding43, 2023-10-28:
Thanks Trefor, are you saying that would address all my points?

if I understand correctly I would need to then manually track dates, times and rates and manually add/update them each time?

springfall2008, 2023-10-28:
The manual effort is only for power ups or saving sessions. The normal day to day rates come directly from the Octopus plug-in

gcoan, 2023-10-28:
This is what I do for the octopus power ups, have to add a zero rate override for apps.yaml.
At the moment there isn’t an octopus api with these dates on yet so can’t really do much better

Redding43, 2023-10-29:
There is a new(?) entity for saver sessions added on the octopus api. I think it will show either a time or a false state. I guess that is the first step to solving part of the problem

Redding43, 2023-10-29:
[image]

springfall2008, 2023-10-29:
When the saving sessions come though this year I'll see if it's possible to automate them, but it's hard to know the price in £, maybe needs some further information on the points etc.

adamwylde, 2023-11-06:
I see that some changes for this have been going in @springfall2008 
I’ve had a poke around, updated my existing apps.yaml to include:
octopus_saving_session: 're:binary_sensor.octopus_energy_saving_session(s|)'
As per your latest commit (nice update as my sensor is the singular so was previously not matched to the regex).

But I’m not seeing the input metric_octopus_saving_rate (input_number.predbat_metric_octopus_saving_rate)

(Not worried, just thought you should know).

Thanks for all your hard work.

gcoan, 2023-11-06:
My octopus sensor is in the plural and I’m seeing input_number.predbat_metric_octopus_saving_rate appearing ok in my home assistant

adamwylde, 2023-11-06:
Hmmm thanks for the info. You made me dig deeper. 

Log suggests the regex isn’t matching:
2023-11-06 21:39:59.024244 INFO pred_bat: Regular expression argument octopus_saving_session matched ^binary_sensor.octopus_energy_saving_session(s|)$ with 

Modified config to use:
octopus_saving_session: 're:binary_sensor.octopus_energy_saving_session'
  
And its now matching:
2023-11-06 21:47:44.690024 INFO pred_bat: Regular expression argument octopus_saving_session Matched ^binary_sensor.octopus_energy_saving_session$ with binary_sensor.octopus_energy_saving_session

And now my input has appeared!

Thanks

adamwylde, 2023-11-06:
Ah worked out my mistake. Bloody regex…

I missed the outer brackets of the regex. Fixed now:

2023-11-06 21:55:44.738265 INFO pred_bat: Regular expression argument octopus_saving_session matched ^(binary_sensor.octopus_energy_saving_session(s|))$ with binary_sensor.octopus_energy_saving_session

gcoan, 2023-11-15:
First saving session is tomorrow and I’ve signed up to it so was just looking at the octopus integration and predbat to see what needs doing

there’s new sensors exposed https://github.com/BottlecapDave/HomeAssistant-OctopusEnergy/blob/develop/_docs/entities/saving_sessions.md

but looking at mine in HA there’s nothing in them !
[image]

so per the above I’ll add an export rate override to apps.yaml

adamwylde, 2023-11-15:
You need to set that middle sensor predbat_metric_ocotopus_saving_rate to 225 which is the price per kWh for tomorrows saving session. Predbat will do the rest assuming you have Calculate Best Discharge enabled.

Hats off to @springfall2008

adamwylde, 2023-11-15:
PS. As you mentioned your saving session attibutes are empty - they should pull through with the session like this:

```
joined_events: 
- start: '2023-11-16T17:30:00+00:00'
  end: '2023-11-16T18:30:00+00:00'

next_joined_event_start: 2023-11-16T17:30:00+00:00
next_joined_event_end: 2023-11-16T18:30:00+00:00
next_joined_event_duration_in_minutes: 60
icon: mdi:leaf
friendly_name: Octopus Energy Saving Session
```

springfall2008, 2023-11-15:
Mine's picked it up correctly at least.

Lets see if it stays when it starts :)

gcoan, 2023-11-15:
Thanks @adamwylde I had looked through the predbat sensors, controls and docs and hadn't spotted that that was all I needed to do.  I'll have a go at setting the predbat_metric_ocotopus_saving_rate and see what happens.  

I was suspicious though that despite opting in on the Octopus website there was nothing in the API call so I'd have to manually script something.... but looking again just now, as if by magic (well synchronisation of systems within Octopus...)

[binary_sensor.octopus_energy_saving_sessions](https://homeassistant.local:8123/developer-tools/state#) now contains
```
joined_events:  - start: '2023-11-16T17:30:00+00:00'   
end: '2023-11-16T18:30:00+00:00'  
next_joined_event_start: 2023-11-16T17:30:00+00:00 
next_joined_event_end: 2023-11-16T18:30:00+00:00 
next_joined_event_duration_in_minutes: 60 
icon: mdi:leaf friendly_name: Octopus Energy Saving Session
```

Well since I have just finished the manual export override config, here's the plan before:
[image]

and after (the 1% discharge at 13:30 is a bit weird):
[image]

I think I've figured it out now from trial and error and watching the saving session video Trefor has just put up https://www.loom.com/share/57066bec571d42099fa1175059c837ee?sid=acee151f-cda4-40e7-9b3d-9de070882a53 but :
(a) there needs to be something in the predbat docs as to how to set up saving sessions as other than the video I couldn't find anything
(b) I think the setup could be made even simpler.

From what you'd said above I thought all I needed to do was to set the predbat input variable and verify that the octopus data looks correct (which it is):
[image]

Before I watched the video I thought I wouldn't need rates_export_override in apps.yaml as I assumed that predbat was getting the rate from this new input and the next session start/end times from the octopus API call, but if I did this (just set the predbat input number and no rates_override), predbat went back to predicting the old plan without taking accoutn of the saving session rates.

Final plan, setting the predbat input number to 225, adding import and export rate overrides, 
[image]

Much the same as the plan with just the export rate override added. I didn't think the import override was needed as I should be using the battery during that high rate period anyway.

adamwylde, 2023-11-15:
You definitely shouldn’t need the rates override. Mine is showing dumping 3.6kwh tomorrow for the duration of the session and I’ve not entered manual overrides.

arti273, 2023-11-15:
> I think I've figured it out now from trial and error and watching the saving session video Trefor has just put up [...]

In reference to (a); here is the section of the Read Me: https://springfall2008.github.io/batpred/energy-rates/#octopus-saving-sessions

I have to say the new Read Me docs are excellent and the addition of a search box is great.

I didn't encounter any setup issues myself but then I tend to click random stuff first and then if that fails then go looking for the instruction manual!

gcoan, 2023-11-15:
Thanks for the link to the appropriate part of the docs, I'd been looking in the configuration guide section https://springfall2008.github.io/batpred/configuration-guide/ not the rates section.  Doh.

OK, so great that you tell me I shouldn't need the rate override, but that was my experience and it is what Trefor's video currently says.

So, going through it again, I've deleted the import and export rate overrides from apps.yaml.  I had verified before that appdaemon was finding the saving session sensor through the regex, but to be sure, I've restarted appdaemon again and the logfile says:

2023-11-15 23:12:51.771034 INFO pred_bat: Regular expression argument octopus_saving_session matched ^(binary_sensor.octopus_energy_saving_session(s|))$ with binary_sensor.octopus_energy_saving_sessions

The Octopus and predbat entities are set correct
[image]

combine charge slots is now off (it was on, a hang-over from previously being on flux) until I saw Trefor's video and turned it off, calculate max windows is 96, rate low and high threshold are zero

and the logfile identifies that a saving session is coming:
2023-11-15 23:25:03.633695 INFO pred_bat: Next Octopus saving session: 2023-11-16T17:30:00+00:00 - 2023-11-16T18:30:00+00:00 at assumed rate 225 state False

so it _should_ be working, but it isn't.  Without the override rates its not planning a discharge tomorrow:
[image]

Something is wrong and I can't work out what it is.   I was running the beta version of main with the nordpool changes, so have upgraded to the 7.12.2 release and still not working

Here's the logfile.  For now I will put the rate overrides back in as I know that works
[appdaemon (10).log](https://github.com/springfall2008/batpred/files/13370697/appdaemon.10.log)

gcoan, 2023-11-16:
Notwithstanding above about filling in the savings period rate not flowing through automatically into extra discharge activity for my predbat, I've just realised that the export rate used in calculations isn't as simple as *just* being the saving session rate.

The saving session rate (£2.25) is in addition to the normal export rate so maybe we should be showing the sum of the export rate plus the saving session rate?
But then also, the saving session extra rate is only paid on top of the amount you normally export, so if you typically export 1kW and can manage to export 2kW then you'll get the export payment for the 2kW but only the saving session rate for 1kW.

Probably going to be too hard to get this totally accurate, but maybe adding the export and saving session rate is worthwhile?

gcoan, 2023-11-28:
I'm still not getting predbat automatically planning for the saving session.
Upgraded to v9.1 of bottlecapdave's octopus integration
Upgraded to v7.13.17 of predbat

I can see the saving session data in the octopus api's

I've double checked the docs https://springfall2008.github.io/batpred/energy-rates/

predbat says it knows about the saving session:
```
2023-11-28 21:55:01.568205 INFO pred_bat: Joined Octopus saving session: 2023-11-16T17:30:00+00:00 - 2023-11-16T18:30:00+00:00 at assumed rate 400.0 state False
2023-11-28 21:55:01.570408 INFO pred_bat: Joined Octopus saving session: 2023-11-29T17:00:00+00:00 - 2023-11-29T18:30:00+00:00 at assumed rate 400.0 state False
```
but just to be sure I edited apps.yaml, removing the regular expression and pointing predbat directly to the saving session binary sensor

input_number.predbat_metric_octopus_saving_rate is set to 400.0
combine charge slots is off
rate low threshold and high threshold are both zero

And yet there is nothing in the predbat plan for the more expensive export rates
[image]

I have not added an export price override to the apps.yaml, but per other people's experience and the manual, I shouldn't have to?
If it doesn't appear tomorrow then I will manually add the override, but why is it not working?

here's the logfile with debug turned on
[appdaemon (18).log](https://github.com/springfall2008/batpred/files/13493896/appdaemon.18.log)

gcoan, 2023-11-28:
> I'm still not getting predbat automatically planning for the saving session.

fixed it by reinstalling predbat and setting up apps.yaml again

why, I have no idea as I thought it was a good mirror of each release, and the saving sessions were being identified ok
[image]

closed 2023-12-29 by springfall2008