springfall2008/batpred #214: ERROR: Exception raised

slopemad, 2023-10-28:
I upgraded to v7.10.1 tonight at about 21:38, and all seemed to work fine.

The run at 22:40 though (so an hour after the upgrade), it errored out with:

```
2023-10-28 22:40:04.978098 INFO pred_bat: New Inverter 0 with soc_max 8.36 kWh nominal_capacity 10.44 kWh battery rate raw 2600.0 w charge rate 2.6 kw discharge rate 2.6 kw battery_rate_min 0.0 w ac limit 2.6 kw export limit 2.6 kw reserve 4.0 % current_reserve 40.0 %
2023-10-28 22:40:06.218160 INFO pred_bat: ERROR: Exception raised 'Load_Power'
```

and then subsequent runs all errored out at this point:

```
2023-10-28 23:10:03.574731 INFO pred_bat: Historical data totals for days [1, 7, 14] are [23.8, 16.25, 20.47] - min 16.25
2023-10-28 23:10:03.577844 INFO pred_bat: Historical days now [1, 7, 14] weight [1, 0.8, 0.6]
2023-10-28 23:10:03.579053 INFO pred_bat: ERROR: Exception raised float division by zero
```

slopemad, 2023-10-28:
I restarted Appdaemon (it was restarted after the upgrade, of course) and then it ran fine.

gcoan, 2023-10-28:
It’s been running fine for me

springfall2008, 2023-10-29:
I've put in some guards against this, but if this case happens again see if you can get the full error from the logfile

closed 2023-10-29 by springfall2008