springfall2008/batpred #215: Load today doesn't seem to be correct

gcoan, 2023-10-29:
I've been saving this up as an enhancement suggestion, and it pretty much works, but I've found one issue that throws it off...

I've mentioned before that I've had problems with getting historic load information into predbat, caused by having two hybrid inverters that "share" the load - they definitely don't record the same load values all the time:
[image]

I had been defining both inverter load_energy_today_kwh sensors in apps.yaml

```
load_today:
    - sensor.g_{geserial}_load_energy_today_kwh
    - sensor.h_{geserial2}_load_energy_today_kwh
```

But this would tend to over-state the daily house load.

And so for a little while I have been using a custom template sensor to calculate house load, bottom up, in the same way that the energy dashboard does:

```
{{ states('sensor.sum_pv_energy_today')|float(0) 
+ states('sensor.sum_battery_discharge_today')|float(0)
- states('sensor.sum_battery_charge_today')|float(0) 
+ states('sensor.g_sd2237g182_import_energy_today_kwh')|float(0)
- states('sensor.g_sd2237g182_export_energy_today_kwh')|float(0) }}
```

This new sensor, calc_house_load_today works fine except for occasional dips that are caused (I think) by one of the solar sensors not returning a value.  I can probably fix this by not summing the raw sensor.  But the overall shape is good and the in-day and end-of-day figures match the Energy dashboard:
[image]

Yesterday's total load was 31.5kWh and the day before 32.5kWh, day before 28.2kWh and day -4 was 39.9kWh (Octopus power up day!).

I define to use this sensor directly in apps.yaml (no intervening utility meter, etc):
```
load_today:
    - sensor.calc_house_load_today

days_previous:
    - 1
    - 2
    - 3
    - 4

days_previous_weight:
    - 1
```

Today I notice that predbat is retrieving historical daily load figures that are quite a bit higher than the sensor is recording:

```
2023-10-29 09:40:02.140157 INFO pred_bat: Today's load divergence 108.3 % in-day adjustment 100.0 % damping 1.0x, Predicted so far 6.87 kWh with 0.0 kWh car excluded and 10.19 kWh import ignored and 0.0 forecast extra, Actual so far 9.63 KWh with 0.51 kWh car excluded and 6.67 kWh import ignored
2023-10-29 09:40:02.213796 INFO pred_bat: Historical data totals for days [1, 2, 3, 4] are [42.21, 41.82, 33.37, 33.41] - min 33.37
2023-10-29 09:40:02.220687 INFO pred_bat: Historical days now [1, 2, 3, 4] weight [1, 1, 1, 1]
```

42kW hasn't been seen as a daily load for 4 days and I don't understand where this is coming from.  Especially not two consecutive days??

And as I type this in, predbat has continued running, and the most recent logfile extract shows different daily load figures:
```
2023-10-29 10:35:01.273610 INFO pred_bat: Current data so far today: load 17.97 kWh import 17.7 kWh export 7.51 kWh pv 5.28 kWh
...
2023-10-29 10:35:02.276533 INFO pred_bat: Today's load divergence 105.92 % in-day adjustment 100.0 % damping 1.0x, Predicted so far 8.85 kWh with 0.0 kWh car excluded and 10.19 kWh import ignored and 0.0 forecast extra, Actual so far 10.77 KWh with 0.51 kWh car excluded and 6.67 kWh import ignored
2023-10-29 10:35:02.350957 INFO pred_bat: Historical data totals for days [1, 2, 3, 4] are [41.32, 41.55, 34.16, 34.17] - min 34.16
2023-10-29 10:35:02.353035 INFO pred_bat: Historical days now [1, 2, 3, 4] weight [1, 1, 1, 1]
```

And surprisingly the 'current data today' doesn't match either:
[image]

From the sensors, load is 10.9kWh, import is 20.4kWh, export 8.2kWh and PV is 0.8+4.7=5.5kWh

I suspect that there's something weird with timespans of data selected going on??

Clearly getting accurate load data into predbat is critical to getting the forecast right.  I know it'll never be 100% because the ASHP load is going to vary quite a bit depending on the outside temperature, but I can live with that

Let me know if you need a logfile with debug on or any more information

Thanks !

springfall2008, 2023-10-29:
I'm not sure it's fully the cause of the problem, but day '1' isn't a good choice as it will wrap, you clearly don't have data for the future and so as the load is projected beyond 24 hours it repeats. The other days will be using real data. Keep in mind this isn't just 24 hours as the prediction runs by default for 24-48 hours depending on the next charge slot.

slopemad, 2023-10-30:
See I never realised '1' isn't a good choice.  I set mine to:

  days_previous: 
    - 1
    - 7
    - 14

  days_previous_weight:
    - 1
    - 0.8
    - 0.6

I assumed 1 day previous meant 'yesterday, midnight to midnight'?

My rationale for using 'yesterday', is because of the heat pump, it's a reasonable assumption that today will be similar to yesterday, but that any extra used by the heat pump will be captured by the in day adjustment.  I may have made an incorrect assumption here (although it seems to be working out at the moment).
.
But also for using 'a week ago' is to make sure my usual higher load/lower load days are captured.

gcoan, 2023-10-30:
No I didn’t realise that -1 was a bad idea also.  I did have mine set to 1, 2, 3, 4, 5, 6, 7, 14, 21 to average out the week as we’re at home in the week and weekends.  I was thinking more days would be better than less to smooth out peaks and troughs of daily consumption with the heat pump, etc.

I’ve changed it now to 2, 3, 4, 5, 6 and will add 7 and 8 when the history is there.

gcoan, 2023-10-30:
Its changed to 2, 3, 4, 5 now and the log file reports:

```
2023-10-30 18:00:02.522030 INFO pred_bat: Today's load divergence 101.99 % in-day adjustment 100.0 % damping 1.0x, Predicted so far 23.65 kWh with 0.0 kWh car excluded and 10.25 kWh import ignored and 0.0 forecast extra, Actual so far 24.27 KWh with 2.71 kWh car excluded and 6.05 kWh import ignored
2023-10-30 18:00:02.606863 INFO pred_bat: Historical data totals for days [2, 3, 4, 5] are [41.45, 38.39, 37.08, 39.52] - min 37.08
2023-10-30 18:00:02.616227 INFO pred_bat: Historical days now [2, 3, 4, 5] weight [1, 1, 1, 1]
```

The sensor data is:
[image]

The end of day sensor peaks are: 35.8 (yesterday), 31.5 (day before), 32.8 (day before that), 28.2 (day preceding), 39.9, 24.3, 28.0

None of these seem to match what predbat is saying is the historical day totals, and even the "shape" seems wrong where my day -4 was significantly lower.  I had assumed that these are whole day figures and based on the guidance in apps.yaml 7 is the same day a week ago so 1 should be yesterday, 2 is day before, etc.

But if I look at the following run of predbat, just 5 minutes later, the previous days history figures are slightly different:
```
2023-10-30 18:05:02.177106 INFO pred_bat: Historical data totals for days [2, 3, 4, 5] are [41.29, 38.59, 37.07, 39.59] - min 37.07
2023-10-30 18:05:02.179408 INFO pred_bat: Historical days now [2, 3, 4, 5] weight [1, 1, 1, 1]
```

Don't wholly understand why these "previous day" figures are changing in 5 minutes.  Could it be that they're not from a whole calendar day, they're from a preceding 24 hour period, so day 2 might be (now)-24 to (now)-48??
I'd probably better stop trying to guess how I am interpreting the data.

BTW the load today figure still also looks wrong:
```
2023-10-30 18:05:01.255546 INFO pred_bat: Current data so far today: load 33.23 kWh import 16.4 kWh export 9.7 kWh pv 11.1 kWh
```
Whereas the sensor says load today is 23.4

I've noticed in the past that predbat is predicting really heavy load overnight and ensuring the battery is charged enough to deliver it, which may be all linked up in this ...

Happy of course to change the sensor in any way needed to generate house load figures that predbat "likes" if that's a part of the problem?

gcoan, 2023-10-30:
The predbat plan is really really useful to be able to see what is planned for the next day.  Its a really useful additional feature that you could sort of work out from looking at the shape of the graphs, but this brings it together in a much more accessible way.
It does unfortunately now cause me to ask some more questions about why the plan is the way it is !
[image]

Its not a very strong solar PV day so the battery needs to be charged to support the house load before the start of the day.  That makes sense.  

But what I don't understand is why the battery is charged to 100% by midnight.  Yes take advantage of the 12.81p at 22:30, but why continue to charge at 21.83p and then hold the 100% charge at 20.95, 23.15 and 19.8 when its cheaper to charge the battery from 2am-4:30am?

I had a thought that this might be due to the metric battery cycle cost I had set to 4 that was dissuading predbat from letting the battery % drop and then recharging in the cheaper period when the rates were only 4 or 5p cheaper.
But changing metric battery cycle to 0, it doesn't seem to have changed this part of the plan for tonight, still charges to 100% by 00:00 and I don't see why it does this when cheaper electricity is coming - either let the battery soc drop or hold the battery at 92% from 23:00 until the recovering to 100% within the cheap charging window at 3:00am?
[image]
although there is less charging periods tomorrow 9-12am, the battery is allowed to drop before being recharged.  which makes sense (metric prevents extra cycling of the battery when the rate difference is so small)

springfall2008, 2023-10-31:
Yeh, I don't get that either. 

Can you share your battery chart and a log also?

gcoan, 2023-10-31:
It looks to be doing the same for tonight, charging prematurely from 20:00-21:30 towards 100% and not letting the battery soc drop or at least be held at a constant level until the cheaper rates from 02:00-04:30.
There is ~1kW load every half hour :-( which may or may not be wholly accurate, but with 14kW of batteries I can sustain that for 7 hours and not need to charge.
[image]

Battery chart:
[image]

Latest logfile (debug not turned on currently)
[appdaemon (1).log](https://github.com/springfall2008/batpred/files/13215227/appdaemon.1.log)

slopemad, 2023-10-31:
I have also noticed the past few nights that the plan involves charging at higher rates and holding at lower rates.  I don't have any screenshots or logs at this point, will keep an eye on it for tonight when the new rates are in at 4pm.

springfall2008, 2023-10-31:
So the log you provided did have metric cycle at 4p, together with 16% round trip losses (discharge, inverter, charge) which would make the later slots perhaps not be as cheap as you think as you need to charge more.

It maybe second pass optimisation might help if your machine has the performance to copy with it.

But as an experiment try setting battery cycle cost to 0 and losses to 0 and see what you get?

gcoan, 2023-10-31:
I had set the losses to being fairly conservative to stop too much charging and discharging of the battery for negligible gain when I was on Flux.  I'm still not sure what is the "right" losses figures as I've heard 5%, 10% or 20% mentioned by different people at different times on the GE community.

And I don't really understand why having a high loss figure would cause the battery to charge on the more expensive rates ahead of the rates dropping later on.  I can understand that it would discourage charging and discharging so I'd have thought it would be more doing 'hold charging', letting the grid supply the house load from say 8pm to midnight and then doing a charge at the cheapest rates it can for next day's load?

Changed all the losses and the battery metric to zero, the shape is now much more as expected, letting the soc drop, enough to get through the evening peak, a bit of charging after the peak to get up to midnight and then big charging overnight at the cheaper rates:
[image]

[image]

[appdaemon (2).log](https://github.com/springfall2008/batpred/files/13216973/appdaemon.2.log)

gcoan, 2023-10-31:
Latest version v4.10.9 losses=0/0/0%
[image]
[image]

Changing charging and discharging battery to 5%, leaving inverter loss on 0, metric cycle count on 0:
[image]
[image]

This looks OK also.  I do want to have _some_ energy conversion losses configured because there are some of course, but maybe they were too high?
[appdaemon (3).log](https://github.com/springfall2008/batpred/files/13219843/appdaemon.3.log)

I could turn on second pass if you think it will further optimise the plan.
Am surprised about how long its taking to run now.  I have HA running on an i5-7400 as a VM with 4vCPU and 4096Mb memory when the recommended minimum is 2vCPU and 2048Mb so I'd thought it'd be enough.  I've upped the memory a bit more but 4vCPU is the maximum

springfall2008, 2023-11-01:
So I think the real issue here is Predbat has no way to stop the battery discharging at a set level and just import, if it did then you could achieve what you wanted. Based on the options available with the higher losses it's cheaper to hold at 100% and import to avoid losses.

I might think of adding a freeze charge feature to allow this in future releases.

gcoan, 2023-11-01:
To be honest I'm pretty happy with this charging plan. By removing the inverter loss and dropping the battery loss slightly It's stopped trying to keep the battery so fully charged and isn't charging on higher rates ahead of lower rates coming overnight.  Using the full range of the battery.

Can we return back to the load figures and whether my custom sensor is right for predbat or not.

springfall2008, 2023-12-20:
Resolving this old thread

closed 2023-12-20 by springfall2008