springfall2008/batpred #225: Installation Issue

Dave-Parry, 2023-10-30:
I am a new Home Assistant user, with a Givenergy System and on an Agile tariff. I am having problems installing batpred in that in Hacs it won't accept the GitHub address and there is no AppDaemon tab under categories. I would use the Manual install method, but I'm not sure where to locate the directories "config/appdaemon/apps/". I'm sure in time I'll get used to doing this, but any help would be appreciated.

springfall2008, 2023-10-30:
Hi Dave,

Have you looked at my installation videos, I do explain how to install Predbat? https://github.com/springfall2008/batpred/blob/main/README.md#video-guides

I think you probably haven't enabled AppDeamon in HACS yet, there's a configuration option you have to set before adding the repository for Predbat.

springfall2008, 2023-10-30:
[image]

Dave-Parry, 2023-10-30:
Thanks for the quick reply Trefor, 
Didn’t see the vids at the bottom.
I will check the appdeamon setup, Thought I had done it, but probably not!!

Regards
David Parry
***@***.***

Dave-Parry, 2023-10-30:
Hi Trevor,
Have been following your first 3 vids and all seemed to go as shown, but when I got to the creating graphs part there were missing entities.
On checking entities there was only one! Predbat update. So I obviously missed a step or something. 
Also Predbat doesn’t seem to appear as an integration, add on or anything else. I can only see it in file editor under Deamon.
Will try again, but if you’ve any ideas they would be appreciated.
 
Regards
David Parry
***@***.***

springfall2008, 2023-10-30:
Check the AppDeamon log file to see if it's running and look for errors.

Predbat isn't an integration itself, it's part of AppDeamon so you can only see that inside the AppDeamon console.

springfall2008, 2023-10-30:
AppDeamon web interface is here:

http://homeassistant.local:5050/aui/index.html#/state?tab=apps

Dave-Parry, 2023-10-30:
Hi Trefor
logs say 
2023-10-30 21:24:05.239856 INFO pred_bat: ERROR: You still have a template configuration, please edit apps.yaml or restart AppDeamon if you just updated with HACS
2023-10-30 21:24:05.285643 INFO AppDaemon: App initialization complete
Have restarted app so must be yaml
Regards
David Parry
***@***.***

springfall2008, 2023-10-30:
Did you edit the apps.yaml and remove the template setting?

Dave-Parry, 2023-10-30:
Trefor,
Sorry to be a pain, but following your vid, but not sure. Should I  place the geserial on line 20 with my serial, or do I have to change all the {geserial} instances with my serial.
Promise won’t bother you again tonight!!

Regards
David Parry
***@***.***

springfall2008, 2023-10-30:
If you only have one inverter then you probably don’t need to change that at all as it will find it automatically from the template there

Dave-Parry, 2023-10-30:
Hi Trefor,
Thank you VERY much for your patience and help.
I now have it running and am seeing graphs etc. Whether the data is correct is a job for another day as is setting it up for Agile.
That you again.

Regards
David Parry
***@***.***

springfall2008, 2023-11-01:
Closing this ticket, feel free to open a new one if need be

closed 2023-11-01 by springfall2008