springfall2008/batpred #231: "State attributes for predbat.plan_html exceed maximum size"

slopemad, 2023-10-31:
Seeing this in home-assistant.log every 5 minutes

`2023-10-31 12:15:07.988 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for predbat.plan_html exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored. `

I'm not aware of it causing any actual issues though.

gcoan, 2023-10-31:
Thinking about it, the predbat html card is being generated as a sensor every 5 minutes and the history of that sensor is getting stored in the database which isn't great and it's pretty useless history.

Maybe we need to change the logging to exclude storing that sensor?  Or would creating a HA helper and putting the html in that help or is that just going to get stored as history as well?

springfall2008, 2023-10-31:
I’m pretty sure that the history of attributes are not kept, which is why I used one rather than the primary state. I think the warning is saying if you reboot you lose this value as it’s not kept at all.

gcoan, 2023-10-31:
This suggests that attribute history is retained https://community.home-assistant.io/t/history-of-sensor-attributes/393209

I know the database has been refactored and I couldn’t find anything explicit either way in history or recorder pages https://www.home-assistant.io/integrations/history/
It does say “everything “ is saved and if attribute history wasn’t you’d think this’d be clearly stated

Geoffrey

springfall2008, 2023-11-01:
It could be, I'm not sure. This message is saying it's not stored which I think is okay in this case.

I might find a better way to achieve this with a file or something in future versions, but I don't think it's an actual problem right now is it?

springfall2008, 2023-12-13:
This has now been documented with a way to disable the warning also

closed 2023-12-13 by springfall2008