springfall2008/batpred #247: Some weird interplay between battery charging and EV charging

adamwylde, 2023-11-04:
Hi @springfall2008 

I'm seeing some odd behavior where predbat seems to be not making best use of the overnight cheap window. As you'll see from the below, when it hit 23:30 last night it charged up the battery although not fully despite the dreadful weather, then let it discharge to the house overnight. A few minutes ago the graph suggested it was then going to charge for half an hour at 06:00-06:30 because I have an extra intelligent charging slot then - but now thats plan has disappeared and its not charging up. You can see in the predicition we will run out of battery today.

[a0d7b954_appdaemon_2023-11-04T05-55-14.484Z.log](https://github.com/springfall2008/batpred/files/13255915/a0d7b954_appdaemon_2023-11-04T05-55-14.484Z.log)

[image]

[image]

Any ideas please?

Thanks

adamwylde, 2023-11-04:
(I've just set best_soc_min to 9.5 to force it to charge temporarily)

adamwylde, 2023-11-04:
A couple of other observations:

I've looked through the log and it looks like I've upload one that doesnt match the images - as I mentioned at some points the plan did show charging during the intelligent slot - the log does show that, the images were slightly later where it had reverted to not charging. Of course, either way, the point is it didnt charge fully, or the whole night like usual, regardless of this bonus slot.

FYI Now that I've set best_soc_min to 9.5, the prediction is showing the 'charge_limit_base' for tomorrow matching todays extra Intelligent slot (6-6:30) which is odd:
[image]

springfall2008, 2023-11-04:
Somehow your plan period seems too short.
What do you have it set to in apps.yaml ?

adamwylde, 2023-11-04:
Set to defaults:

  forecast_hours: 48
  forecast_plan_hours: 24

adamwylde, 2023-11-05:
Seems to have been fine overnight, maintaining the full battery as expected right through some extra slots this morning. I didnt change anything!

[image]

springfall2008, 2023-11-06:
I think you need forecast_plan_hours to be 48 in this case, it's confused because the extra slots are assumed to be repeated tomorrow (which they aren't).

That said I normally have the allow car to charge from battery turned off and that has the same effect overall.

What are you other settings, do you have combine charge turned on?

adamwylde, 2023-11-06:
Allow car to charge from battery is off. (I'm not overly bothered it showed up tomorrows slots in the plan as by the time they are relevant they disappear anyway).
Combine charge slots is on.

I'm largely following your Intelligent settings guide.

Full options below:

[image]

Thank you

adamzest, 2023-11-08:
Having had the car fail to charge overnight I got an extra Intelligent slot this morning which I'm currently in the middle of, and the rest of the plan is now not right, although I'm sure it will fix its-self once this slot has finished! (Best SOC Min is 9.5, so the overnight period this evening should be to the max). You mentioned that the plan needed increasing to prevent the repeating IO slot which I haven't changed, but the overnight charge is well within the next 24 hours and as I say, its min 9.5kwh anyway).

[image]

Having updated the plan hours to 48 fixes the overnight charge, but also continues to show todays bonus charge on tomorrows plan?

[image]

springfall2008, 2023-11-08:
Isn’t the bonus charge the one in green as base, not the one Predbat has picked in yellow?

adamwylde, 2023-11-08:
Yes your quite right 🤦‍♂️

I guess it’s working correctly then with 48 hour plan! Thanks

closed 2023-11-12 by adamwylde